arduino/arduino-ide #1953: Add an option to favorite some boards

Zhek1ch, 2023-03-12:
### Describe the request

Would be really great to be able to add some boards to favorites, so you don't have to search up manually different boards to select one. After you added them to favorites, it would be great if it would automatically move to top and stay there as long as it is favorited. Here is a screenshot that shows possible way it can be implemented:
[image]

### Describe the current behavior

Currently that feature isn't available (

### Arduino IDE version

2.0.5

### Operating system

macOS

### Operating system version

Big Sur 11.7.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-03-15:
Two questions for refining the requirements of this ticket:
  - Is adding a not yet installed board to the favorites possible? Probably not. When a board is not installed, the `fqbn` is unavailable, and storing the board's name could be incorrect. (https://github.com/arduino/arduino-cli/issues/620)
  - Can you share the list of favorite boards with others? In other words, where should IDE2 store the user's favorite boards (platforms, examples, etc.)? IDE2 can keep it in the `settings.json`.

Zhek1ch, 2023-04-03:
I also thought of another small addition, would be great if IDE could store board associated with the port. I use same esp every day, and it's always` /dev/cu.usbserial-0001`, but every time I reopen ide and my project, or create a new project, it resets, and I have to search for the board and select it every time.