arduino/arduino-ide #1960: {build.variant} recipe is not expanded properly for debug

gautierg-st, 2023-03-15:
### Describe the problem

We are currently trying to add the debug support for STM32 in Arduino IDE 2.x.

We added the following to platform.txt
```
debug.executable={build.path}/{build.project_name}.elf
debug.toolchain=gcc
debug.toolchain.path={compiler.path}
debug.toolchain.prefix=arm-none-eabi-
debug.server=openocd
debug.server.openocd.path={runtime.tools.xpack-openocd-0.12.0-1.path}/bin/openocd
debug.server.openocd.scripts_dir={runtime.tools.xpack-openocd-0.12.0-1.path}/openocd/scripts
debug.server.openocd.script={runtime.platform.path}/variants/{build.variant}/{build.openocdscript}
```
An openocd.cfg file was added in each variant directory and a the boards.txt looks like this
```
################################################################################
# Nucleo 64 boards

Nucleo_64.name=Nucleo-64

Nucleo_64.build.core=arduino
Nucleo_64.build.board=Nucleo_64
Nucleo_64.build.variant_h=variant_{build.board}.h
Nucleo_64.build.st_extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial}
Nucleo_64.build.openocdscript=openocd.cfg ####### openocd file defined here ######
Nucleo_64.upload.maximum_size=0
Nucleo_64.upload.maximum_data_size=0

# NUCLEO_C031C6 board
Nucleo_64.menu.pnum.NUCLEO_C031C6=Nucleo C031C6
Nucleo_64.menu.pnum.NUCLEO_C031C6.node="NOD_C031C6"
Nucleo_64.menu.pnum.NUCLEO_C031C6.upload.maximum_size=32768
Nucleo_64.menu.pnum.NUCLEO_C031C6.upload.maximum_data_size=12288
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.mcu=cortex-m0plus
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.board=NUCLEO_C031C6
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.series=STM32C0xx
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.product_line=STM32C031xx
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.variant=STM32C0xx/C031C(4-6)(T-U)
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.cmsis_lib_gcc=arm_cortexM0l_math
Nucleo_64.menu.pnum.NUCLEO_C031C6.build.extra_flags=-D{build.product_line} {build.xSerial} -D__CORTEX_SC=0

# NUCLEO_F030R8 board
Nucleo_64.menu.pnum.NUCLEO_F030R8=Nucleo F030R8
Nucleo_64.menu.pnum.NUCLEO_F030R8.node="NODE_F030R8,NUCLEO"
Nucleo_64.menu.pnum.NUCLEO_F030R8.upload.maximum_size=65536
Nucleo_64.menu.pnum.NUCLEO_F030R8.upload.maximum_data_size=8192
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.mcu=cortex-m0
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.board=NUCLEO_F030R8
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.series=STM32F0xx
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.product_line=STM32F030x8
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.variant=STM32F0xx/F030R8T
Nucleo_64.menu.pnum.NUCLEO_F030R8.build.cmsis_lib_gcc=arm_cortexM0l_math

```

With the above configuration, if we run a compilation for Nucleo F030R8, it works as expected, but if we try to debug we have the following message:
```
[2023-03-15T16:31:11.029Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\gautierg\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-openocd\\0.12.0-1/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\gautierg\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2023215-25036-1m4spc4.y2xv\\sketch_mar15a" -f "C:/Users/gautierg/AppData/Local/Programs/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f "C:\\Users\\gautierg\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\hardware\\stm32\\2.5.0-dev/variants/STM32C0xx/C031C(4-6)(T-U)/openocd.cfg"
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
C:\Users\gautierg\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.5.0-dev/variants/STM32C0xx/C031C(4-6)(T-U)/openocd.cfg:15: Error: Can't find target/.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 28
at file "C:\Users\gautierg\AppData\Local\Arduino15\packages\STMicroelectronics\hardware\stm32\2.5.0-dev/variants/STM32C0xx/C031C(4-6)(T-U)/openocd.cfg", line 15
[2023-03-15T16:31:11.162Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
As we can see, the {build.variant} is not properly expanded to our STM32F030 but to a STM32C0 which is the first to appear in the boards.txt after the declaration of the `build.openocdscript`

### To reproduce

Use code from https://github.com/stm32duino/Arduino_Core_STM32/pull/1976 and try debugging a Nucleo that is not the first in its category.

### Expected behavior

{build.variant} should be expanded with the correct value so that we could debug.

### Arduino IDE version

2.0.4

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-03-15:
Probably related: https://github.com/arduino/arduino-ide/issues/1905.

fpistm, 2023-03-17:
Hi @kittaakos 
the mentionned PR https://github.com/stm32duino/Arduino_Core_STM32/pull/1976 have been updated to workaround this issue.
Original code of this PR available here:
https://github.com/fpistm/Arduino_Core_STM32/tree/pr-1976-ori-openocd

fpistm, 2024-07-10:
Hi,
@gautierg-st I guess this one could be closed as debug specifications has changed. So this issue is deprecated.

closed 2024-07-11 by gautierg-st