arduino/arduino-ide #1968: Compiler work for ESP32 and AVR64DD32 is broken on Arduino IDE 2.0.4

HonestQiao, 2023-03-20:
### Describe the problem

When I use Arduino IDE 2.0.3, everything is OK!
After upgraded to Arduino IDE 2.0.4, Compiler work for ESP32 and AVR64DD32 is broken.

The error is :
```
exec: "bash": cannot run executable found relative to current directory

Compilation error: exec: "bash": cannot run executable found relative to current directory
```

When I use the Nightly Builds, version 2.0.5-nightly-20230320, the same error occurs.

### To reproduce

First, Upgrade to Arduino IDE 2.0.4

Then, use Blink example for test.

Test for ESP32:
1. Add board support for ESP32, keyword is esp32, version is [2.0.7](https://github.com/espressif/arduino-esp32)
2. Open a Blink example in 01.Basic
3. set Board as an any type of ESP32. For instance , choose ESP32S2 Dev Module
4. Compile it
5. Error throw:
[image]

```
Use the platform's 'esp32' development board, in the folder listed: /Users/HonestQiao/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7
Use the platform's 'esp32' code, in the folder listed: /Users/HonestQiao/Library/Arduino15/packages/esp32/hardware/esp32/2.0.7

bash -c "[ ! -f \"/private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/.arduinoIDE-unsaved2023220-73893-wztoq8.ykqbh/Blink\"/partitions.csv ] || cp -f \"/private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/.arduinoIDE-unsaved2023220-73893-wztoq8.ykqbh/Blink\"/partitions.csv \"/private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D\"/partitions.csv"

exec: "bash": cannot run executable found relative to current directory

Compilation error: exec: "bash": cannot run executable found relative to current directory
```

Test for AVR64DD32:
1. Add board support for AVR64DD32, keyword is DxCore, version is [1.5.6](https://github.com/SpenceKonde/DxCore)
2. Open a Blink example in 01.Basic
3. set Board as AVRDD-serises(no bootloader)
4. Compile it
5. Error throw:
```
Use the platform's 'avrdd' development board, in the folder listed: /Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6
Use the platform's 'dxcore' code, in the folder listed: /Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6
The library being used is being detected. . .
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr64dd32 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DTWI_MORS_SINGLE -DMILLIS_USE_TIMERB2 -DCORE_ATTACH_ALL -DARDUINO=10607 -DARDUINO_avrdd -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.5.6\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=5UL -DDXCORE_PATCH=6UL -DDXCORE_RELEASED=1 -DSPM_FROM_APP=-1 -DMVIO_ENABLED -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore/api/deprecated -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/variants/32pin-ddseries /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/sketch/Blink.ino.cpp -o /dev/null
Generate a function prototype. . .
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr64dd32 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DTWI_MORS_SINGLE -DMILLIS_USE_TIMERB2 -DCORE_ATTACH_ALL -DARDUINO=10607 -DARDUINO_avrdd -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.5.6\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=5UL -DDXCORE_PATCH=6UL -DDXCORE_RELEASED=1 -DSPM_FROM_APP=-1 -DMVIO_ENABLED -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore/api/deprecated -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/variants/32pin-ddseries /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/sketch/Blink.ino.cpp -o /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/HonestQiao/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/preproc/ctags_target_for_gcc_minus_e.cpp
Project is being compiled. . .
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-g++ -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mrelax -mmcu=avr64dd32 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DTWI_MORS_SINGLE -DMILLIS_USE_TIMERB2 -DCORE_ATTACH_ALL -DARDUINO=10607 -DARDUINO_avrdd -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.5.6\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=5UL -DDXCORE_PATCH=6UL -DDXCORE_RELEASED=1 -DSPM_FROM_APP=-1 -DMVIO_ENABLED -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore/api/deprecated -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/cores/dxcore -I/Users/HonestQiao/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.6/variants/32pin-ddseries /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/sketch/Blink.ino.cpp -o /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/sketch/Blink.ino.cpp.o
The library is being compiled. . .
Compiling the kernel. . .
Use precompiled code: /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/cores/e0dbb896863e7bbeb58d4cc0a5d38be7/core.a
Link everything together. . .
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-gcc -Wall -Os -g -flto -fuse-linker-plugin -mrelax -Wl,--gc-sections,--section-start=.text=0x0,--section-start=.FLMAP_SECTION1=0x8000,--section-start=.FLMAP_SECTION2=0x10000,--section-start=.FLMAP_SECTION3=0x18000 -mmcu=avr64dd32 -o /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.elf /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/sketch/Blink.ino.cpp.o /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/../../cores/e0dbb896863e7bbeb58d4cc0a5d38be7/core.a -L/private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D -lm
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy -O binary -R .eeprom /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.elf /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.bin

/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.elf /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.eep
/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objcopy -O ihex -R .eeprom /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.elf /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594931CEABD473D/Blink.ino.hex

bash -c "/Users/HonestQiao/Library/Arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/avr-objdump --disassemble --source --line-numbers --demangle --section=.text /private/var/folders/4n/psszlnqn73x1vwzl44cq258h0000gn/T/arduino/sketches/92B375FE5BC74A234594 [... cut ...]
### Operating system

macOS

### Operating system version

macOS Big Sur 11.7.4

### Additional context

OS: macOS Big Sur 11.7.4
bash: GNU bash 5.2.2(1)-release (x86_64-apple-darwin20.6.0)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

cjolly, 2023-03-20:
I can confirm the ESP32 behavior. I get the same error message when trying to compile / upload on MacOS BigSur 11.7

HonestQiao, 2023-04-28:
On Arduino IDE 2.1.0, the same problem still occurs.

imRohan, 2023-06-28:
Same result on Ardunio IDE 2.1.1

daz, 2023-07-16:
Still the same issue on 

macOS 13.0

Arduino Version: 2.1.2-nightly-20230716
Date: 2023-07-16T03:17:59.660Z (9 hours ago)
CLI Version: 0.33.1

jackzzj, 2023-09-06:
Just exists for apple silicon in the latest version ...

JaguarRocka, 2024-03-11:
I have the same issue with 2.3.2 version and Apple M1 Pro Ventura 13.2.1.  I can't verify my board or upload ... 

Someone found an solution ?

1990eam, 2024-05-20:
Having this issue with Arduino IDE 2.3.2 and macOs 13.5.1.