arduino/arduino-ide #1977: Serial monitor can't auto scroll [Version 2.0.4]

sarfat96, 2023-03-23:
### Describe the problem

Serial monitor can't auto scroll

### To reproduce

```c
void setup() {
  Serial.begin(9600);
}

void loop() {
  static int number = 0;
  Serial.println(number);
  number++;
  delay(200);
}
```

### Expected behavior

The serial monitor must be auto scroll, my Arduino IDE have been updated to V2.0.4 but still have same issue like V2.0.3

### Arduino IDE version

2.0.4

### Operating system

Windows

### Operating system version

Windows 11

### Additional context

Version: 2.0.4
Date: 2023-02-27T16:14:28.576Z
CLI Version: 0.31.0

Copyright © 2023 Arduino SA

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-03-24:
Hi @sarfat96. Thanks for your report.

Do you have autoscroll enabled in Serial Monitor? It can be turned on and off by clicking the icon at the right side of the Serial Monitor toolbar:

[image]

sarfat96, 2023-03-24:
User error: fixed

closed 2023-03-24 by sarfat96