arduino/arduino-ide #2001: Flashing ATtiny13 with avrdude works but UI reports munmap_chunk(): invalid pointer

pololevu, 2023-04-06:
### Describe the problem

I am flashing a program to an ATtiny13 using a usbtinyisp programmer. I have done this many times, I am confident my config and code are good. I am using the 'Sketch -> Upload using programmer' menu item. This invokes avrdude as expected and appears to do the right thing but after avrdude completes a successful verification of the target flash, the arduino UI reports:

munmap_chunk(): invalid pointer
Failed programming: uploading error: signal: aborted

and the UI pops up a dialog claiming the programming failed (which it didn't).

Actual error message:

Sketch uses 994 bytes (97%) of program storage space. Maximum is 1024 bytes.
Global variables use 18 bytes (28%) of dynamic memory, leaving 46 bytes for local variables. Maximum is 64 bytes.
"/home/simon/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/home/simon/.arduino15/packages/diy_attiny/hardware/avr/2022.3.10-gcc7.3/avrdude.conf" -v -pattiny13 -cusbtiny  "-Uflash:w:/tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex:i"

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/simon/.arduino15/packages/diy_attiny/hardware/avr/2022.3.10-gcc7.3/avrdude.conf"
         User configuration file is "/home/simon/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 001:007
         AVR Part                      : ATtiny13
         Chip Erase delay              : 4000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65     5     4    0 no         64    4      0  4000  4000 0xff 0xff
           flash         65     6    32    0 yes      1024   32     32  4500  4500 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          2    0      0     0     0 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9007 (probably t13)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "/tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex"
avrdude: writing flash (994 bytes):

Writing | ################################################## | 100% 0.64s

avrdude: 994 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex:
avrdude: load data flash data from input file /tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex:
avrdude: input file /tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex contains 994 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.60s

avrdude: verifying ...
avrdude: 994 bytes of flash verified

avrdude done.  Thank you.

munmap_chunk(): invalid pointer
Failed programming: uploading error: signal: aborted

### To reproduce

Code I'm using is at: https://github.com/mstratman/relay-bypass/tree/main/relay-bypass/relay-bypass.ino
Board is 'ATtiny13' from DIY ATtiny (https://raw.githubusercontent.com/sleemanj/optiboot/master/dists/package_gogo_diy_attiny_index.json)
Programmer is usbtinyisp connected directly via USB.
Choose 'Sketch -> Upload using programmer' menu item.

### Expected behavior

UI should report success. 

### Arduino IDE version

Version: 2.0.4 Date: 2023-02-27T16:11:28.105Z CLI Version: 0.31.0  Copyright © 2023 Arduino SA

### Operating system

Linux

### Operating system version

bookworm (ie debian testing)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-04-07:
Hi @pololevu. I tried it with a similar configuration:

- Ubuntu 22.04
- Arduino IDE 2.0.4
- `diy_attiny:avr@2022.3.10-gcc7.3` platform
- USBtinyISP programmer
- ATtiny85 target (I don't have an ATtiny13 on hand)

I was not able to reproduce the fault you reported.

Please try this:

1. Attempt an "**Upload Using Programmer**" operation in Arduino IDE, just as you did before.
1. Wait for the operation to fail.
1. Scroll up the "**Output**" panel at the bottom of the Arduino IDE window until you see the `avrdude` command line.
   For example, in the output you shared above, the command line is:
   ```text
   "/home/simon/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C/home/simon/.arduino15/packages/diy_attiny/hardware/avr/2022.3.10-gcc7.3/avrdude.conf" -v -pattiny13 -cusbtiny  "-Uflash:w:/tmp/arduino/sketches/B1E4685C13C2DECE87BADD606D0239A0/relay-bypass.ino.hex:i"
   ```
1. Select the full `avrdude` command line.
1. Press the <kbd>Ctrl</kbd>+<kbd>C</kbd> keyboard shortcut to copy the selected text.
1. Open a terminal window.
1. Press the <kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>V</kbd> keyboard shortcut to paste the `avrdude` command line.
1. Press the <kbd>Enter</kbd> key.
1. Wait for the command to run.
1. Comment here to let me know whether that "`munmap_chunk(): invalid pointer`" error still occurs when you run the `avrdude` command directly from the terminal.

pololevu, 2023-04-10:
Thanks for the rapid response and good call, I get the same error in a terminal so that implies an issue with the version of avrdude I have. Do you know what version i should have? Currently I have:

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

Trying the command using avrdude 7.1 (which I have installed globally on my computer for flashing other devices) I get many error messages related to the diy_attiny supplied avrdude.conf file so I wonder if the issue is related to the diy_attiny board config. 

Ok, thanks for the response, I suspect the issues is actually not with the Arduino IDE after all, sorry for the bum steer. Since the flashing actually works and the error is harmless I don't really feel there is anything else to do here - I might reach out to the diy_attiny devs.

Thanks again!

closed 2023-04-10 by pololevu