arduino/arduino-ide #2007: Compiling Arduino IDE from source throws errors

eldarj, 2023-04-10:
### Describe the problem

I'm running Ubuntu and I am following the [github guide](https://github.com/arduino/arduino-ide/blob/main/docs/development.md) on how to compile Arduino IDE, but `yarn` is failing.

I installed all the dependencies needed that are listed on the above page, as well as on [Theia's prereqs guide](https://github.com/eclipse-theia/theia/blob/master/doc/Developing.md#prerequisites) 

**Issue**
When executing `yarn` in the repo root, it fails wih the following:

```
arduino-ide-extension: task: Failed to run task "go:build": exit status 127
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.

 —————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————————

 >  Lerna (powered by Nx)   Running target "prepare" failed

   Tasks not run because their dependencies failed or --nx-bail=true:
   
   - electron-app:prepare
   
   Failed tasks:
   
   - arduino-ide-extension:prepare

error Command failed with exit code 1.
info Visit https://yarnpkg.com/en/docs/cli/install for documentation about this command.
```

For some reason it seems `go:build` fails? I'm running `go1.18.1`. Do I need a different go version? Also, why isn't go documented as a prereq anywhere?

**Expected**
Succesfully run commands and move on to building, running the project, etc.

**Prereq versions**
```
$ lsb_release -d
Description:	Ubuntu 22.04.1 LTS
$ pkg-config --version
0.29.2
$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
$ make --version
GNU Make 4.3
$ node --version
v18.13.0
$ npm --version
8.19.3
$ yarn --version
1.22.19
$ go version
go version go1.18.1 linux/amd64
```

### To reproduce

* Checkout arduino-ide
* Install all dependencies listed (node >16 & <17, gcc, pkg-config, make, native-keymap, keytar, etc.)
* Run `yarn`

~ Fails with the above errors

### Expected behavior

* Expected to successfully finish the yarn install step and move on to building and running the projects 

### Arduino IDE version

branch=main;commit = eb1f247296a1993c38a65b1b5707a238a8d8a77d

### Operating system

Linux

### Operating system version

Ubuntu 22.04.1 LTS

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-04-10:
Hi! Thanks for taking the time to report the issue. You need [`task`](https://taskfile.dev/) to be available from the `$PATH` for the build.

We have to improve the documentation. Closing as a duplicate of https://github.com/arduino/arduino-ide/issues/474.

If you have `task` available and the build fails, ping me here, and I will reopen this issue.

closed 2023-04-10 by kittaakos

kittaakos, 2023-04-11:
> docs needs updating indeed.

No questions about it. That's why we have #474.

----

IDE2 needs to build the CLI if it's using [a non-released version](https://github.com/arduino/arduino-ide/blob/eb1f247296a1993c38a65b1b5707a238a8d8a77d/arduino-ide-extension/package.json#L166-L170). For the time being, please reference the CLI prerequisites [here](https://arduino.github.io/arduino-cli/0.32/CONTRIBUTING/#prerequisites).

I do not have the bandwidth to look into this issue right now, but here are my local versions:

```
% node -v && yarn -v && go version && task --version
v16.15.0
1.22.18
go version go1.19.1 darwin/amd64
Task version: v3.14.1 (h1:myTmEIbMbQx2D+g5lJvnbSqX531HmxiaQuefAqz8ISA=)
```

I will get back to this issue hopefully soon. Please bear with me. Thanks!