arduino/arduino-ide #2018: **BREAKING**: Current test/nightly builds not compatible with Ubuntu 18.04

ubidefeo, 2023-04-14:
### Describe the problem

**TLDR**:
For the time being we cannot support nightly and tester builds for Ubuntu 18.04

**Keep going, I'm interested**:
A few days back GitHub phased out runners based on Ubuntu 18.04, which resulted in breaking builds on our PR and alerted us on looking deep into the issue.
Unfortunately it seems that building Arduino IDE 2 on Ubuntu 20.04 (lowest version available for Linux runners) will not start on Ubuntu 18.04 yielding the following error
```
Error: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_arduinLHEbAx/resources/app/node_modules/nsfw/build/Release/nsfw.node)
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1800)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1170:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:1800)
    at Module.load (node:internal/modules/cjs/loader:982:32)
    at Module._load (node:internal/modules/cjs/loader:823:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1006:19)
    at require (node:internal/modules/cjs/helpers:93:18)
    at Object.<anonymous> (/tmp/.mount_arduinLHEbAx/resources/app/node_modules/nsfw/js/src/index.js:4:14)
    at Module._compile (node:internal/modules/cjs/loader:1110:14) {
  code: 'ERR_DLOPEN_FAILED'
}
```
We are aware that some of our users are still on this version, hence it is only fair that we notify you of this.

This is only true for test and nightly, of course, as we will be using 18.04 for release builds and make them available for download.
We are investigating possible ways of building compatible builds in the available GitHub runners, and have some plan B in case this cannot be fixed.

If you want to contribute to our investigation building on 22.04 and testing on 18.04 to help us find what the breaking point might be, we'd be glad to listen to your feedback.

For the time being we are still committed to making Arduino IDE 2 available to those users who need to run Ubuntu 18.04, but we're aware that a day will come when we'll have to deprecate support for it.

Thank you for your support 🙏🏼 
Arduino Tooling Team

### To reproduce

n/a

### Expected behavior

Nightly builds will successfully be released with Ubuntu 18.04 compatibility

### Arduino IDE version

any

### Operating system

Linux

### Operating system version

18.04

### Additional context

n/a

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

theofficialgman, 2023-06-21:
@ubidefeo you might want to look into what chromium does to lower the required GLIBC and GLIBCXX as part of the chromium sysroot creation scripts.

I have implemented these in other projects as part of github actions and have been able to successfully lower the required GLIBC version there.

https://github.com/libgdx/Jamepad/pull/22/commits/5eb5f30fe9d5134ffd8ef2ea44e05b59fca0433c

I am not sure if the above changes would fix this particular issue though. Chromium to my knowledge avoids anything newer than C++17. The nsfw.node binary is probably being built targeting C++20. What symbols is getting linked to when you build the `nsfw.node` binary? you should be able to see them by doing an `objdump -T nsfw.node` on the binary

kittaakos, 2023-08-01:
Let's keep an eye on the work-in-progress changes in Theia: https://github.com/eclipse-theia/theia/pull/12784.

closed 2023-10-12 by per1234