arduino/arduino-ide #2028: Update Boards version number to a different version and then change back may not work.

KurtE, 2023-04-20:
### Describe the problem

Suppose I have the Teensy board types installed, like 1.58.1, and I did a few builds, and then decide I wish to try something out on the current beta build 0.59.2.   Run a test or two, then decide to update back to 1.58.1 

And now the builds fail. 

The problem is that the tool Teensy.exe was running when I installed or reinstalled the different versions. 

### To reproduce

Update from to 0.59.2 The output window shows:
```
Tool teensy:teensy-compile@11.3.1 already installed
Tool teensy:teensy-discovery@1.58.0 already installed
Tool teensy:teensy-monitor@1.58.0 already installed
Downloading packages
teensy:teensy-tools@0.59.2
teensy:avr@0.59.2
Installing teensy:teensy-tools@0.59.2
Configuring tool.
teensy:teensy-tools@0.59.2 installed
Replacing platform teensy:avr@1.58.1 with teensy:avr@0.59.2
Uninstalling teensy:avr@1.58.1
Platform teensy:avr@1.58.1 uninstalled
Uninstalling teensy:teensy-compile@11.3.1, tool is no more required
Uninstalling teensy:teensy-tools@1.58.0, tool is no more required
Uninstalling teensy:teensy-discovery@1.58.0, tool is no more required
Uninstalling teensy:teensy-monitor@1.58.0, tool is no more required
Configuring platform.
Platform teensy:avr@0.59.2 installed
```
Now I convert back to 1.58.1  
```
Tool teensy:teensy-tools@1.58.0 already installed
Tool teensy:teensy-compile@11.3.1 already installed
Tool teensy:teensy-discovery@1.58.0 already installed
Tool teensy:teensy-monitor@1.58.0 already installed
Downloading packages
teensy:avr@1.58.1
Replacing platform teensy:avr@0.59.2 with teensy:avr@1.58.1
Uninstalling teensy:avr@0.59.2
Platform teensy:avr@0.59.2 uninstalled
Uninstalling teensy:teensy-compile@11.3.1, tool is no more required
Uninstalling teensy:teensy-tools@0.59.2, tool is no more required
Tool teensy:teensy-tools@0.59.2 uninstalled
Uninstalling teensy:teensy-discovery@1.58.0, tool is no more required
Uninstalling teensy:teensy-monitor@1.58.0, tool is no more required
Configuring platform.
Platform teensy:avr@1.58.1 installed
```
I then try building the same sketch over again and see:
[image]

```
FQBN: teensy:avr:teensyMM
Using board 'teensyMM' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1
Using core 'teensy4' from platform in folder: C:\Users\kurte\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1

Detecting libraries used...
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\cores\\teensy4" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\D900ADA56C36DEBF2BF727755AF16A83\\sketch\\borg_cube_ili9341_t3n.ino.cpp" -o nul
Alternatives for ILI9341_t3n.h: [ILI9341_t3n@1.1.0]
ResolveLibrary(ILI9341_t3n.h)
  -> candidates: [ILI9341_t3n@1.1.0]
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\cores\\teensy4" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\ILI9341_t3n\\src" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\D900ADA56C36DEBF2BF727755AF16A83\\sketch\\borg_cube_ili9341_t3n.ino.cpp" -o nul
Alternatives for SPI.h: [SPI@1.0]
ResolveLibrary(SPI.h)
  -> candidates: [SPI@1.0]
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10607 -DARDUINO_TEENSY_MICROMOD -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\cores\\teensy4" "-Ic:\\Users\\kurte\\Documents\\Arduino\\libraries\\ILI9341_t3n\\src" "-IC:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\libraries\\SPI" "C:\\Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\D900ADA56C36DEBF2BF727755AF16A83\\sketch\\borg_cube_ili9341_t3n.ino.cpp" -o nul
Alternatives for tgx.h: [TGX@1.0.0]
ResolveLibrary(tgx.h)
[... 45 lines cut ...]
### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

11

### Additional context

I did a quick search and did not find anything. 

I resolved it this time, but closing the teesy.exe application, and change versions twice again and this time it removed the old version and the whole tools directory.

@PaulStoffregen - not sure anything we can do for this on the Teensy Side/ 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

PaulStoffregen, 2023-04-22:
Pretty sure the fundamental issue is Teensy tools continue running, but Arduino CLI & IDE have an implicit (and undocumented) expectation that tools completely exit.

Later when you try to update or uninstall the Teensy tools, Arduino CLI (or IDE) isn't able to deal with the fact the a tool is still running in the background, which causes the operating system to disallow deleting its executable file.

I don't see any easy solution.  Normal tools don't have any defined protocol like pluggable discovery and monitor do, to tell the tool running in the background to quit.  But even without this, I would imagine the Arduino software could be much better about checking for errors while deleting files.

Defragster, 2023-04-22:
Trying to repro observe this - with regard to Teensy_Ports and Serialmonitor active the version change similar to KurtE was noted when Teensy.exe was CLOSED. 1.58.1 was installed then updated to Beta 0.59.2. That folder created holding only Teensy.exe it seems. SO could not complete a build until the tools were copied from a IDE 1.8.19 folder.

Went back to 1.58.1 and the 0.59.2 folder was emptied but not removed. I cannot remove this directory manually either from file explorer of a cmdline PowerShell window as it is marked 'in use' - though EMPTY?

Trying again to upgrade from updated release 1.58.1 to beta 0.59.2 I got this:

`Tool teensy:teensy-compile@11.3.1 already installed
Tool teensy:teensy-discovery@1.58.0 already installed
Tool teensy:teensy-monitor@1.58.0 already installed
Downloading packages
teensy:teensy-tools@0.59.2
teensy:avr@0.59.2
Installing teensy:teensy-tools@0.59.2
Failed to install platform: teensy:avr.
Error: 13 INTERNAL: Cannot install tool teensy:teensy-tools@0.59.2: moving extracted archive to destination dir: destination c:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\0.59.2 already exists
`
As noted it exists, though empty:
` Directory of C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools

[.]      [..]     [0.59.2] [1.58.0]
               0 File(s)              0 bytes

 Directory of C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\0.59.2

[.]  [..]
               0 File(s)              0 bytes

 Directory of C:\Users\Tim\AppData\Local\Arduino15\packages\teensy\tools\teensy-tools\1.58.0

[.]                       stdout_redirect.exe       teensy_ports.exe          teensy_secure.exe
[..]                      T4Security.jar            teensy_post_compile.exe   teensy_serialmon.exe
mktinyfat.exe             teensy.exe                teensy_reboot.exe         teensy_size.exe
precompile_helper.exe     teensy_gateway.exe        teensy_restart.exe
              13 File(s)      6,191,304 bytes`