arduino/arduino-ide #2029: Mac OS 12.6.5 CMD & S is not working

CptHolzschnauz, 2023-04-21:
### Describe the problem

The shortcut is enabled in config, the dropdown "Save" is working but the shortcut cmd & s isn't.

Swiss German Keyboard.

### To reproduce

Get a swiss german keyboard.

### Expected behavior

Save the file after pressing cmd + s

### Arduino IDE version

2.1.0

### Operating system

macOS

### Operating system version

12.6.5

### Additional context

Very well done with the cloud/local integration in 2.1! I confess, i did not check with nightly build.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-04-27:
Thanks for taking the time and reporting it to us. I could not reproduce the bug with a Swiss German keyboard layout. It worked fine with 2.1.0.

https://user-images.githubusercontent.com/1405703/234816477-c88f2721-5398-42b9-8edb-5a383901c00d.mp4

Could you please try to see if you can save resources with <kbd>⌘</kbd>+<kbd>S</kbd> with the latest Theia Blueprint application? Here is the download link: https://theia-ide.org/docs/blueprint_download#downloads

CptHolzschnauz, 2023-04-27:
Hmm, i see. Then it must be the hardware (the swiss keyboard on a 2017 iMac), i don't know.
I installed the IDE on my M2 Air with swiss keyboard and the behaviour is opposite,
the IDE does like autosave (local and remote sketch) immediately (I just changed the code without pressing a key).
See the white dot disappears automatic:

https://user-images.githubusercontent.com/68701579/234965889-15a614b4-64b2-40ea-883f-eb114508f735.mov

I don't get it tbh, is there a setting for this feature?

kittaakos, 2023-04-28:
Hi, auto-save is enabled by default in IDE2. Whenever you type, IDE2 will save your changes after a couple of hundred milliseconds. If you want to disable it, you can do it from the settings dialog.

https://user-images.githubusercontent.com/1405703/235099184-5bc5538e-2cbd-4235-8eb7-11a90f3385d2.mp4

Now that you know how to turn auto-save on/off, please help me understand your problem with CMD+S on a Swiss German keyboard. Do you have auto-save off, and when you press CMD+S, the editor content is not saved? Thanks

CptHolzschnauz, 2023-04-29:
Hi 
Didn't know that. With my M2 Air everything works a expected with the swiss german shortcut.
I'm abroad for the next days and i will test again on my Intel iMac after my return and post the results here.
Thank you for your work!
PS: The new release was a big step forward with the "git" features.

per1234, 2023-09-28:
Hi @CptHolzschnauz. We are still waiting for you to provide the information requested in the previous replies so that we can proceed with the investigation or evaluation of this.

Please comment here with the requested information.

CptHolzschnauz, 2023-10-01:
Hi @per1234 
Sorry, i forgot.
Behaviour is still the same, on the M2 Mac Air everything works as intended, with my iMac (Retina 5K, 27-inch, Late 2015) 
it's not saving with cmd+s, so 
_"Do you have auto-save off, and when you press CMD+S, the editor content is not saved? Thanks"_
Exactly, it's not saved on my intel based imac. Auto save works perfectly. 

IDE 2.2.1, both computers have a Swiss-German keyboard.

per1234, 2024-08-14:
I see we have another report of this problem: https://github.com/arduino/arduino-ide/issues/2263 Even though this one was submitted first, it is a bit ambiguous so I'm going to close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

Thanks for your report!

closed 2024-08-14 by per1234