arduino/arduino-ide #2030: Add search field to "SKETCHBOOK" view

per1234, 2023-04-22:
### Describe the request

Add a search capability to the "**SKETCHBOOK**" view.

🙂 Users will be able to quickly find a specific sketch from their sketchbook.

### Describe the current behavior

The "sketchbook" folder serves as a convenient place to store sketches. The "**SKETCHBOOK**" view of the Arduino IDE Side Panel shows a list of all the sketches from that folder in their tree structure.

Users often have a large collection of sketches.

🙁 It can be time consuming to find a specific sketch to open via the Arduino IDE UI.

### Arduino IDE version

b451e2db8c35aa119f33385de5adcf291d2fba02

### Operating system

All

### Operating system version

Any

### Additional context

The scope of the search should be set by the tab selection in the "**SKETCHBOOK**" view:

- Local Sketchbook
- Cloud Sketchbook

---

The semi-equivalent "**EXPLORER**" view (accessed via the "**Toggle Explorer View**" command) does have a filename search capability:

TODO

However, the design makes the feature difficult to discover and easy to trigger accidentally.

A design based on the Boards/Library Manager search capability will be more user friendly and consistent:

TODO

#### Additional requests

- https://forum.arduino.cc/t/arduino-ide-2-1-0-is-now-available/1117237/6

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

BallscrewBob, 2023-04-24:
https://forum.arduino.cc/t/add-sketch-name-search-capability/1118306

Include "SAMPLES" in search.

V205Github, 2023-07-09:
Yes. Would be nice!