arduino/arduino-ide #2032: Sketchbook - Opening a folder, list jumps to off-screen highlighted item

Jeff-Haas, 2023-04-24:
### Describe the problem

If you have selected something in the Sketchbook, and scroll it off the top, when you open a folder, the highlight jumps back up to the selected item.

### To reproduce

Have a list of sketches in the Sketchbook that is longer than what can be seen in the current view.  Make sure some of these are folders and not just individual sketches.

Click one to select it (it can be a sketch or a folder) so it is highlighted, and then scroll it off the screen.

Find a folder.  Click the carat on the folder to open it.  The folder will open, and the Sketchbook list will jump back to the previously selected item.  If you scroll back to the folder you clicked on, you will see that it was opened before the view jumped back to the originally selected item.

Note that this only happens if you **expand** a folder.  If you have a sketch or folder highlighted, scroll it off and then click a folder or a sketch, then the highlight moves to the new item and the view stays in the correct position.

### Expected behavior

The highlight should move to the folder that was expanded.

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

Win 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-04-25:
Thanks for taking the time and reporting it to us. Are you referring to this behavior?

https://user-images.githubusercontent.com/1405703/234285255-2c1b20f8-3d8f-4676-a98e-a7efc1470652.mp4

Jeff-Haas, 2023-04-25:
Yes, but it's not necessary to open the first folder (1736).  You only need to click it to highlight it.  Then scroll down and try to open another folder.

Note that this persists if you close the sketchbook list.  When you reopen the left-side list, the previous selection is (correctly) remembered.  When you scroll down to another folder to open it, your view jumps back up to the earlier selected item.