arduino/arduino-ide #2033: Arduino IDE launch takes minutes

SConaway, 2023-04-26:
### Describe the problem

## Issue 
When I open the IDE normally on my mac, the IDE always takes over two minutes before a window appears. Until then, the icon bounces in the Dock and Activity Monitor lists the `Arduino IDE` process as not responding (and it has launched no other processes yet). 

If I open the application in the terminal via its binary (`/Applications/Arduino\ IDE.app/Contents/MacOS/Arduino IDE`), then this problem goes away. 

## Attempted solutions
1. I've tested it with other 2.x versions, including nightlies.
2. removed all configuration (`~/.arduinoIDE`, `~/Library/Arduino15`, `~/Library/Application Support/arduino-ide`) and still no luck
3. tried packaging from source but this fails due to an error

### To reproduce

1. Install IDE
2. attempt to open

### Expected behavior

The software opens in seconds like any other application.

### Arduino IDE version

2.1.0

### Operating system

macOS

### Operating system version

anything ventura-based

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-04-27:
Thank you for taking the time and reporting the issue to us.

> ### Additional context
> I am willing to provide any information needed.

Great! Here are a few questions that could help to figure out the problem:
 - M1 or Intel chip?
 - Does the slow startup happen with earlier (2.x) versions?
 - Somebody else [has reported](https://forum.arduino.cc/t/arduino-ide-2-1-0-is-very-slow-on-my-macbook-pro/1119007/) a similar issue on the Arduino Forum. What happens when you try disconnecting from the Internet or disabling the auto-update feature? Please reference this [post](https://forum.arduino.cc/t/arduino-ide-2-1-0-is-very-slow-on-my-macbook-pro/1119007/11) on how to do it.

Thanks!

SConaway, 2023-04-30:
> M1 or Intel chip?

M1 Max

> Somebody else [has reported](https://forum.arduino.cc/t/arduino-ide-2-1-0-is-very-slow-on-my-macbook-pro/1119007/) a similar issue on the Arduino Forum. What happens when you try disconnecting from [...]

Neither disabling WiFi nor disabling autoupdates (from settings) solved the issue. 

> You could share the startup log with us. Follow [these](https://forum.arduino.cc/t/arduino-ide-2-1-0-is-very-slow-on-my-macbook-pro/1119007/6?u=kittaakos) excellent steps to gather the log files from IDE2 on macOS.

When I run from terminal, the issue goes away and it launches normally.

kittaakos, 2023-05-02:
Thank you for the update!

> > M1 or Intel chip?
> 
> M1 Max
> 
> >

Please confirm you have downloaded the IDE2 application for M1 and not Intel. Thanks!

From the download page (https://www.arduino.cc/en/software):

[image]