arduino/arduino-ide #2034: Arduino IDE 2.10: No "Go to definition" in context menu

wavoigt, 2023-04-27:
### Describe the problem

Hi, I'm missing the "Go to definition" topic in the context menu of IDE 2.10
The Autocomplete feature is working partially, it only suggests "abc" items, but no functions .

### To reproduce

Tested on AI Thinker ESP32-CAM, Adafruit ESP32 Feather and more

### Expected behavior

"Go to definition" in the context menu (right click menu)
Autocomplete  suggesting function calls

### Arduino IDE version

2.10

### Operating system

Windows

### Operating system version

10 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-04-27:
Thanks for the information. Even though the bug is not in this repository's codebase, since the symptom is most impactful to Arduino IDE users and we have already received multiple reports about it on Arduino Forum I'll go ahead and leave this issue open for now to document the bug and available workaround.

wh201906, 2023-05-28:
So this should be fixed when Arduino CLI 0.33 is used in the IDE, right?