arduino/arduino-ide #2038: Update avrdude

landfillbaby, 2023-04-28:
### Describe the request

avrdude 6.3.0-arduino17, as included with Adruino IDE 2.1.0 on Windows doesn't work with [USBasp v1.08](https://github.com/nerdralph/usbasp/releases/tag/v1.08).
I see there's 2 repos for it:
- https://github.com/arduino/avrdude-packing is on 7.1-arduino.1. Why isn't it used?
- https://github.com/arduino/avrdude-build-script is on 6.3.0-arduino19 which is still newer than the included one.
- There's also https://github.com/arduino/toolchain-avr?

### Describe the current behavior

Hangs indefinitely when connecting to my USBasp, I have to stop it with Task Manager.
I tried a test read with standalone https://github.com/avrdudes/avrdude/releases/download/v7.1/avrdude-v7.1-windows-x64.zip and it worked fine.

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

10 22H2 build 19045.2846

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

landfillbaby, 2023-04-28:
...Should this be moved to https://github.com/arduino/ArduinoCore-avr?

kittaakos, 2023-05-02:
Hi @landfillbaby, thank you for taking the time and reporting your problem.

This issue tracker is for reporting IDE2 bugs and requesting new features.

> avrdude 6.3.0-arduino17, as included with Arduino IDE 2.1.0

No platforms or libraries are included in IDE2. There is a logic to install the Arduino AVR platform (`arduino:avr`) and the [`Arduino_BuiltIn`](https://github.com/arduino-libraries/Arduino_BuiltIn) library on the first IDE2 start, but none of the automatically installed things related to your request.

Your topic is more appropriate for the Arduino Forum. I'm sure people will be able to help you out over there:

https://forum.arduino.cc/

I appreciate your understanding.

closed 2023-05-02 by kittaakos

landfillbaby, 2023-05-02:
https://github.com/search?q=org:arduino+"6.3.0-arduino17"&type=code
https://github.com/search?q=org:arduino-libraries+"6.3.0-arduino17"&type=code
where is the source code for https://downloads.arduino.cc/packages/package_index.json?
where is the appropriate place to ask to update it?

kittaakos, 2023-05-02:
You can try it here: https://github.com/arduino/library-registry.

Library Manager FAQ: https://github.com/arduino/library-registry/blob/main/FAQ.md#arduino-library-manager-faq
Library specification: https://arduino.github.io/arduino-cli/latest/library-specification/
Writing a library for Arduino: https://docs.arduino.cc/learn/contributions/arduino-creating-library-guide

mirh, 2023-05-10:
That's for `library_index.json` though, isn't it? 
What you want here instead is the list for the boards manager. 
https://arduino.github.io/arduino-cli/latest/package_index_json-specification/

https://github.com/arduino/Arduino/commit/3af92cdfc8a90874461f021bdc397c29a78dd23f would seem to give some further clue, but alas they stopped to update it years ago (god forbid [this](https://github.com/arduino/Arduino/wiki/Boards-Manager-FAQ#how-is-the-boards-list-generated) or [this](https://github.com/arduino/arduino-cli/commits/master/internal/integrationtest/testdata/installed.json) provided more details). And it's certainly not something architecture specific since many devices share the same package. 

https://github.com/arduino/toolchain-avr/issues/17
https://github.com/arduino/avrdude-build-script/issues/15
https://forum.arduino.cc/t/failure-with-ide-2-0-and-megacorex/1032918/18
If you follow the epic, I believe long time ago development moved on the second repository, and then since a year it moved on the first one mentioned in this thread. 
It had one official dainty release this winter though, so it's anybody's guess why it still didn't roll out. 
Maybe they hit https://github.com/avrdudes/avrdude/issues/1162#issuecomment-1374962281, or maybe they still didn't have the time to do proper regression testing?

landfillbaby, 2023-05-10:
i'm going to copy and paste something i wrote in discord a few days ago and then unsubscribe from this thread.

arduino's documentation is frequently wrong or just missing, their software is broken and outdated in at least 10 different ways, and their support team don't even know the product lineup. at least the hardware works. i think i'll just use c to program them from now on, without their libraries, and if i want a bootloader i'll use atmel dfu :)