arduino/arduino-ide #2042: First run hangs if IDE can't access Internet

per1234, 2023-05-04:
### Describe the problem

On the first run after a fresh installation, Arduino IDE downloads some files from the Internet:

- The `builtin:serial-discovery` [pluggable discovery](https://arduino.github.io/arduino-cli/latest/pluggable-discovery-specification/) tool
- The `builtin:mdns-discovery` pluggable discovery tool
- The Library Manager index
- The primary Boards Manager package index

🐛 If the download of these files is not possible due to lack of Internet access, the startup hangs.

This is a change from Arduino IDE 1.x, which is in a fully functional state from the start with no hard dependency on an Internet connection (though obviously Library/Boards Manager can't be used).

### To reproduce

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Rename or delete (:warning: cautiously) the following folder to effectively simulate the environment of a first run:
   - **If you are using Windows:**
     ```text
     C:\Users\<username>\AppData\Local\Arduino15
     ```
     (where `<username>` is your Windows username)
   - **If you are using Linux:**
     ```text
     ~/.arduino15
     ```
   - **If you are using macOS:**
     ```text
     ~/Library/Arduino15
     ```
1. Disconnect your computer from the Internet.
1. Start Arduino IDE.

🐛 Startup hangs at the startup animation.

🐛 The cause of the hang is not communicated to the user.

### Expected behavior

Complete startup even when the essential resources could not be installed.

This is important because, once it has started up, Arduino IDE shows notifications that communicate the Internet access dependency to the user.

- Record whether the `arduino:avr` and "Arduino_BuiltIn" installations were successful, and if not then retry on subsequent startups.
  - Note: This must be done based on a dedicated record, and not on the presence/absence of these installations because users may have later removed unwanted installations after they were successfully installed automatically.

### Arduino IDE version

097c92d904ca9b126e54b6a58030b5f4c650f075

### Operating system

Windows

### Operating system version

11

### Additional context

The regression was introduced at https://github.com/arduino/arduino-ide/commit/a36524e02ab5fbc2ef333ff67a047dfe5718b306

---

The logs include this:

```text
2023-05-03T23:57:19.823Z root ERROR Uncaught Exception: 
2023-05-03T23:57:19.823Z root ERROR Error: The index of the cores and libraries must be updated before initializing the core gRPC client.
The following problems were detected during the gRPC client initialization:
[platform-index] - code: 9, message: Loading index file: loading json index file C:\Users\per\AppData\Local\Arduino15\package_index.json: open C:\Users\per\AppData\Local\Arduino15\package_index.json: The system cannot find the file specified.
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:serial-discovery not found
[platform-index] - code: 9, message: Error loading hardware platform: discovery builtin:mdns-discovery not found
[library-index] - code: 9, message: Loading index file: reading library_index.json: open C:\Users\per\AppData\Local\Arduino15\library_index.json: The system cannot find the file specified.

    at isIndexUpdateRequiredBeforeInit (C:\arduino-tools\193-097c92d\resources\app\node_modules\arduino-ide-extension\lib\node\core-client-provider.js:388:11)
    at CoreClientProvider.evaluateErrorStatus (C:\arduino-tools\193-097c92d\resources\app\node_modules\arduino-ide-extension\lib\node\core-client-provider.js:200:16)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async ClientReadableStreamImpl.<anonymous> (C:\arduino-tools\193-097c92d\resources\app\node_modules\arduino-ide-extension\lib\node\core-client-provider.js:184:31)
```

#### Related

- https://github.com/arduino/arduino-ide/issues/784
- https://github.com/arduino/arduino-ide/issues/786

#### Additional reports:

- https://forum.arduino.cc/t/arduino-2-0-will-not-start-past-the-loading-screen/1049023/1
- https://forum.arduino.cc/t/arduino-2-0-will-not-start-past-the-loading-screen/1049023/31
- https://forum.arduino.cc/t/arduino-2-0-will-not-start-past-the-loading-screen/1049023/56
- https://forum.arduino.cc/t/arduino-ide-wont-start-on-the-windows-10/1052335
- https://forum.arduino.cc/t/arduino-ide-doesnt-get-any-further-than-splash-screen/1084721
- https://forum.arduino.cc/t/arduino-is-not-starting-hangs-on-launch-up-animation/1093871
- https://forum.arduino.cc/t/arduino-ide-2-04-not-starting-up-stuck-on-logo/1102153
- https://forum.arduino.cc/t/stuck-on-loading-screen-on-arduino-2-0-4/1101395
- https://forum.arduino.cc/t/arduino-2-0-4-not-starting/1097641
- https://forum.arduino.cc/t/macos-m1-arduino-ide-2-0-3-is-not-working-anymore/1078952/11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

dkorzhevin, 2023-11-28:
Probably, also related to the "Could not install Arduino_BuiltIn library: Error: Could not find library" error.

[image]

IM-coding, 2026-08-25:
Shouldn't user be asked if application starts to use network? 
Something like "We need to download X Mb of code to use in Y so that Z could work. Proceed?"

Had the same issue for linux. And now i have a trust issues ... what else IDE will do without me knowing?