arduino/arduino-ide #2044: macOS Apple Silicon (arm64) release is missing

martinvandiemen, 2023-05-04:
### Describe the problem

The latest release doesn't contain a build for Apple Silicion (arm64). I've noticed based on the dates in previous versions it has been added later.

### To reproduce

https://github.com/arduino/arduino-ide/releases/tag/2.1.0 < missing arm64
https://github.com/arduino/arduino-ide/releases/tag/2.0.4 < has arm64

### Expected behavior

When a new release is made also include the arm64 version.

### Arduino IDE version

2.1.0

### Operating system

macOS

### Operating system version

13.3

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-05-04:
Thanks for bringing this to our attention! The M1 build is still a manual process, and we forgot to upload the artifacts to GitHub release: #408 

Please get the `2.1.0` release for M1 from the software download page instead of the GH release page: https://www.arduino.cc/en/software.

Thank you!

per1234, 2023-06-07:
I have now added the Apple Silicon builds to the 2.1.0 GitHub release assets.

closed 2023-06-07 by per1234

HelgeSeidel, 2023-07-01:
Same issue with 2.1.1 release.