arduino/arduino-ide #2049: IDE fails to save sketch

CreativeRobotics, 2023-05-05:
### Describe the problem

Sometimes the IDE will stop saving sketches.

I just spent two hours editing code in the IDE, I was saving code regularly through the session. When I tried to compile the code the IDE wouldn't compile, the compile button got 'stuck' and the 'Compiling' pop up appeared but nothing happened. A message appeared across the bottom that said 'Offline'.

I tried to close the app but it wouldn't respond.

All through this the editor continued to work and the save option in the menu could be selected.

When I forced quit the applicatiopn and restarted it ALL the code I had written in the last two hours was gone - it had failed to save anything.

### To reproduce

No idea how to reproduce apart from having a sketch open and sending the computer to sleep several times in a row.
Perhaos this is an issue related to windows in speep mode.

I have NOT verified this in the nightly build because I can't easily reproduce it. If you insist on excluding bug reports for hard to reproduce items then you will miss some important bugs.

### Expected behavior

The IDE shouldn't allow code to be edited, and respond to the save menu being selected, but not actually save anything.

### Arduino IDE version

Version: 2.1.0 Date: 2023-04-19T15:31:10.185Z CLI Version: 0.32.2 

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

P3Fab, 2023-06-13:
I have run into a similar issue where the IDE will not save when offline, even the save popup when closing the IDE doesn't save the changes. 
To reproduce the issue: go online, open the IDE, make some changes, and disconnect from the internet. 
After the "Cloud save not available", try to save changes, I will receive a popup saying "Could not save the sketch. Please copy your unsaved work into your favorite text editor, and restart the IDE."   
Another scenario is never being online in the first place and not being able to save. I will see a popup "Arduino CLI not available." 
Sometimes you get a notification that the code did not save and other time you get no notification and you think it has saved only time find out later all of your work was lost.

per1234, 2023-06-14:
@P3Fab that is a different bug: https://github.com/arduino/arduino-ide/pull/2080

https://github.com/arduino/arduino-ide/pull/2080 was already fixed and the fix is in the nightly build of Arduino IDE. The nightly build download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

per1234, 2023-06-17:
Thanks so much for reporting this @CreativeRobotics!

I see there was a later report of the bug here: https://github.com/arduino/arduino-ide/issues/2081

The mitigation measure made in https://github.com/arduino/arduino-ide/pull/2082 is to clearly communicate to the user that the IDE was unable to save and provide instructions for how they can avoid the loss of their sketch data:

[image]

I will close this issue as resolved by https://github.com/arduino/arduino-ide/pull/2082 as was done for https://github.com/arduino/arduino-ide/issues/2081

That change does not address the cause of the IDE going into that state. I am still very interested in that. Unfortunately it would be difficult to investigate or prepare a fix for the problem if we don't have a clear understanding of the conditions it occurs under.

You mention putting the PC through a sleep cycle. The same was also mentioned as a possible factor in these forum posts:

- https://forum.arduino.cc/t/is-save-save-as-being-worked-on/1124043
- https://forum.arduino.cc/t/ide-2-1-0-freezes-during-compile-after-a-period-of-sleep/1138808

But I frequently (far more than the average user since I am regularly switching between three multiple machines as I do testing on each platform) put my computers through sleep cycles while Arduino IDE is running, yet I don't experience this problem of the IDE going into an "offline" state (where it loses communication with the backend). So it definitely doesn't occur universally on a sleep cycle.

If anyone can identify the specific conditions that produce the fault and provide use with detailed instructions we can follow to reproduce the fault, please do submit a dedicated issue report here. That would be very helpful.

closed 2023-06-17 by per1234

CreativeRobotics, 2023-11-15:
This issue is NOT resolved. I'm experiencing the same issue on windows 10 where the IDE can't save a sketch after a number of sleep cycles and when more than one instance is open. The issue should remain open until it is fixed so that everyone who is still dealing with this issue can be directed to an open issue thread to contribute information.

spilz87, 2024-03-16:
> This issue is NOT resolved. The IDE can't save a sketch after a number of sleep cycles and when more than one instance is open. The issue should remain open until it is fixed so that everyone who is still dealing with this issue can be directed to an open issue thread to contribute information. The mitigation posted doesn't resolve the actual issue, its a crude workaround that is seriously impractical if you have sketches with multiple tabs.

I confirm, I still have this issue :(

loopyengineeringco, 2024-05-26:
Also have this issue. IDE 2.3.2, Arm Mac.
The backend going offline is annoying, but it should still allow saving. Why would it not, it makes no sense. If a program crashes and can't continue, it needs to be able to save your work.
Now I have to start multiple temporary text files to copy & paste my tabs into, to then force quit the program, to then paste the contents back in. Beyond ridiculous