arduino/arduino-ide #2050: gdb-server session closed GDB server session ended. This terminal will be reused, waiting for next session to start.

krhackbarth, 2023-05-07:
### Describe the problem

I'm following the procedure "Debugging with the Arduino IDE 2" to learn how to use the Arduino debugger.  However, when I press the button to "Start debugging" I get the error "gdb-server session closed GDB server session ended. This terminal will be reused, waiting for next session to start."

### To reproduce

follow the instructions at "https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-debugger"

### Expected behavior

I would expect to be able to see the next step which is to see the code stop at the breakpoint for line 33.

### Arduino IDE version

2.1.0 and 2.1.1-nightly-20230506

### Operating system

Windows

### Operating system version

11

### Additional context

I'm testing with a MKRZERO.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

victoryred, 2023-12-21:
I'd like to add this for more detail from the GDB-server window...

Waiting for gdb server to start...[2023-12-21T17:49:53.205Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/libraries/Hash/examples/sha1 -f /tmp/.mount_arduinu9wPiS/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/lib/picoprobe_cmsis_dap.tcl
[2023-12-21T17:49:53.257Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

victoryred, 2023-12-21:
Partially solved- From the Debug Console...
Cortex-Debug: VSCode debugger extension version 1.5.1 git(be7d3c8+dirty). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-objdump --syms -C -h -w /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
Reading symbols from /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-nm --defined-only -S -l -C -p /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
Launching GDB: /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-gdb -q --interpreter=mi2 /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: /home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/libraries/Hash/examples/sha1 -f /tmp/.mount_arduinu9wPiS/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/lib/picoprobe_cmsis_dap.tcl
    Please check TERMINAL tab (gdb-server) for output from /home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd
Finished reading symbols from objdump: Time: 35 ms
/home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-gdb: error while loading shared libraries: libtinfo.so.5: cannot open shared object file: No such file or directory
Error: Unable to start GDB. Make sure you can start gdb from the command-line and run
    any command like "echo hello". If you cannot, it is most likely because "libncurses5" is not installed.
Finished reading symbols from nm: Time: 71 ms
OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.
GDB never responded to an interrupt request. Trying to end session anyways
Failed to launch OpenOCD GDB Server: Timeout.

After that, tried again and in the Debug console.....

victoryred, 2023-12-21:
Continuing after loading required libraries...fro mthe Debug Console...

Cortex-Debug: VSCode debugger extension version 1.5.1 git(be7d3c8+dirty). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-objdump --syms -C -h -w /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
Reading symbols from /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-nm --defined-only -S -l -C -p /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
Launching GDB: /home/penta/.arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/arm-none-eabi-gdb -q --interpreter=mi2 /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: /home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/libraries/Hash/examples/sha1 -f /tmp/.mount_arduinu9wPiS/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/lib/picoprobe_cmsis_dap.tcl
    Please check TERMINAL tab (gdb-server) for output from /home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd
Finished reading symbols from objdump: Time: 38 ms
Reading symbols from /tmp/arduino/sketches/8CF8FC96861539E476DC0E8FD09DD06D/sha1.ino.elf...
GNU gdb (GDB) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
.
Find the GDB manual and other documentation resources online at:
    .
For help, type "help".
Type "apropos word" to search for commands related to "word".
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
Finished reading symbols from nm: Time: 76 ms
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
sleep_until (t=) at /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/common/pico_time/time.c:398
398	/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/common/pico_time/time.c: No such file or directory.
Program stopped, probably due to a reset and/or halt issued by debugger
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x10005612 msp: 0x20041f68
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00

victoryred, 2023-12-21:
And after that, in the Debug Console I typed

>show configuration
{"output":"","token":14,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
This GDB was configured as follows:
   configure --host=
x86_64-linux-gnu --target=arm-none-eabi
	     --with-auto-load-dir=$debugdir:$datadir/auto-load
	     --with-auto-load-safe-path=
$debugdir:$datadir/auto-load
	     --with-expat
	     --with-gdb-datadir=/workdir/arm-none-eabi.x86_64/share/gdb (relocatable)
	     --with-jit-reader-dir=/workdir/arm-none-eabi.x86_64/lib/gdb (relocatable)
	     --without-libunwind-ia64
	     --without-lzma
	     --without-babeltrace
	     --without-intel-pt
	     --with-mpfr
	     --without-xxhash
	     --without-python
	     --without-python-libdir
	     --without-debuginfod
	     --without-guile
	     --disable-source-highlight
	     --enable-threading
	     --with-separate-debug-dir=/workdir/arm-none-eabi.x86_64/lib/debug (relocatable)
("Relocatable" means the directory can be moved with the GDB installation
tree, and GDB will still find it.)

victoryred, 2023-12-21:
Finally in  the gdb-server window....

Open On-Chip Debugger 0.12.0-g4d87f6dca (2023-09-11-23:12)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E661640843626C27
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0x61000000 pc: 0x10005612 msp: 0x20041f68
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Info : Padding image section 0 at 0x100121d8 with 40 bytes (bank write end alignment)
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (2184 ms). Workaround: increase "set remotetimeout" in GDB
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00

victoryred, 2023-12-21:
Retry a debugging session.....again, this time in gdb-server-1 window

Waiting for gdb server to start...[2023-12-21T20:55:59.327Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
/home/penta/.arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd -c "gdb_port 50007" -c "tcl_port 50008" -c "telnet_port 50009" -s /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/libraries/Hash/examples/sha1 -f /tmp/.mount_arduinu9wPiS/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl -f /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.2/lib/picoprobe_cmsis_dap.tcl
Open On-Chip Debugger 0.12.0-g4d87f6dca (2023-09-11-23:12)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 50008 for tcl connections
Info : Listening on port 50009 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E661640843626C27
Warn : could not claim interface: Resource busy
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error reading data: Input/Output Error
Error: error writing data: Input/Output Error
Error: CMSIS-DAP command CMD_INFO failed.

[2023-12-21T20:55:59.396Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

Bennyggcc, 2025-01-10:
@krhackbarth @victoryred Hi, do you solve this issue? I meet the same problem.