arduino/arduino-ide #2066: need to make ESP8266 and ESP32 Data upload available in Tools

bmck202, 2023-05-11:
### Describe the request

unable to upload data files in tools before sketch is uploaded.

### Describe the current behavior

unable to load data heavy files in conventional sketch format

### Arduino IDE version

IDE 2.1

### Operating system

macOS

### Operating system version

13.3 Venture

### Additional context

none at this time.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-05-11:
Thanks for taking the time to create this enhancement ticket, @bmck202!

> ### Describe the request
> unable to upload data files in tools before sketch is uploaded.
> 
> ### Describe the current behavior
> unable to load data heavy files in conventional sketch format

Could you please elaborate on what exactly is your feature request? Thank you!

frank26080115, 2023-05-11:
Please implement a method by which files from the `<sketch_folder>/data` can be uploaded to the ESP8266 or ESP32's SPIFFS memory. In IDE v1.x this was accomplished via https://randomnerdtutorials.com/install-esp32-filesystem-uploader-arduino-ide/ and https://github.com/me-no-dev/arduino-esp32fs-plugin

Yea I want this too, can't upgrade the IDE until this is available.

kittaakos, 2023-05-11:
Thank you! We have an opened ticket to track the missing support of tools in IDE2: https://github.com/arduino/arduino-ide/issues/58.

We want to keep one open issue for the same feature, so I have moved the request to https://github.com/arduino/arduino-ide/issues/58#issuecomment-1544503051.

I am closing it as a duplicate. Thank you for your understanding.

closed 2023-05-11 by kittaakos