arduino/arduino-ide #2068: Default sketch names sort out of order

drf5n, 2023-05-14:
### Describe the problem

The default sketch name template of 'sketch_MMMDDx.ino' produces names that do not sort in a sane order.

```
sketch_apr03a	sketch_apr24a	sketch_dec20b	sketch_feb09a	sketch_feb23b	sketch_jan21a	sketch_mar08b	sketch_mar27b	sketch_nov18a
sketch_apr06a	sketch_apr25a	sketch_dec21a	sketch_feb09b	sketch_feb23c	sketch_jan22a	sketch_mar09a	sketch_mar27c	sketch_nov24a
sketch_apr07a	sketch_apr27a	sketch_dec28a	sketch_feb10a	sketch_feb24a	sketch_jan23a	sketch_mar12a	sketch_mar28b	sketch_nov24b
sketch_apr12a	sketch_dec05a	sketch_dec29a	sketch_feb14a	sketch_feb26a	sketch_jan23c	sketch_mar12b	sketch_mar28c	sketch_nov25a
sketch_apr13a	sketch_dec07a	sketch_feb01a	sketch_feb16a	sketch_feb28b	sketch_jan23d	sketch_mar16a	sketch_mar29a	sketch_oct22a
sketch_apr14a	sketch_dec10a	sketch_feb01b	sketch_feb16b	sketch_feb28c	sketch_jan24a	sketch_mar18a	sketch_mar30a	sketch_sep12b
sketch_apr15a	sketch_dec17a	sketch_feb01c	sketch_feb19a	sketch_jan02a	sketch_jan29a	sketch_mar18b	sketch_may02a	sketch_sep12c
sketch_apr16a	sketch_dec18a	sketch_feb01d	sketch_feb21a	sketch_jan03a	sketch_jan31a	sketch_mar20a	sketch_may02b
sketch_apr19a	sketch_dec18b	sketch_feb03a	sketch_feb21b	sketch_jan09a	sketch_mar05a	sketch_mar21a	sketch_may02c
sketch_apr21a	sketch_dec18c	sketch_feb03b	sketch_feb22a	sketch_jan11b	sketch_mar05b	sketch_mar25a	sketch_may03a
sketch_apr23a	sketch_dec20a	sketch_feb07a	sketch_feb23a	sketch_jan13a	sketch_mar08a	sketch_mar26a	sketch_nov07a
```

January sorts after February, April comes first, and December comes second.

Using a pattern like `sketch_YYYYMMDDx` would produce names which sort in a sane order.

### To reproduce

Produce sketches with default names over the course of a year or multiple years.

### Expected behavior

I would expect auto-generated sketch names to be created with an ISO-9660-ish name and sort in a sane, chronological order.

Either include the year and numeric month, or if awkward names that encourage renaming are the goal, use a simpler scheme like `sketch_xxxx`.

### Arduino IDE version

2.1.0

### Operating system

macOS

### Operating system version

13.2.1

### Additional context

[Per  @ptillisch](https://forum.arduino.cc/t/change-default-sketch-mmmddx-names-to-sketch-yyyymmddx/1126195/8?u=davex), the code is here:

https://github.com/arduino/arduino-ide/blob/31deeebb4970b4de9925161c291084d8d5f8273d/arduino-ide-extension/src/node/sketches-service-impl.ts#L372-L415

I'm not sure how that code pads the day numbers with zeros, but maybe something like this would work as I expect it should:

    const sketchBaseName =`sketch_${today.getYear()}${today.getMonth()+1}${today.getDate()}`;

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-05-16:
> I would expect auto-generated sketch names to be created with an ISO-9660-ish name and sort in a sane, chronological order.

Hi, thanks for sharing your idea. Nice one!

@drf5n, would you be interested in contributing to this feature to IDE2? I am happy to guide you, or you can reference a very much related external contribution to change the default `.ino` file content with an advanced setting: https://github.com/arduino/arduino-ide/pull/1559/

A few remarks:

 - I would expect that the users can specify a valid date format as an [advanced setting](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings).
 - If no custom format is defined, IDE2 uses the current one.
 - IDE2 should fall back to the default behavior if an invalid date format is specified.
 - Custom date formats must not conflict with the sketch naming [spec](https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-folders-and-files). 

Alternatively, IDE2 can provide a set of predefined date formats (as an enum preference type) if such flexibility is not required.

kittaakos, 2023-05-17:
Thanks for looking at the code, @drf5n. We will handle it later, then.

> comfortable programming in .ts or how to modify the UI for advanced settings.

You don't need to change the UI at all. That's why I have referenced the other PR. Please see [this](https://github.com/arduino/arduino-ide/pull/1559#issuecomment-1281226860) comment:

> > I don't feel comfortable making change to the UI.
> 
> You do not have to. Checking your changeset, I am positive you can do it. Here are a few pointers: