arduino/arduino-ide #2069: Linux build (Appimage) fails to detect USBASP or USBTinyISP for bootloader flashing

AndKe, 2023-05-17:
### Describe the problem

Appimage of Arduino 2.1.0 claims that it can't see the USBASP or USBTinyISP programmer.
running Arduino 1.8.19  (no appimage)  uses that programmer just fine 

### To reproduce

Try to use USBASP or USBTinyISP with the appimage.

[image]

### Expected behavior

same as 1.8.19 - which just works

### Arduino IDE version

2.1.0

### Operating system

Linux

### Operating system version

Ubuntu 23.04

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-05-22:
Hi @AndKe. I'm going to ask you to provide the full output from the upload when in verbose mode.

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
   The "**Preferences**" dialog will open.
1. Uncheck the box next to **Show verbose output during: > `☑` compilation** in the "**Preferences**" dialog.
1. Check the box next to **Show verbose output during > ☐ upload**.
1. Click the <kbd>**OK**</kbd>** button.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for upload to fail.
1. After the upload fails you'll see a button on the "**Upload error**" notification at the bottom right corner of the IDE window: <kbd>**COPY ERROR MESSAGES**</kbd>. Click that button.
1. [Paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the error messages in a reply here (please use [code fencing](https://guides.github.com/features/mastering-markdown/)).

AndKe, 2023-05-22:
please note that the instructions requested for sketch->upload , but I am submitting output from "burn bootloader"
Uploading sketch (via a serial connection) works fine
I tried to post output from the bootloader-flashing using USBASP
..but it works fine now ... I do not remember if I rebooted the system in the past days or something.
Sorry - the issue went away.
```

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/andre/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/home/andre/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "/home/andre/.arduino15/packages/arduino/hardware/avr/1.8.6/bootloaders/caterina/Caterina-Leonardo.hex"
avrdude: writing flash (32730 bytes):

Writing | ################################################## | 100% 10.69s

avrdude: 32730 bytes of flash written
avrdude: verifying flash memory against /home/andre/.arduino15/packages/arduino/hardware/avr/1.8.6/bootloaders/caterina/Caterina-Leonardo.hex:
avrdude: load data flash data from input file /home/andre/.arduino15/packages/arduino/hardware/avr/1.8.6/bootloaders/caterina/Caterina-Leonardo.hex:
avrdude: input file /home/andre/.arduino15/packages/arduino/hardware/avr/1.8.6/bootloaders/caterina/Caterina-Leonardo.hex contains 32730 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 9.47s

avrdude: verifying ...
avrdude: 32730 bytes of flash verified
avrdude: reading input file "0x2F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x2F:
avrdude: load data lock data from input file 0x2F:
avrdude: input file 0x2F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.
```

closed 2023-05-22 by AndKe