arduino/arduino-ide #2073: Arduino 2.1.0 fails to open, stuck in loading page

theopenhighway, 2023-05-23:
### Describe the problem

I have installed Arduino 2.0 around April 2023 on an Asus laptop (with Windows) and been using it on a regular basis. However, today I encountered a problem where the IDE stuck in the loading page, as shown in this figure below. Never done any modifications on the software itself. Tried to reinstall the program but the problem still occurs. Have looked through similar problems with this but no documented issue on version 2.1.0, as far as I am aware of.

[image]

### To reproduce

I just clicked on the exe file and it happened. So I don't know what would be needed to reproduce it.

### Expected behavior

Arduino IDE running normally (directly showing IDE after it loads)

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

11 ver 22H2

### Additional context

I have included additional information that may help to identify the problem. Basically went to an endless loop of error codes.
[Windows PowerShell.txt](https://github.com/arduino/arduino-ide/files/11547249/Windows.PowerShell.txt)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

dankeboy36, 2023-05-24:
> I have included additional information that may help to identify the problem

Please reference this Arduino forum thread to see if it fixes your issue: https://forum.arduino.cc/t/arduino-ide-v2-hangs-at-start-up/1030052/7.

theopenhighway, 2023-05-30:
sorry for the late update. still stuck in the loading page. no error code.

[image]

kittaakos, 2023-06-02:
@theopenhighway, could you share the log files with us? You find the log files on Windows under `C:\Users\<username>\AppData\Roaming\Arduino IDE\` where `<username>` is your Windows username. Please start the app; it does not load. Quit the app, go to the log folder, and attach the most recent log file to this GitHub issue. Thank you!

theopenhighway, 2023-06-08:
anyway, i did uninstall and reinstall the software today. still show the same error code as the error code in the original post
[2023-06-08_log.log](https://github.com/arduino/arduino-ide/files/11691150/2023-06-08_log.log)

kittaakos, 2023-06-08:
Please check what does the following prints:
 - Open the _Run_ dialog with <kbd>⊞</kbd>+<kbd>R</kbd> (where ⊞ is the Windows sign on your keyboard),
 - Type `cmd` and press <kbd>Enter</kbd> (a terminal opens), and
 - Type `echo %ComSpec%` and press <kbd>Enter</kbd>.

theopenhighway, 2023-06-10:
[image]

bobqianic, 2023-07-21:
I've just run into the same issue on Windows 11. When I click on the IDE icon, I get the loading animation. But interestingly, the CPU usage remains quite low, and I've noticed that there are roughly 10 instances of the IDE executable running simultaneously.

**I've tested all the IDE2 versions and they all seem to be experiencing the exact same issue.**
@theopenhighway

per1234, 2023-09-28:
Hi @theopenhighway. Please comment here to tell us whether you are still experiencing the problem if you try again with the latest version of Arduino IDE (currently 2.2.1).

This type of problem should now be completely avoided because Arduino IDE no longer uses `cmd.exe` thanks to https://github.com/arduino/arduino-ide/pull/2067 and https://github.com/arduino/arduino-ide/issues/2112

theopenhighway, 2023-10-01:
Hello @per1234 just want to let you know, it's running properly now. Thanks for your help.

[image]

closed 2023-10-01 by theopenhighway