arduino/arduino-ide #2077: when saving as, sketch folder fills with files (libraries???)

jsmwrench, 2023-05-27:
### Describe the problem

When I save a current sketch using the Save As control, the IDE produces a folder with the same name as the sketch (this is common, I believe) but then fills the folder with numerous other files (36 other files in this case).  A lot of the files that were produced look  to be the library files from within the libraries being "included" in my sketch.

I'm not sure if this matters, but the original file was opened from an external drive.  I then used File/Save As to save it to my "sketches" folder.  Below is a screenshot of all of the files produced.

[image]

### To reproduce

I opened a sketch from an external drive, edited a couple of lines, then saved it to the Arduino/sketches folder on my computer using the File/Save As command.

### Expected behavior

Normally, the IDE creates a folder with the same name as my sketch and places the sketch inside the folder.  No other files are created.

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-05-29:
Hi @jsmwrench. Thanks for your report.

>the original file was opened from an external drive. I then used File/Save As to save it to my "sketches" folder.

Please check to see whether the unexpected 36 files are also present in the folder on that external drive.

A common misconception is that an Arduino sketch is a file. The sketch is actually a folder. When you do a "Save As" operation with a sketch, all the files from the original sketch folder are saved to the folder that is the target of the "Save As" operation. So if you had previously placed all those 36 files in the original sketch folder then it is expected that they would also be present in the new folder as well.

fwlrichard, 2023-05-29:
Save as is also dumping All Files from all sub folders into the Sketch folder.

Original Sketch Folder Containing only the Sketch and a data folder. Data folder contains 4xfile.txt files.

[image]

Open Sketch click "Save As" and as you can see now the new Sketch folder contains the sketch and all 4 of the data files from the "data" folder that is now gone.

[image]

@per1234 - All previous versions of the Arduino IDE have been able to maintain file structure when doing a "Save As. " 
Problem started with IDE Version: 2.1.0
Date: 2023-04-19T15:31:10.185Z
CLI Version: 0.32.2

jsmwrench, 2023-05-30:
None of the new files are in the original folder.  The original folder only had the sketch itself and the AccelStepper library in it.  I opened the sketch in VS Code and saved as a new file on my computer and when I did, the IDE created all of these files in the same folder as the newly saved sketch.
Folder I took it from:
[image]

New Folder:
[image]

per1234, 2023-07-13:
I am able to reproduce the fault and have updated the report to clearly describe the bug. Thanks for bringing this to our attention @jsmwrench and for your clarification @fwlrichard

commit 193583a7 (2023-11-23): fix: sketch Save As preserves the folder structure

This commit rewrites how IDE copies sketches as part of the _Save As_
operation. Instead of copying to the destination, IDE copies the sketch
into a temporary location, then to the desired destination.

This commit drops [`cpy`](https://www.npmjs.com/package/cpy).
Ref: https://github.com/sindresorhus/cpy/commit/47b89a70b51af32133750ea2c73bed65fbe0ce56

Closes #2077
commit 601ef545 (2023-11-24): fix: sketch Save As preserves the folder structure

This commit rewrites how IDE copies sketches as part of the _Save As_
operation. Instead of copying to the destination, IDE copies the sketch
into a temporary location, then to the desired destination.

This commit drops [`cpy`](https://www.npmjs.com/package/cpy).
Ref: https://github.com/sindresorhus/cpy/commit/47b89a70b51af32133750ea2c73bed65fbe0ce56

Closes #2077
commit 074f6544 (2024-01-15): fix: sketch Save As preserves the folder structure

This commit rewrites how IDE copies sketches as part of the _Save As_
operation. Instead of copying to the destination, IDE copies the sketch
into a temporary location, then to the desired destination.

This commit drops [`cpy`](https://www.npmjs.com/package/cpy).
Ref: https://github.com/sindresorhus/cpy/commit/47b89a70b51af32133750ea2c73bed65fbe0ce56

Closes #2077
closed 2024-01-15 by kittaakos