arduino/arduino-ide #2079: Sketch not saving while Offline

fwlrichard, 2023-05-29:
### Describe the problem

Sketch not saving while dis-connected to the internet. If I'm online Sketch saves just fine.  
[image]

### To reproduce

Disconnect form the internet. Open a new sketch and try to save File|Save. 
You get Error: Could not save the sketch. Please copy your unsaved work into your favorite text editor, and restart the IDE.
Trun WIFI back on or reconnect to the internet and clicked File|Save and the Sketch saves.
 

### Expected behavior

I would expect the Sketch to save locally regardless of being online.

### Arduino IDE version

Version: 2.1.0 Date: 2023-04-19T15:31:10.185Z CLI Version: 0.32.2  Copyright © 2023 Arduino SA

### Operating system

Windows

### Operating system version

Windows 11 Home Version	10.0.22621 Build 22621

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-05-30:
There is another regression. When the IDE2 backend process crashes, IDE2 is used to warn users to copy the sketch content and save it with their favorite text editor. It does not work anymore. It's been broken since Theia started to use `socket.io`, and the `connectionStatusService` is no longer hitting [the expected error](https://github.com/eclipse-theia/theia/blob/c1a2b7b021cb67db4cc05a57cd3561bcba023ab9/packages/core/src/browser/connection-status-service.ts#L155-L156) as the connection tries to [reconnect forever](https://github.com/eclipse-theia/theia/blob/c1a2b7b021cb67db4cc05a57cd3561bcba023ab9/packages/core/src/browser/messaging/ws-connection-provider.ts#L123). Ref: cff2c956845e04d320231e8a924d1a47ad016af7

kittaakos, 2023-05-30:
@fwlrichard, thank you for taking the time and reporting the issue to us. I have created a PR with the proposed fixes: https://github.com/arduino/arduino-ide/pull/2080. If you are willing to help us with the verification, please follow the [beta-testing](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md) documentation.

PR #2080 (2023-06-08): fix: relaxed `saveAll` if no Internet connection
### Motivation

Saving any (local and cloud) sketches without an Internet connection must be possible. (Pulling/pushing cloud sketch changes are expected to not work without the Internet.)

### Change description

The previous logic has incorrectly bailed the save when there is no Internet connection. The corrected logic disallows saving files if there is no connection between the front and backend.

### Other information

How to verify the fix:
 - open IDE2, create a new sketch, turn auto-save off, disconnect from the Internet (you will see the [... cut]
commit 4af488b0 (2023-06-08): fix: relaxed `saveAll` if no Internet connection

The previous logic has incorrectly bailed the save when there is no
Internet connection. The corrected logic disallows saving files if there
is no connection between the frontend and the backend.

Ref: cff2c956845e04d320231e8a924d1a47ad016af7
Closes #2079
closed 2023-06-08 by kittaakos