arduino/arduino-ide #2085: Compiler Never terminates if compilation error occurs

marcusobrien, 2023-06-01:
### Describe the problem

Have a ESP32 sketch loaded. Hit the compile Button (tick mark top left on IDE GUI), the system compiles the sketch - but encounters an error, reports the error through the output terminal, and compilation has stopped - but GUI has not registered this state change, so the GUI has

1. The Notification message "Compiling", which stays on the screen and wont clear unless I explicitly clear it via Notification button
2. All re compile buttons greyed out in the menus eg Sketch->Verify/Compile, Sketch->Upload, Sketch->Configure and Upload, Sketch->Upload using programmer, and Sketch->Export compiled binary
3. I can still edit the sketch file and press other buttons - so the App hasn't crashed, and the GUI is still repsonding
4. The round Upload Arrow button is now yellow and cant be clicked, the round Verify button is still green and also cant be clicked

### To reproduce

1. Open .ino Sketch 
2. Press Round Right Arrow button on GUI to Compile and Upload
3. Make sure the compilation fails
4. Should now be in the same state as me, cant now re-compile as the GUI still thinks the compilation process is running.

Only was to now recover is reset application ie close it and re-open

### Expected behavior

Should be able to re-compile when compilations has stopped, without having to shut down and restart the IDE

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

10

### Additional context

Compiling project for ESP32S3 using libraries GFX and LVGL

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-01:
Hi @marcusobrien. Thanks for taking the time to submit an issue.

This is caused by a bug in the Arduino CLI tool that handles the compilation for Arduino IDE: https://github.com/arduino/arduino-cli/issues/2169

That bug has already been fixed in Arduino CLI, so the only work that remains is to update the version of Arduino CLI used by Arduino IDE, which is already in progress: https://github.com/arduino/arduino-ide/pull/2083

If you would like, you can test that fix by downloading the tester build for https://github.com/arduino/arduino-ide/pull/2083. There are instructions for doing that here:

https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md#testing-pull-requests

marcusobrien, 2023-06-03:
Hi @per1234 - thanks for the info - I followed the instructions - but this doesnt fix my bug of control returning back to the Application 

I used this version (taken from the About Menu)

Version: 2.1.1-snapshot-f6453a1
Date: 2023-06-01T07:03:41.627Z
CLI Version: 0.33.0

Copyright © 2023 Arduino SA

per1234, 2023-06-05:
Thanks so much for testing the potential fix and reporting your results. It is unfortunate the fault still occurs for you.

Does the problem of the IDE not returning to a usable state after compilation error occur for any sketch compiled for any board, or is it specific to the board and project you mentioned?:

>Compiling project for ESP32S3 using libraries GFX and LVGL

For example, what happens if you select **Tools > Board > Arduino AVR Boards > Arduino UNO** from the Arduino IDE menus and then compile this sketch?:

```cpp
void setup() {
  oops
}
void loop() {}
```

marcusobrien, 2023-06-10:
Hi, 

Using the simple project with the Uno as the target (but no real target is connected - so executing compile not upload from the GUI) , causes the IDE to fails to compile and then the compilation phase terminates excatly as expected (so the fauilt is not replicated in that simple project/sketch case).

I noticed on the instance of the IDE that is failing - the LVGL and ESP32S3, this session is complaining about the language server when I open the sketch. Then the compilation phase starts working ie it starts to build the project, then the link phase fails (due to undefined reference), then the toolchain never terminates (linker I presume ?).

If there is no board connected then the compialtion phase does terminate on my project also ie with ESP32 S3 and LVGL - so the problem is easier to reproduce when their is a target connected, so maybe its that ?

per1234, 2023-06-10:
If you compile that simple sketch from my previous comment for the ESP32-S3 board, does the problem of the IDE not returning to a usable state after a compilation error occur?

>this session is complaining about the language server when I open the sketch.

Please provide the full and exact text of any error or warning messages you are getting about the language server.

>If there is no board connected then the compialtion phase does terminate on my project also ie with ESP32 S3 and LVGL - so the problem is easier to reproduce when their is a target connected

Are you clicking the tick mark button ("Verify") or the right arrow button ("Upload")? I was under the impression the fault was occurring when clicking the "Verify" button because you mentioned it in the introduction of your original report:

>Hit the compile Button (tick mark top left on IDE GUI), the system compiles the sketch - but encounters an error, reports the error through the output terminal, and compilation has stopped - but GUI has not registered this state change

but I now see you mention using the "Upload" button later in the report:

>2. Press Round Right Arrow button on GUI to Compile and Upload

We do have a previous report related to this type of fault occurring during an upload:

https://github.com/arduino/arduino-ide/issues/1801

As well as some related reports:

- https://github.com/arduino/arduino-ide/issues/1357
- https://github.com/arduino/arduino-ide/issues/1356

But those are a bit different in that the fault is occurring after a successful compilation phase.

I was actually hoping that https://github.com/arduino/arduino-ide/pull/2083 would fix those issues as well (I can't reproduce https://github.com/arduino/arduino-ide/issues/1801 et al. so I can't check it myself).

per1234, 2023-09-28:
Hi @marcusobrien. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this. Please comment here to provide the requested information.

vnp85, 2024-03-29:
Hi everyone, this issue has been plaguing my workflow for quite some time: on compilation error, the IDE becomes unusable and needs to be closed from task manager. I just installed the latest Arduino IDE 2.3.2, with ESP32 2.0.14. On compilation error, press close window, save [yes/no] doesn't matter, and the window doesn't even close.