arduino/arduino-ide #2086: Arduino IDE2.1.0 editor issue

David2153, 2023-06-04:
### Describe the problem

There seems to be a problem with the new IDE 2.1.0 editor when I try to copy a section of code from one sketch to another sketch using the standard right click COPY and PASTE. The copy and paste seems to work however I then use SAVEAS to save the new sketch into a new sketch/file and the file that is created is unexpectedly truncated (ie. missing lines of code at the bottom of the sketch). I have used this technique successfully on many occasions with the older IDE 1.8.19 without this issue.

### To reproduce

1. open an existing sketch
2. open a second existing sketch
3. highlight and COPY a section of code in the 1st sketch
4. PASTE this section of code to the 2nd sketch
5. SAVEAS the 2nd sketch to a new file name/location
6. the saved file is unexpectedly truncated

### Expected behavior

I'd expect the new sketch not to be truncated.

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

WIN10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2023-06-05:
@David2153 
I cannot reproduce this on Mac, I'll check with the rest of the team :)

David2153, 2023-06-06:
Further information.... the original compiled .ino file size is 28.4 kB for the two opened sketches and the saved to disk  .ino file size is 101kB.
The truncated saved  .ino file  size is 65kB and it will not compile. 
The sketches are intended for a UNO board.

David2153, 2023-06-07:
Further information...... the problem seems related to file size. I tried the same process using smaller sketches and it worked as it should. I tried it again using the larger sketches where I initially observed the problem and it failed again. I have expanded/clarified the "To Reproduce" procedure below.

1. open an existing sketch
2. open a second existing sketch
3. using a mouse, highlight and right click COPY a section of code in the 1st sketch
4. using a mouse, highlight the section of code in the second sketch to be replaced with the clipboard contents
5. right click PASTE the clipboard contents to the highlighted code in the second sketch
6. the COPY and PASTE appears to work OK with no lost lines of code at the bottom of the sketch
7. SAVE AS the 2nd sketch to a new file name/location
8. the saved file is unexpectedly truncated

kittaakos, 2023-06-07:
Thank you for taking the time and writing the bug report.

I could reproduce the bug with `2.1.0` when the file size exceeds 64kb. Upstream issue: https://github.com/eclipse-theia/theia/issues/12406.

We have fixed this by updating the version of the underlying [Theia IDE framework](https://theia-ide.org/) in IDE2. See https://github.com/arduino/arduino-ide/pull/2027.

Please try with the nightly build and let us know if the defect is still present. You can download the nightly IDE2 builds from https://www.arduino.cc/en/software#nightly-builds.

I also downloaded IDE2 nightly (`2.1.1-nightly-20230607`), which worked for me locally. `2.1.0` does not. I appreciate your help.

David2153, 2023-06-08:
The latest nightly version has fixed the issue. Thanks.

per1234, 2023-06-08:
Thanks for the follow-up @David2153.

Closing as resolved by https://github.com/arduino/arduino-ide/pull/2027.

closed 2023-06-08 by per1234