arduino/arduino-ide #2087: Outline view says "The active editor cannot provide outline information."

Mek7, 2023-06-04:
### Describe the problem

Outline view says "The active editor cannot provide outline information."
It does not display any outline.

### To reproduce

Open any sketch. For example:
```
/*

 * EEPROM Clear

 *

 * Sets all of the bytes of the EEPROM to 0.

 * Please see eeprom_iteration for a more in depth

 * look at how to traverse the EEPROM.

 *

 * This example code is in the public domain.

 */

#include <EEPROM.h>

void setup() {

  // initialize the LED pin as an output.

  pinMode(13, OUTPUT);

  

  /***

    Iterate through each byte of the EEPROM storage.

    Larger AVR processors have larger EEPROM sizes, E.g:

    - Arduino Duemilanove: 512 B EEPROM storage.

    - Arduino Uno:         1 kB EEPROM storage.

    - Arduino Mega:        4 kB EEPROM storage.

    Rather than hard-coding the length, you should use the pre-provided length function.

    This will make your code portable to all AVR processors.

  ***/

  for (int i = 0 ; i < EEPROM.length() ; i++) {

    EEPROM.write(i, 0);

  }

  // turn the LED on when we're done

  digitalWrite(13, HIGH);

}

void loop() {

  /** Empty loop. **/

}
```
Then press F1, write "Toggle outline view" and select it from the dropdown to open outline view. Observe it displays "The active editor cannot provide outline information."

### Expected behavior

Show outline of the sketch - function list. Such as in the video in #237 

### Arduino IDE version

Version: 2.1.1-nightly-20230604 Date: 2023-06-04T03:16:46.254Z CLI Version: 0.32.2  Copyright © 2023 Arduino SA

### Operating system

Windows

### Operating system version

Win 10 Pro x64

### Additional context

The sketch is for Arduino Nano with old bootloader.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

francwalter, 2023-06-05:
Yes, I noticed this too, since maybe v2.1 the outline view is missing. It worked on 2.0 something but now it doesnt show anymore 😢 
Outline View is important and useful. I have a long sketch and scrolling and searching is annoying.
I use 2.1.0 on Windows 10 (64-Bit).

francwalter, 2023-06-13:
Is there at least any trick to bring it back?
I remember on the early betas, there was always issues with that outline information (see 124), but often it was present.
There must be something to do about it?
At worst a downgrade to 2.0.x?

francwalter, 2023-06-29:
OK, no fix since yet, I downgraded to [2.0.4](https://github.com/arduino/arduino-ide/releases/tag/2.0.4), where the outline is still working.
Hope this could be fixed in 2.1+ once 😄 
The 2.1 I have in my C:\Program Files folder (I had installed it with Administrator rights) and the 2.0.4 I have in the local programs folder (c:\Users\f\AppData\Local\Programs\Arduino-IDE\) so I can keep both.
They use the same settings (in: c:\Users\f\.arduinoIDE\settings.json) and recent sketch (in: c:\Users\f\.arduinoIDE\recent-sketches.json) so no real difference (apart from the working oulines).

When the 2.1 is fixed, I will switch back.
Thanks

Mek7, 2023-09-05:
No, it's broken and considering it's a critical functionality, I'm suprised it's getting no attention.
I am still waiting for a version that fixes this, and am not aware of any workaround :(

francwalter, 2023-09-05:
Is the feature surplus in 2.1+ worth that missing outline? 
For me that outline is so useful, that I wait with any update till the outline is working again. On 2.0.4 it works (as I wrote).

per1234, 2023-09-17:
Thanks for your report @Mek7. Please add a comment here to tell us which board you have selected from the **Tools > Board** menu in Arduino IDE when you experience this issue.

---

For all other interested parties: Please refrain from replying unless you have significant and unique information to add. If you only want to express your support for the PR or a comment in the thread, use the ["Reactions" feature](https://github.blog/2016-03-10-add-reactions-to-pull-requests-issues-and-comments/) to add a "👍", etc.

Everyone who watches the repository receives a notification whenever any reply is made on any of the issues and PRs. It is important for the developers and maintainers of the project to watch the repository to make sure we see the valuable feedback from the contributors. But if the repository becomes the source of a large number of low value notifications we must either waste our time sorting through them instead of doing productive work, or else unwatch the repository.

Unstructured conversations about Arduino-related subjects are always welcome over on [Arduino Forum](https://forum.arduino.cc/).

Mek7, 2023-09-17:
Hello, the board is Arduino Nano.
I launched Arduino IDE to check it out today, it asked me to update to version 2.2.1 and now I see the outline view is working already, so it looks like it was fixed in the meantime.

closed 2023-09-17 by per1234