arduino/arduino-ide #2091: On some ESP32 boards, the default Arduino upload speed is too high, resulting in errors

fdb, 2023-06-07:
### Describe the problem

When trying to upload to a NodeMCU ESP32-S on Mac M1, the default upload speed of 921,600 is too high resulting in errors like this:

```
A fatal error occurred: Unable to verify flash chip connection (Invalid head of packet (0xE0): Possible serial noise or corruption.).
```
Or this:
```
A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 7 bytes)).
```
This only affects some boards. The ESP-WROOM-32 by AZ-Delivery has no issues. The two NodeMCU ESP32-S's do have issues.

Here is the full output:

```
Sketch uses 235937 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22056 bytes (6%) of dynamic memory, leaving 305624 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-4140
Connecting......
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:e5:7c:db:1c:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 8 bytes)).
Failed uploading: uploading error: exit status 2
````

Setting the upload rate to a lower value (e.g. 460800) fixes the error.

### To reproduce

- Open the Basics>Blink Example
- Select the NodeMCU32-S board and Serial Port (USB)
- Choose Upload

### Expected behavior

Uploading proceeds, and upload percentages rise. This works if I lower the baud rate to 460800:

```
Sketch uses 235937 bytes (18%) of program storage space. Maximum is 1310720 bytes.
Global variables use 22056 bytes (6%) of dynamic memory, leaving 305624 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-4140
Connecting.....
Chip is ESP32-D0WD (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: a4:e5:7c:db:1c:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00049fff...
Compressed 17520 bytes to 12170...
Writing at 0x00001000... (100 %)
Wrote 17520 bytes (12170 compressed) at 0x00001000 in 0.7 seconds (effective 213.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 267.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 400.8 kbit/s)...
Hash of data verified.
Compressed 236304 bytes to 129987...
Writing at 0x00010000... (12 %)
Writing at 0x0001e222... (25 %)
Writing at 0x0002421b... (37 %)
Writing at 0x000293fa... (50 %)
Writing at 0x0002ea8c... (62 %)
Writing at 0x00036fdb... (75 %)
Writing at 0x0003f0ee... (87 %)
Writing at 0x000446ea... (100 %)
Wrote 236304 bytes (129987 compressed) at 0x00010000 in 3.6 seconds (effective 527.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

### Arduino IDE version

Version 2.1.0

### Operating system

macOS

### Operating system version

Ventura 13.1 on Apple M1 (both Mac Studio and MacBook Air)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-14:
Hi @fdb. The default for custom board options such as the **Tools > Upload Speed** menu that appears when an ESP32 board is selected are set by the order of the option definitions in the `boards.txt` configuration file of the boards platform.

For example:

https://github.com/espressif/arduino-esp32/blob/72c41d09538663ebef80d29eb986cd5bc3395c2d/boards.txt#L219

So this is something the ESP32 boards platform developers have configured. For this reason, I'll close it as off topic. You can take it up in the ESP32 boards platform's repository if you like.

closed 2023-06-14 by per1234

fdb, 2023-06-15:
Thanks, filed an issue at arduino-esp32: https://github.com/espressif/arduino-esp32/issues/8320

WidgetVRP, 2023-11-03:
It worked, you guys are the best