arduino/arduino-ide #2093: Arduino IDE 2.1.0 Serial Monitor

IvanchoIotata, 2023-06-09:
### Describe the problem

Cannot copy the entire output on displayed on the Serial monitor. Only the visible part can be copied. Additionally when two colon data is copied it looks like it is copied in some sort formatted text instead of pure ASCII like the legacy Arduino 1.8.19. Therefore when the two colon data is copied from the serial monitor and pasted in Excel for example it is pasted as text in one colon. In the legacy Arduino this would paste as two colon data. The Arduino IDE 2.x.x displays data on the serial monitor in some sort of HTML format while keeping the invisible data in a buffer. Scrolling is pulling the data out of a buffer instead of scrolling a pure text window. 

### To reproduce

Run two colon of numbers separated by comma or space. Try to copy the whole list of data including the invisible part. Paste the copied data to Excel. 

### Expected behavior

Being able to copy the entire data on the Serial monitor, not only the one that is displayed on the visible part. 
Expecting to behave exactly like the legacy Serial Monitor. Or any standard Terminal. 

### Arduino IDE version

Any version 2

### Operating system

Windows, Linux, macOS

### Operating system version

This is universal behavior

### Additional context

People have been complaining about this since Arduino IDE 2 was in alpha testing mode. When this will be addressed?

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-06-09:
Thanks for taking the time and reporting it to us, @IvanchoIotata.

You're describing at least two issues here:

> Cannot copy the entire output on displayed on the Serial monitor. Only the visible part can be copied.

> Additionally when two colon data is copied it looks like it is copied in some sort formatted text instead of pure ASCII like the legacy Arduino 1.8.19.

The following issues track all the monitor widget-related bugs and enhancements:
 - https://github.com/arduino/arduino-ide/issues/247,
 - https://github.com/arduino/arduino-ide/issues/812, and
 - https://github.com/arduino/arduino-ide/issues/1729.

In the context of this GitHub issue, I focus on the copy-pasting of the colon data.

> Run two colon of numbers separated by comma or space. Try to copy the whole list of data including the invisible part. Paste the copied data to Excel.

Please share a bare minimum sketch to produce the monitor output you mentioned, and I will check the problem. Thank you!

IvanchoIotata, 2023-06-09:
``Hello,
Thank you for the response.
Actually the main issue is selecting and copying the entire output on Serial monitor if it the output does not fit on visible part on the Serial Monitor. The two colon issue have workaround, that adds extra steps and is annoying, but works. Legacy Arduino does not have that problem
Here is a simple code that illustrates both problems.
`
/*************************
 * Test copy a two colon tab separated numbers
 * from serial monitor to clipboard and paste in Excel
 * On Arduino 1.8.19 works fine
 * On Arduino IDE 2.X.X the copied data pasted in only one colon
 * Also the entire output cannot be copied to the clipboard if
 * all the data is not visible on the display. Ctrl+A does not work on 2.X.X either.
 *************************/
void setup() {
  Serial.begin(115200);
  delay(500);
  for(int i = 0; i < 51; i++){
    Serial.print( i );
    Serial.print("\t");
    Serial.println( i * i );
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}`

Thanks

IvanchoIotata, 2023-06-09:
Here is the code in zip file:
[SerialMonitoryCopyPasteTest.zip](https://github.com/arduino/arduino-ide/files/11707109/SerialMonitoryCopyPasteTest.zip)

IvanchoIotata, 2023-06-09:
I have one more suggestion regarding Serial Monitor. To be placed on the side of the Arduino IDE window instead of the bottom. With modern day display there always more depth(height) than width. Also the data in the Serial monitor is rarely 10 spaces wide, but can be quite long. It makes sense to place the Serial monitor in a place that more data can be visible.

Nick-Norton, 2023-06-10:
> I have one more suggestion regarding Serial Monitor. To be placed on the side of the Arduino IDE window instead of the bottom. 
Or have the ability to pop the Serial Monitor out, into its own Window. Like the Serial Plotter does.

IvanchoIotata, 2023-06-12:
@per1234 Thank you for summarizing the problem in more concise manner.

IvanchoIotata, 2023-07-18:
I want to check if there is any progress on this issue.

avandalen, 2024-09-14:
Can this issue be put on a priority list, it is important!

per1234, 2026-11-13:
Closing as resolved by https://github.com/arduino/arduino-ide/pull/2936.

closed 2026-11-13 by per1234