arduino/arduino-ide #2096: Quick suggestion does not work on MacOs Ventura.

ubaldot, 2023-06-12:
### Describe the problem

I followed the [guideline](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature) but when I hit CTRL+Space nothing happens. It also looks like that it is not possible to generate any LSP log (or perhaps it is just me that I haven't found). 

### To reproduce

Install Arduino IDE

```
Version: 2.1.0
Date: 2023-04-17T11:52:02.744Z
CLI Version: 0.32.2
```

on `MacOs 13.4 (22F66)`, attach a ArduinoUno and test the autocomplete.

### Expected behavior

Autocomplete suggestions to pop up.

### Arduino IDE version

2.1.0

### Operating system

macOS

### Operating system version

13.4 (22F66)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-12:
>### Arduino IDE version
>2.1.0

There was a bug in Arduino IDE 2.1.0 that could cause the language server features to not be available under certain conditions (https://github.com/arduino/arduino-ide/issues/2034). That bug has already been fixed (https://github.com/arduino/arduino-ide/pull/2083) since the time of the 2.1.0 release you are using. Please try with the latest nightly build of Arduino IDE, which contains the fix, then comment here to let us know whether the problem still occurs with the nightly build.

The download links for the nightly build are listed here:

https://www.arduino.cc/en/software#nightly-builds

> It also looks like that it is not possible to generate any LSP log

I'll provide instructions:

## A. Enable Language Server Logging

1. Use the <kbd>Ctrl</kbd>/<kbd>⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. In the "**Search Settings**" field, type `arduino.language.log`
1. Under the "**Arduino > Language: Log**", section, check the box next to "**☐ True if the Arduino Language Server should generate log files into the sketch folder. Otherwise, false. It's false by default.**"
1. Close the **Preferences** tab by clicking its **X** icon.

## B. Gather Logs

1. Start Arduino IDE.
1. Perform whatever procedure causes the problem you are experiencing in Arduino IDE.
1. Select **Sketch > Show Sketch Folder** from the Arduino IDE menus.
   The folder containing the sketch will open in your file manager.

You will find several `.log` files in the sketch folder. These are the language server logs.

## C. Disable Language Server Logging

When language server logging is enabled, log files are saved to every sketch you open in Arduino IDE and they steadily grow in size whenever you are editing the sketch. So the logging is something you would usually want to disable whenever generation of logs is not needed.

1. Use the <kbd>Ctrl</kbd>/<kbd>⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. In the "**Search Settings**" field, type `arduino.language.log`
1. Under the "**Arduino > Language: Log**", section, uncheck the box next to "**`☑` True if the Arduino Language Server should generate log files into the sketch folder. Otherwise, false. It's false by default.**"
1. Close the **Preferences** tab by clicking its **X** icon.
1. Select **File > Quit** from the Arduino IDE menus.

You are now welcome to delete the `.log` files that were added to the sketch folder.

closed 2023-06-12 by ubaldot

kilitary, 2025-04-28:
not works in latest 2.3.7