arduino/arduino-ide #2098: Only the visible data can be copied from the serial Monitor

IvanchoIotata, 2023-06-12:
### Describe the problem

When the data is long enough it cannot be copied from the Serial Monitor. 
Arduino 1.8.X does it. Also Ctrl + A works to select all the data on the Serial Monitor and then Ctrl+C will copy it to the clipboard.
Arduino IDE 2.X.X can select and then copy only the data that is visible. 

### To reproduce

This code will generate enough lines to scroll out of the visible window.  Try to select the whole data including the portion that is scrolled out.
`/*************************
 * Test copy all the data on Serial Monitor
 * On Arduino 1.8.19 works fine
 * On Arduino IDE 2.X.X the can select and copy only the visible data
 * The entire output cannot be copied to the clipboard if
 * all the data is not visible on the display. Ctrl+A does not work on 2.X.X either.
 *************************/
void setup() {
  Serial.begin(115200);
  delay(500);
  for(int i = 0; i < 101; i++){
    Serial.print( i );
    Serial.print("\t");
    Serial.println( i + i );
  }

}

void loop() {
  // put your main code here, to run repeatedly:

}`

### Expected behavior

To be able to select and copy more than the visible data from the Serial monitor by dragging your cursor through the area or at least use Ctrl+A to select all the data displayed on the Serial monitor. That has been working on Arduino 1.X.X with no problem

### Arduino IDE version

Arduino IDE 2.1.0

### Operating system

Windows, Linux, macOS

### Operating system version

This is universal behavior

### Additional context

This issue has been persistent since the alpha testing of Arduino IDE 2. 
There are several posts that discuss it but I have not seen detailed explanation with a code that reproduce it.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-13:
Hi @IvanchoIotata. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1081 or https://github.com/arduino/arduino-ide/issues/812.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2023-06-13 by per1234

IvanchoIotata, 2023-06-13:
Hello Per1234. 
Thank you for responding. I opened that issue because the other report was split in two issues and the issue that is described here was not addressed. It is not even clear if every this issue will be addressed within the modern Arduino system. I see a lot of dancing around across the forums here, with suggestions to use alternative serial terminals etc. That is creating a pseudo solution and the core issue is not addressed.

gaby64, 2024-05-25:
The issue remains, we would like a fix please.