arduino/arduino-ide #2100: Serial monitor 'No monitor available for port protocal serial`

m410, 2023-06-15:
### Describe the problem

I'm not able to use the serial monitor with arduino-pico. I've only tried using 'raspberry pi pico w'. Scripts build and flash the device as expected but the serial monitor never attaches.  

Error from commandline:
`
2023-06-15T13:58:16.718Z root ERROR Uncaught Exception: 
2023-06-15T13:58:16.718Z root ERROR Error: 3 INVALID_ARGUMENT: No monitor available for the port protocol serial
    at Object.callErrorFromStatus (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/call.js:31:19)
    at Object.onReceiveStatus (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/client.js:190:52)
    at Object.onReceiveStatus (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:365:141)
    at Object.onReceiveStatus (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/client-interceptors.js:328:181)
    at /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/call-stream.js:187:78
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
for call at
    at ServiceClientImpl.makeUnaryRequest (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/client.js:160:30)
    at ServiceClientImpl.enumerateMonitorPortSettings (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/@grpc/grpc-js/build/src/make-client.js:105:19)
    at resp (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/monitor-service.js:346:20)
    at new Promise (<anonymous>)
    at MonitorService.portMonitorSettings (/Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/lib/node/monitor-service.js:345:28)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
`

### To reproduce

Follow these install instructions: https://github.com/earlephilhower/arduino-pico#installation
I've tried library versions 3.2.2 & 3.2.1.

I used this basic blick script, but others show the same behavior.  This same script works for me on IDE 1.8.19.

`#include "Adafruit_TinyUSB.h"
#include "hardware/pio.h"

void setup() {
  Serial.begin(115200);
  
  while (!Serial);
  delay(100); 

  Serial.println("Startup");
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  Serial.println("Looping...");
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(1000);   
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000); 
}
`

### Expected behavior

Serial monitor should work.

### Arduino IDE version

2.1 & 2.1.1-nightly

### Operating system

macOS

### Operating system version

Ventura 13.4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-06-16:
>  No monitor available for the port protocol serial

Thanks for reporting it to us. Interesting one. Please check what IDE2 logs when you plug/unplug the 'raspberry pi pico w' board.

You can start IDE2 by adjusting the log level to `debug`. On my macOS, it looks like this:

```sh
/Applications/Arduino\ IDE\ 2.1.0.app/Contents/MacOS/Arduino\ IDE --log-level debug
```

When IDE2 has started, please clean the terminal, and plug/unplug the board. IDE2 will log things. Please share everything, but I am looking for such events in particular. These are the events when I plugin in the board:

```
2023-06-16T13:15:40.306Z discovery-log INFO {"eventType":"add","port":{"matchingBoardsList":[{"name":"Arduino Nano RP2040 Connect","fqbn":"arduino:mbed_nano:nanorp2040connect","isHidden":false}],"port":{"address":"/dev/cu.usbmodem14301","label":"/dev/cu.usbmodem14301","protocol":"serial","protocolLabel":"Serial Port (USB)","propertiesMap":[["pid","0x005E"],["serialNumber","FFFFFFFFFFFFFFFF"],["vid","0x2341"]],"hardwareId":"FFFFFFFFFFFFFFFF"}},"error":""}
```

Or the `Board config changed:  ` is also very interesting in this case:

```
2023-06-16T13:15:40.317Z root DEBUG Board config changed:  {"selectedBoard":{"name":"Arduino Nano RP2040 Connect","fqbn":"arduino:mbed_nano:nanorp2040connect","port":{"address":"/dev/cu.usbmodem14301","addressLabel":"/dev/cu.usbmodem14301","protocol":"serial","protocolLabel":"Serial Port (USB)","properties":{"pid":"0x005E","serialNumber":"FFFFFFFFFFFFFFFF","vid":"0x2341"},"hardwareId":"FFFFFFFFFFFFFFFF"}},"selectedPort":{"address":"/dev/cu.usbmodem14301","addressLabel":"/dev/cu.usbmodem14301","protocol":"serial","protocolLabel":"Serial Port (USB)","properties":{"pid":"0x005E","serialNumber":"FFFFFFFFFFFFFFFF","vid":"0x2341"},"hardwareId":"FFFFFFFFFFFFFFFF"}}
```

Thank you for helping!

(Normally, I would have asked you to attach the log files, but messages with [`debug` level are missing](https://github.com/arduino/arduino-ide/blob/d79bc0d083c40b3244264696d2d6703756a8bd53/electron/build/patch/backend/main.js#L21) from the log 🤦)

m410, 2023-06-17:
Hi! I’m away from my computer, I’ll rerunnwirh debug on by Monday morning EST

m410, 2023-06-19:
Here's the output
[arduino-ide-debug.txt](https://github.com/arduino/arduino-ide/files/11788880/arduino-ide-debug.txt)

kittaakos, 2023-06-21:
Thanks for the logs, @m410.

I have received some internal assistance, and the problem seems to be in the [`platform.txt`](https://github.com/earlephilhower/arduino-pico/blob/21d1a285dcca7fcd0e1dd4f6cf89967a4e44aa51/platform.txt) of the core. All discoveries and monitors must be specified in the `platform.txt`. Something like this:

```properties
# Required discoveries and monitors
# ---------------------------------
pluggable_discovery.required.0=builtin:serial-discovery
pluggable_discovery.required.1=builtin:mdns-discovery
pluggable_monitor.required.serial=builtin:serial-monitor
```

Here are two PRs as an example of how people ported a legacy platform to the newer [pluggable discovery](https://arduino.github.io/arduino-cli/dev/pluggable-discovery-specification/):
 - https://github.com/arduino/ArduinoCore-avr/pull/426
 - https://github.com/arduino/ArduinoCore-avr/pull/440

If you wonder why the same sketch works in IDE 1.x but not in 2.x, well, I had the same question, and maintainers of the core logic explained to me that this is expected.

Since the necessary changes have to be performed in the core, I decided to close this IDE2 issue.

You know where to make the changes if you are the core maintainer. If you are a user of the core, please request a change on their GitHub page, or ask around on the Arduino forum: https://forum.arduino.cc/c/software/ide-2-x/93.

closed 2023-06-21 by kittaakos