arduino/arduino-ide #2103: Upload doesnt just upload

marcusobrien, 2023-06-17:
### Describe the problem

There are 6 sketch build and upload buttons in the Arduino 2 IDE. 

Toolbar **Verify** Button - Builds the sketch
Toolbar **Upload** Button - Builds the sketch, then uploads to device

Menu Bar Sketch Menu - **Verify/Compile** MenuButton - Builds the sketch
Menu Bar Sketch Menu - **Upload** MenuButton - Builds the sketch and then uploads to device
Menu Bar Sketch Menu - **Configure and Upload** MenuButton - ??
Menu Bar Sketch Menu - **Upload Using Programmer** MenuButton - Builds the sketch and then uploads to device

1. Why does Upload build/verify before uploading ? Can't it just upload the existing binary to the device if there is one already ?
2. Whats the difference between the Upload Button, Upload MenuButton, Upload using Programmer and COnfigure and Upload ?
3. How can I upload just an existing binary to the device, without any compilation/build/link/config etc, My workflow is sometimes develop, test - compile and upload, experiment with the device causing firmware wipe, then I want to only re-apply/upload the same firmware that I built earlier, I dont weant to rebuilod anything.

### To reproduce

1. Create a simple project and sketch .ino
2. Configure IDE for ESP32 for example (can be any device AFAIK)
3. Press Upload (Right Arrow) on Toolbar - IDE will build the sketch first and then upload firmware binary
4. Press Upload (Right Arrow) on Toolbar - IDE will build the sketch AGAIN (even without any code changes being made), and then it will upload the newly built firmware binary 

### Expected behavior

1. Create a simple project and sketch .ino
2. Configure IDE for ESP32 for example (can be any device AFAIK)
3. Press Upload (Right Arrow) on Toolbar - IDE will build the sketch first and then upload firmware binary
4. Press Upload (Right Arrow) on Toolbar - IDE will **NOT** build the sketch AGAIN - it will just upload the **existing** firmware binary 

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-18:
Hi @marcusobrien. Thanks for taking the time to submit an issue.

> Why does Upload build/verify before uploading ?

Because Arduino IDE is designed to be friendly to new users. Having a requirement to manually trigger a separate "Verify" operation after any change to the sketch and sketch dependencies before doing an "Upload" operation would make it significantly less friendly to new users, many of whom would end up being confused after forgetting to do that and uploading a binary that doesn't match their sketch code.

In return, separating the two operations would only benefit users in the relatively rare use case of using Arduino IDE to deploy a sketch to multiple boards of the same model.

> Can't it just upload the existing binary to the device if there is one already ?

That binary might be outdated. The build phase of the upload ensures that the binary always matches the current sketch code and its dependencies.

The build system caches the compilation output, so the full sketch program is only compiled the first time and when one of the cached objects has been made outdated by a change to the source. There is work in progress to make these subsequent builds more efficient:

- https://github.com/arduino/arduino-cli/issues/1996
- https://github.com/arduino/arduino-cli/issues/2204
- https://github.com/arduino/arduino-cli/pull/1735
- https://github.com/arduino/arduino-cli/pull/1766

>Whats the difference between the Upload Button, Upload MenuButton, Upload using Programmer and COnfigure and Upload ?

Discussion of these subjects is off topic for this issue tracker. I'll be happy to answer your questions if you make a post on Arduino Forum:

https://forum.arduino.cc/c/software/ide-2-x/93

> How can I upload just an existing binary to the device, without any compilation/build/link/config etc

You can use [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/). Unlike Arduino IDE, since **Arduino CLI** is a tool for advanced users, the `arduino-cli upload` command does not do a compile, instead requiring that the used has previously generated the binary via an `arduino-cli compile` command.

Alternatively, you can use the upload tool of your board (e.g., [**esptool**](https://github.com/espressif/esptool)) directly to upload the binary.

Because their command line interface enables use in scripts and automated applications, these tools will be much more appropriate than Arduino IDE in the use case of deploying a binary to multiple of the same model board. We will not add such a capability to Arduino IDE (https://github.com/arduino/arduino-ide/issues/333).

If you would like assistance with using **Arduino CLI**, **esptool**, etc., you are welcome to post on [Arduino Forum](https://forum.arduino.cc/).

closed 2023-06-18 by per1234

marcusobrien, 2023-06-23:
Added question to Arduino Forumas requested, even though IMO its excessively pedantic not to just answer it here.