arduino/arduino-ide #2104: nioght build, no ports detected, windows 10, device manager and 1.8 see port ok

rfaass, 2023-06-18:
### Describe the problem

Do not know how to get my Arduino Mega 256, no clone, to show up. it was working before, upgraded to 2.1.0.

Restated, reloaded, downloaded the night build, changed usb ports, nothing helps. Lost my afternoon, trying to get this to work. 
1.8 recognizes it fine.

### To reproduce

Good question, I just have my Mega attached...

### Expected behavior

recognize the board... ahum

### Arduino IDE version

2.1.0 last nightly as well as the 2.1.0 from arduino cc

### Operating system

Windows

### Operating system version

win10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

rfaass, 2023-06-19:
No ports discovered; show all ports does not show any port....

per1234, 2023-06-19:
Hi @rfaass. Thanks for your report. I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Start [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1).
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<username>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

rfaass, 2023-06-19:
PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Arduino IDE 2.1.0
Starting backend process. PID: 4684
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///c%3A/Users/Robert%20Faass/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-06-19T00:52:20.759Z root INFO Backend Object.initialize: 5.8 ms [Finished 2.275 s after backend start]
2023-06-19T00:52:20.760Z root INFO Backend Object.configure: 4.7 ms [Finished 2.275 s after backend start]
2023-06-19T00:52:20.761Z root INFO Backend MessagingContribution.onStart: 2.1 ms [Finished 2.275 s after backend start]
2023-06-19T00:52:20.763Z root INFO Backend OSBackendApplicationContribution.configure: 0.6 ms [Finished 2.329 s after backend start]
2023-06-19T00:52:20.764Z root INFO Backend DefaultWorkspaceServer.onStart: 0.5 ms [Finished 2.329 s after backend start]
2023-06-19T00:52:20.766Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.2 ms [Finished 2.329 s after backend start]
2023-06-19T00:52:20.767Z root WARN Backend PluginLocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 53.9 ms [Finished 2.330 s after backend start]
2023-06-19T00:52:20.767Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.4 ms [Finished 2.330 s after backend start]
2023-06-19T00:52:20.768Z root INFO Backend ElectronTokenBackendContribution.configure: 3.0 ms [Finished 2.333 s after backend start]
2023-06-19T00:52:20.769Z config INFO >>> Initializing CLI configuration...
2023-06-19T00:52:20.769Z root INFO Backend PluginDeployerContribution.initialize: 0.9 ms [Finished 2.333 s after backend start]
2023-06-19T00:52:20.770Z root INFO Backend ConfigServiceImpl.onStart: 1.6 ms [Finished 2.335 s after backend start]
2023-06-19T00:52:20.771Z config INFO Loading CLI configuration from c:\Users\Robert Faass\.arduinoIDE\arduino-cli.yaml...
2023-06-19T00:52:20.772Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.5 ms [Finished 2.336 s after backend start]
2023-06-19T00:52:20.772Z root INFO Backend ArduinoDaemonImpl.onStart: 1.6 ms [Finished 2.337 s after backend start]
2023-06-19T00:52:20.772Z root INFO Backend NodeFileUploadService.configure: 3.1 ms [Finished 2.338 s after backend start]
2023-06-19T00:52:20.773Z daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
2023-06-19T00:52:20.773Z discovery-log INFO start
2023-06-19T00:52:20.773Z discovery-log INFO start new deferred
2023-06-19T00:52:20.773Z root INFO Backend HostedPluginLocalizationService.initialize: 0.6 ms [Finished 2.339 s after backend start]
2023-06-19T00:52:20.774Z root INFO Backend BoardDiscovery.onStart: 1.5 ms [Finished 2.340 s after backend start]
2023-06-19T00:52:20.774Z root INFO Backend FileDownloadEndpoint.configure: 1.3 ms [Finished 2.341 s after backend start]
2023-06-19T00:52:20.774Z root INFO Backend AuthenticationServiceImpl.onStart: 0.5 ms [Finished 2.343 s after backend start]
2023-06-19T00:52:20.774Z root INFO Backend PluginApiContribution.configure: 0.6 ms [Finished 2.343 s after backend start]
2023-06-19T00:52:20.776Z root INFO Backend HostedPluginReader.configure: 0.7 ms [Finished 2.344 s after backend start]
2023-06-19T00:52:20.776Z root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 2.345 s after backend start]
2023-06-19T00:52:20.776Z root INFO Backend PlotterBackendContribution.configure: 1.4 ms [Finished 2.347 s after backend start]
2023-06-19T00:52:20.783Z root INFO Theia app listening on http://localhost:62997.
2023-06-19T00:52:20.784Z root INFO Finished starting backend application: 10.2 ms [Finished 2.354 s after backend start]
2023-06-19T00:52:20.786Z root WARN The local plugin referenced by local-dir:/c%3A/Users/Robert%20Faass/.arduinoIDE/plugins does not exist.
2023-06-19T00:52:20.786Z root WARN The local plugin referenced by local-dir:/c%3A/Users/Robert%20Faass/.arduinoIDE/extensions does not exist.
2023-06-19T00:52:20.787Z root WARN The local plugin referenced by local-dir:C:\Users\Robert Faass\.arduinoIDE\plugins does not exist.
Restoring workspace roots: c:\Users\Robert Faass\Documents\Arduino\sketch_jun18a-wdt
isTempSketch: false. Input was c:\Users\Robert Faass\Documents\Arduino\sketch_jun18a-wdt\sketch_jun18a-wdt.ino
2023-06-19T00:52:20.821Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-19T00:52:20.822Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-06-19T00:52:20.823Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-19T00:52:20.824Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.8" with engines:
2023-06-19T00:52:20.826Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-19T00:52:20.828Z root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
2023-06-19T00:52:20.829Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-19T00:52:20.829Z root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
2023-06-19T00:52:20.830Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
INFO[0010] Required tool                                 tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7"
INFO[0010] Required tool                                 tool="arduino:avrdude@6.3.0-arduino17"

2023-06-19T00:52:31.836Z daemon INFO INFO[0010] Required tool                                 tool="arduino:arduinoOTA@1.3.0"
INFO[0010] Required tool                                 tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7"
INFO[0010] Required tool                                 tool="arduino:avrdude@6.3.0-arduino17"

rfaass, 2023-06-19:
Closed by accident

per1234, 2023-06-19:
This is the relevant part of the output:

> ```text
> 2023-06-19T00:52:24.020Z daemon INFO ERRO[0003] Discovery builtin:serial-discovery failed to run: discovery builtin:serial-discovery process not started: exec: "C:\\Users\\Robert Faass\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\serial-discovery\\1.4.0\\serial-discovery": file does not exist
> ```

Arduino IDE gets the list of ports of Arduino boards connected to your computer from helper tools called "pluggable discoveries".

Arduino IDE automatically installs these tools, but in this case the tool is not present at the path where it should be installed:

```text
C:\Users\Robert Faass\AppData\Local\Arduino15\packages\builtin\tools\serial-discovery\1.4.0\serial-discovery
```

The reason for the missing file is likely a false positive by your antivirus or security software. Please check your security software's quarantine. If you can restore the file from quarantine and then restart Arduino IDE, you should find that the ports are detected as expected.

I also recommend adjusting the configuration of your security software so that it won't interfere with any future updates of the tool.

If you are unable to restore the file from quarantine, just comment here to let me know and I'll provide some alternative instructions you can follow to force Arduino IDE to make a fresh installation of the tool.

---

Related: https://github.com/arduino/arduino-cli/issues/1669

closed 2023-06-19 by per1234

rfaass, 2023-06-19:
I am unable to restore the file from quarantine, as I did not find any file quarantined....  Please let me know how to force Arduino IDE to make a fresh installation of the tool. (I only have the native security app and defender)

rfaass, 2023-06-19:
Found these instructions to make it work, thank you for the assistance! : Open the project's "Releases" page in your browser:
[Releases · arduino/serial-discovery · GitHub 55](https://github.com/arduino/serial-discovery/releases)
On that page, find the release for the missing version (likely "1.3.2" in this case).
There will be a list of download links under the "Assets" section of the release (if you can't see the links, click on "Assets" to expand the section). Find the download link for your operating system and click it.
In this case, the correct link will be named "serial-discovery_v1.3.2_Windows_64bit.zip"
Wait for the download to finish.
Extract the downloaded file.
Copy the serial-discovery executable to the empty installation folder.
For example:
C:\Users\per\AppData\Local\Arduino15\packages\builtin\tools\serial-discovery\1.3.2
Restart Arduino IDE if it is running.