arduino/arduino-ide #2105: Mdns-discovery and windows firewall

liquidpower007, 2023-06-20:
### Describe the problem

The issue is that both the installer version of arduino and the portable have it so when a user launches the core program, a folder is made in said users app data folder in windows with the mdns discovery app. This app needs firewall access and will prompt the end user about this at launch. The issue is that A non admin cannot allow access. So normally we would create a firewall exception however, windows firewall doesn’t allow variable paths to an executable which is what is needed since the path is always going to change to reflect the username of whoever is signed into windows. I’ve installed the latest arduino and the problem persists and is seriously holding up an entire image.

### To reproduce

Open the program

### Expected behavior

The application should always be in a static location to allow an exception to be made in windows firewall. 

### Arduino IDE version

Whichever the latest is 

### Operating system

Windows

### Operating system version

Windows 10 LTSC 2021

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

skydiveguy, 2023-06-29:
Same issue here. I'm installing this on new PCs at a school and students do not have admin access to the machines and this will cause repeated calls to our support team every time a new student opens IDE up.
Stop installing items to APPDATA folders.

liquidpower007, 2023-06-30:
You are not going to be able to get around this without going to super extra mile. There is a Powershell script you can find online that essentially adds the firewall rule for each user that signs in. It is way overkill TBH. What I'd suggest doing is asking if they can use the Legacy IDE version 1.8.19. Even with that, make sure you download the ZIP file version and extract the contents to a location that doesn't require admins rights and then make a public desktop shortcut for the main app. 

I've been in touch with Arduino support and they have no solution as far as using the newer version. They are aware this is a problem for an environment like ours.

skydiveguy, 2023-06-30:
Thanks. They are using 1.8.19 now but I wanted to update them over the summer but whatever.

liquidpower007, 2023-06-30:
If this is by any chance for PLTW, they will even tell you to stick with that version. Additionally, it is an optional course in the PLTW program.

skydiveguy, 2023-06-30:
They follow PLTW curriculum but I install the latest version regardless.
I get in arguments all the time that I will not allow computers with outdated and unpatched software on my network.

per1234, 2023-09-28:
Hi @liquidpower007. Although the default location is under `%LOCALAPPDATA%\Arduino15\`, a custom location can be set in Arduino IDE's configuration file.

The file is at this location:

```text
%HOMEDRIVE%%HOMEPATH%\.arduinoIDE\arduino-cli.yaml
```

The path is set in the `directories.data` key of that file. The documentation for the file is available here:

https://arduino.github.io/arduino-cli/latest/configuration/

Arduino CLI's [`arduino-cli config`command](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli_config/) is made specifically for working with this configuration file. 

Since the `arduino-cli.yaml` file uses the standard [YAML](https://www.yaml.info/learn/index.html) data format, you can also use any number of convenient general tools to work with it. I can recommend the excellent [**yq**](https://mikefarah.gitbook.io/yq/).

closed 2023-09-28 by per1234