arduino/arduino-ide #2106: Won't upload data to Arduino Micro

ServerPlayerPro, 2023-06-20:
### Describe the problem

I am getting this error when trying to upload data to my Arduino Micro Board from the IDE version 2.1.0

> Sketch uses 4894 bytes (17%) of program storage space. Maximum is 28672 bytes.
> Global variables use 201 bytes (7%) of dynamic memory, leaving 2359 bytes for local variables. Maximum is 2560 bytes.
> Connecting to programmer: .
> Found programmer: Id = "CATERIN"; type = S
>     Software Version = 1.0; No Hardware Version given.
> Programmer supports auto addr increment.
> Programmer supports buffered memory access with buffersize=128 bytes.
> 
> Programmer supports the following devices:
>     Device code: 0x44

### To reproduce

Arduino Micro connected through Micro USB pushing from the portable version of the 2.1.0 IDE.

**Code:**
```
#include <Mouse.h>

const int upButton = 2;
const int downButton = 3;
const int leftButton = 4;
const int rightButton = 5;
const int mouseButtonL = 6;
const int fmouseButtonR = 7;

void setup() {  
  pinMode(upButton, INPUT);
  pinMode(downButton, INPUT);
  pinMode(leftButton, INPUT);
  pinMode(rightButton, INPUT);
  pinMode(mouseButtonL, INPUT);
  pinMode(fmouseButtonR, INPUT);

  Serial.begin(9600);
  Mouse.begin();
}

void loop() {
  if (Serial.available() > 0) {
    char inChar = Serial.read();

    switch (inChar) {
      case 'u':
        // move mouse up
        Mouse.move(0, -40);
        break;
      case 'd':
        // move mouse down
        Mouse.move(0, 40);
        break;
      case 'l':
        // move mouse left
        Mouse.move(-40, 0);
        break;
      case 'r':
        // move mouse right
        Mouse.move(40, 0);
        break;
      case 'm':
        // perform mouse left click
        Mouse.click(MOUSE_LEFT);
        break;
      case 'f':
        // perform mouse left click
        Mouse.click(MOUSE_RIGHT);
        break;
    }
  }
}
```

### Expected behavior

> Sketch uses 4894 bytes (17%) of program storage space. Maximum is 28672 bytes.
> Global variables use 201 bytes (7%) of dynamic memory, leaving 2359 bytes for local variables. Maximum is 2560 bytes.
> Connecting to programmer: .
> Found programmer: Id = "CATERIN"; type = S
>     Software Version = 1.0; No Hardware Version given.
> Programmer supports auto addr increment.
> Programmer supports buffered memory access with buffersize=128 bytes.
> 
> Programmer supports the following devices:
>     Device code: 0x44

### Arduino IDE version

2.1.0

### Operating system

Windows

### Operating system version

Windows 10 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-06-20:
Hi @ServerPlayerPro. This is the expected output from a successful upload to an ATmega32U4-based board. If Arduino IDE showed a "**Done uploading.**" notification in the bottom right corner of the IDE window: at the end of the upload then all is well and there is no problem:

[image]

If the upload fails, you will see an "**Upload error: ...**" notification in the bottom right corner of the IDE window:

[image]

If your project is not working as expected, you can make a post on Arduino Forum and we'll help you out over there:

https://forum.arduino.cc/

---

I'll share some additional explanations about this strange output:

## Why is this output shown?

If you have used Arduino IDE 1.x previously, this output might be unexpected to you. This is because of a change to how output is handled while uploading in Arduino IDE 2.x.

When the "Show verbose output during: upload" preference is disabled, Arduino IDE 1.x suppresses all output from the upload tool after a successful upload. Arduino IDE 2.x prints all output from the upload tool, leaving the control over the output verbosity to the developer of the boards platform, who defines the patterns of the commands that are run during an upload at each verbosity setting.

In the case of the "**Arduino AVR Boards**" platform, [`-q -q` flags are added](https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L106) to the `avrdude` command when verbose output is disabled.

https://github.com/arduino/ArduinoCore-avr/blob/master/platform.txt#L106

The [AVRDUDE documentation](https://avrdudes.github.io/avrdude/6.4/avrdude_3.html#Option-Descriptions) explains the effect:

> Disable (or quell) output of the progress bar while reading or writing to the device. Specify it a second time for even quieter operation.

So the idea is that by adding these flags, the `avrdude` command should print only the most important output. The AVRDUDE developers apparently considered this particular output to be so important that it should be printed even when the tool is run with these `-q -q` flags.

## Why is the output colored red?

The red color of the upload output text might give the impression that it indicates an error, but it is not so in this case.

The **AVRDUDE** developers [made the decision](https://www.nongnu.org/avrdude/environment.html#:~:text=all%20messages%20of%20any%20kind%20go%20to%20stderr) to print even output that is not related to an error condition to the "standard error" (AKA "stderr") stream instead of the more common practice of printing it on the "standard output" (AKA "stdout") stream. Because it is usually associated with errors or warnings, Arduino IDE colors all output received from the stderr stream red (at least when the default IDE theme is selected).

closed 2023-06-20 by per1234

webcode2, 2023-08-28:
Same here, trying to upload to micro.
here is my log

`Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44`