arduino/arduino-ide #2107: Cannot run Arduino IDE 2.1.0 in Native Wayland

Integral-Tech, 2023-06-20:
### Describe the problem

I'm using Arduino IDE 2.1.0 on KDE Plasma + Wayland, but I find it can only run under XWayland.

After adding ```--enable-features=UseOzonePlatform --ozone-platform=wayland``` parameters to force it to run under native Wayland, it crashes.

### To reproduce

Execute the following command to force it to run under native Wayland:
```bash
arduino-ide --enable-features=UseOzonePlatform --ozone-platform=wayland
```

It crashes:
```bash
zsh: segmentation fault (core dumped)  arduino-ide --enable-features=UseOzonePlatform --ozone-platform=wayland
```

Full log:
```bash
> arduino-ide --enable-features=UseOzonePlatform --ozone-platform=wayland
Arduino IDE 2.1.0
Starting backend process. PID: 126461
[126486:0620/143121.470045:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is egl, ANGLE is 
[126486:0620/143121.476150:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/integral/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-06-20T06:31:22.372Z root INFO Backend Object.initialize: 4.2 ms [Finished 1.007 s after backend start]
2023-06-20T06:31:22.372Z root INFO Backend Object.configure: 3.4 ms [Finished 1.007 s after backend start]
2023-06-20T06:31:22.372Z root INFO Backend MessagingContribution.onStart: 1.5 ms [Finished 1.007 s after backend start]
2023-06-20T06:31:22.373Z root INFO Backend OSBackendApplicationContribution.configure: 0.4 ms [Finished 1.054 s after backend start]
2023-06-20T06:31:22.374Z root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 1.054 s after backend start]
2023-06-20T06:31:22.374Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 1.054 s after backend start]
2023-06-20T06:31:22.375Z root INFO Backend PluginLocalizationBackendContribution.initialize: 46.3 ms [Finished 1.054 s after backend start]
2023-06-20T06:31:22.375Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.3 ms [Finished 1.054 s after backend start]
2023-06-20T06:31:22.376Z root INFO Backend ElectronTokenBackendContribution.configure: 0.7 ms [Finished 1.055 s after backend start]
2023-06-20T06:31:22.376Z config INFO >>> Initializing CLI configuration...
2023-06-20T06:31:22.376Z root INFO Backend PluginDeployerContribution.initialize: 0.5 ms [Finished 1.055 s after backend start]
2023-06-20T06:31:22.377Z root INFO Backend ConfigServiceImpl.onStart: 0.8 ms [Finished 1.056 s after backend start]
2023-06-20T06:31:22.377Z config INFO Loading CLI configuration from /home/integral/.arduinoIDE/arduino-cli.yaml...
2023-06-20T06:31:22.378Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 1.057 s after backend start]
2023-06-20T06:31:22.378Z root INFO Backend ArduinoDaemonImpl.onStart: 0.9 ms [Finished 1.057 s after backend start]
2023-06-20T06:31:22.378Z root INFO Backend NodeFileUploadService.configure: 1.6 ms [Finished 1.057 s after backend start]
2023-06-20T06:31:22.378Z daemon INFO Starting daemon from /opt/arduino-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-06-20T06:31:22.378Z discovery-log INFO start
2023-06-20T06:31:22.378Z discovery-log INFO start new deferred
2023-06-20T06:31:22.378Z root INFO Backend HostedPluginLocalizationService.initialize: 0.4 ms [Finished 1.058 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend BoardDiscovery.onStart: 0.9 ms [Finished 1.059 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend FileDownloadEndpoint.configure: 0.8 ms [Finished 1.059 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 1.060 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend PluginApiContribution.configure: 0.4 ms [Finished 1.060 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend HostedPluginReader.configure: 0.3 ms [Finished 1.061 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 1.061 s after backend start]
2023-06-20T06:31:22.379Z root INFO Backend PlotterBackendContribution.configure: 0.4 ms [Finished 1.062 s after backend start]
2023-06-20T06:31:22.381Z root INFO Theia app listening on http://localhost:41219.
2023-06-20T06:31:22.382Z root INFO Finished starting backend application: 3.2 ms [Finished 1.064 s after backend start]
2023-06-20T06:31:22.383Z root WARN The local plugin referenced by local-dir:/home/integral/.arduinoIDE/plugins does not exist.
2023-06-20T06:31:22.383Z root WARN The local plugin referenced by local-dir:/home/integral/.arduinoIDE/extensions does not exist.
2023-06-20T06:31:22.398Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-20T06:31:22.398Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-06-20T06:31:22.398Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-20T06:31:22.398Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.8" with engines:
2023-06-20T06:31:22.398Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-20T06:31:22.398Z root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
2023-06-20T06:31:22.398Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-20T06:31:22.398Z root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
2023-06-20T06:31:22.398Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-06-20T06:31:22.398Z root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
2023-06-20T06:31:22.399Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... 133 lines cut ...]

It should NOT crash.

### Arduino IDE version

2.1.0

### Operating system

Linux

### Operating system version

Arch Linux, rolling

### Additional context

Arch Wiki: https://wiki.archlinux.org/title/Wayland#Electron
```--ozone-platform-hint=auto``` does NOT work (Arduino is stil running under XWayland with ```--ozone-platform-hint=auto```)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-06-21:
Thank you for reporting it to us. @Integral-Tech, could you please try if starting the app from a terminal with the `--disable-gpu` flag solves the problem on your env? See https://github.com/arduino/arduino-ide/issues/1816#issuecomment-1384948811.

rdlu, 2023-07-02:
Hey, I'm having almost the same issue `arduino-ide killed by SIGSEGV` after adding the wayland flags.

I need it to make it correctly resize on my secondary monitor. It resizes only on primary using xwayland.

OS: Fedora 38
Dell XPS Plus 9320 with Intel GPU / 12th gen i7.
Using Arduino IDE version `2.1.1` with the following command line:

`flatpak run --branch=stable --arch=x86_64 --command=arduino-ide --file-forwarding cc.arduino.IDE2 --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu`

Without the wayland flags it opens fine, but resizing, changing dpi settings, etc shows several weird behaviors whenever I move the IDE window to a secondary display. Also due to xwayland it feels more sluggish than usual.

**Disable GPU makes no difference**

I have several electron apps running fine with wayland flags where needed, from chromium itself, to vscode, slack, obsidian, logseq, etc.

Thanks all!

Log:

```
[⚡] × flatpak run --branch=stable --arch=x86_64 --command=arduino-ide --file-forwarding cc.arduino.IDE2 --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu
Arduino IDE 2.1.1
Starting backend process. PID: 31
[31 zypak-helper] Using spawn strategy test 1 as set by environment
[59:0702/161235.136397:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is swiftshader, ANGLE is
[59:0702/161235.138819:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/rdlu/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-07-02T19:12:35.519Z root INFO Backend Object.initialize: 2.6 ms [Finished 0.434 s after backend start]
2023-07-02T19:12:35.519Z root INFO Backend Object.configure: 2.1 ms [Finished 0.434 s after backend start]
2023-07-02T19:12:35.519Z root INFO Backend MessagingContribution.onStart: 0.9 ms [Finished 0.434 s after backend start]
2023-07-02T19:12:35.520Z root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.456 s after backend start]
2023-07-02T19:12:35.520Z root INFO Backend DefaultWorkspaceServer.onStart: 0.2 ms [Finished 0.456 s after backend start]
2023-07-02T19:12:35.521Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 0.456 s after backend start]
2023-07-02T19:12:35.521Z root INFO Backend PluginLocalizationBackendContribution.initialize: 22.2 ms [Finished 0.456 s after backend start]
2023-07-02T19:12:35.521Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 0.456 s after backend start]
2023-07-02T19:12:35.522Z root INFO Backend ElectronTokenBackendContribution.configure: 0.5 ms [Finished 0.457 s after backend start]
2023-07-02T19:12:35.522Z config INFO >>> Initializing CLI configuration...
2023-07-02T19:12:35.522Z root INFO Backend PluginDeployerContribution.initialize: 0.3 ms [Finished 0.457 s after backend start]
2023-07-02T19:12:35.522Z root INFO Backend ConfigServiceImpl.onStart: 0.5 ms [Finished 0.457 s after backend start]
2023-07-02T19:12:35.523Z config INFO Loading CLI configuration from /home/rdlu/.arduinoIDE/arduino-cli.yaml...
2023-07-02T19:12:35.523Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.2 ms [Finished 0.458 s after backend start]
2023-07-02T19:12:35.523Z root INFO Backend ArduinoDaemonImpl.onStart: 0.5 ms [Finished 0.458 s after backend start]
2023-07-02T19:12:35.523Z root INFO Backend NodeFileUploadService.configure: 1.1 ms [Finished 0.458 s after backend start]
2023-07-02T19:12:35.523Z daemon INFO Starting daemon from /app/arduino-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-07-02T19:12:35.523Z discovery-log INFO start
2023-07-02T19:12:35.523Z discovery-log INFO start new deferred
2023-07-02T19:12:35.523Z root INFO Backend HostedPluginLocalizationService.initialize: 0.3 ms [Finished 0.459 s after backend start]
2023-07-02T19:12:35.523Z root INFO Backend BoardDiscovery.onStart: 0.6 ms [Finished 0.459 s after backend start]
2023-07-02T19:12:35.523Z root INFO Backend FileDownloadEndpoint.configure: 0.5 ms [Finished 0.459 s after backend start]
2023-07-02T19:12:35.523Z root INFO Backend AuthenticationServiceImpl.onStart: 0.3 ms [Finished 0.460 s after backend start]
2023-07-02T19:12:35.524Z root INFO Backend PluginApiContribution.configure: 0.3 ms [Finished 0.460 s after backend start]
2023-07-02T19:12:35.524Z root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 0.461 s after backend start]
2023-07-02T19:12:35.524Z root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 0.461 s after backend start]
2023-07-02T19:12:35.524Z root INFO Backend PlotterBackendContribution.configure: 0.2 ms [Finished 0.461 s after backend start]
2023-07-02T19:12:35.525Z root INFO Theia app listening on http://localhost:34883.
2023-07-02T19:12:35.526Z root INFO Finished starting backend application: 2.1 ms [Finished 0.463 s after backend start]
2023-07-02T19:12:35.526Z root WARN The local plugin referenced by local-dir:/home/rdlu/.arduinoIDE/plugins does not exist.
2023-07-02T19:12:35.526Z root WARN The local plugin referenced by local-dir:/home/rdlu/.arduinoIDE/extensions does not exist.
2023-07-02T19:12:35.533Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-07-02T19:12:35.533Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
[... cut ...]
2023-07-02T19:12:36.004Z root INFO Deployed backend plugin "ms-ceintl.vscode-language-pack-zh-hans@1.53.2" from "/app/arduino-ide/resources/app/plugins/vscode-language-pack-zh-hans/extension": 19.5 ms [Finished 0.939 s after backend start]
2023-07-02T19:12:36.005Z root INFO Deploy plugins list: 51.2 ms [Finished 0.942 s after backend start]
/app/bin/arduino-ide: line 10:     3 Segmentation fault      (core dumped) env TMPDIR=$XDG_CACHE_HOME zypak-wrapper /app/arduino-ide/arduino-ide $CUSTOM_IDE_FLAGS "$@"
```

kittaakos, 2023-07-03:
> [⚡] × flatpak run --branch=stable --arch=x86_64 --command=arduino-ide --file-forwarding cc.arduino.IDE2 --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu

I see you're using FlatPack which is not officially supported by IDE2: https://github.com/arduino/arduino-ide/issues/1848.

@joshuag1000, could you help come over this problem?

> /app/bin/arduino-ide: line 10:     3 Segmentation fault      (core dumped) env TMPDIR=$XDG_CACHE_HOME zypak-wrapper /app/arduino-ide/arduino-ide $CUSTOM_IDE_FLAGS "$@"

The problematic code is with the logger patching https://github.com/arduino/arduino-ide/commit/0b33b517008dc2b5e27e54aafc588200fbac9459, but I do not know much about the FlatPack support.

joshuag1000, 2023-07-03:
> Hey, I'm having almost the same issue `arduino-ide killed by SIGSEGV` after adding the wayland flags.
> 
> I need it to make it correctly resize on my secondary monitor. It resizes only on primary using xwayland.
> 
> OS: Fedora 38 Dell XPS Plus 9320 with Intel GPU / 12th gen i7. Using Arduino IDE version `2.1.1` with the following command line:
> 
> `flatpak run --branch=stable --arch=x86_64 --command=arduino-ide --file-forwarding cc.arduino.IDE2 --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu`
> 
> Without the wayland flags it opens fine, but resizing, changing dpi settings, etc shows several weird behaviors whenever I move the IDE window to a secondary display. Also due to xwayland it feels more sluggish than usual.
> 
> **Disable GPU makes no difference with wayland flags**.
> 
> disable-gpu It makes a bit more bearable on xwayland tho, bit less sluggish but it still tries to use the DPI settings from the primary display on the secondary one. (200% on primary and 100% on secondary).
> 
> I have several electron apps running fine with wayland flags where needed, from chromium itself, to vscode, slack, obsidian, logseq, etc.
> 
> Thanks all!
> 
> Log:
> 
> ```
> [⚡] × flatpak run --branch=stable --arch=x86_64 --command=arduino-ide --file-forwarding cc.arduino.IDE2 --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu
> Arduino IDE 2.1.1
> Starting backend process. PID: 31
> [31 zypak-helper] Using spawn strategy test 1 as set by environment
> [59:0702/161235.136397:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is swiftshader, ANGLE is
> [59:0702/161235.138819:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
> Using browser-only version of superagent in non-browser environment
> Configuration directory URI: 'file:///home/rdlu/.theia'
> Configuring to accept webviews on '^.+\.webview\..+$' hostname.
> 2023-07-02T19:12:35.519Z root INFO Backend Object.initialize: 2.6 ms [Finished 0.434 s after backend start]
> 2023-07-02T19:12:35.519Z root INFO Backend Object.configure: 2.1 ms [Finished 0.434 s after backend start]
> 2023-07-02T19:12:35.519Z root INFO Backend MessagingContribution.onStart: 0.9 ms [Finished 0.434 s after backend start]
> 2023-07-02T19:12:35.520Z root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.456 s after backend start]
> 2023-07-02T19:12:35.520Z root INFO Backend DefaultWorkspaceServer.onStart: 0.2 ms [Finished 0.456 s after backend start]
> 2023-07-02T19:12:35.521Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 0.456 s after backend start]
> 2023-07-02T19:12:35.521Z root INFO Backend PluginLocalizationBackendContribution.initialize: 22.2 ms [Finished 0.456 s after backend start]
> 2023-07-02T19:12:35.521Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 0.456 s after backend start]
> 2023-07-02T19:12:35.522Z root INFO Backend ElectronTokenBackendContribution.configure: 0.5 ms [Finished 0.457 s after backend start]
> 2023-07-02T19:12:35.522Z config INFO >>> Initializing CLI configuration...
> 2023-07-02T19:12:35.522Z root INFO Backend PluginDeployerContribution.initialize: 0.3 ms [Finished 0.457 s after backend start]
> 2023-07-02T19:12:35.522Z root INFO Backend ConfigServiceImpl.onStart: 0.5 ms [Finished 0.457 s after backend start]
> 2023-07-02T19:12:35.523Z config INFO Loading CLI configuration from /home/rdlu/.arduinoIDE/arduino-cli.yaml...
> 2023-07-02T19:12:35.523Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.2 ms [Finished 0.458 s after backend start]
> 2023-07-02T19:12:35.523Z root INFO Backend ArduinoDaemonImpl.onStart: 0.5 ms [Finished 0.458 s after backend start]
> 2023-07-02T19:12:35.523Z root INFO Backend NodeFileUploadService.configure: 1.1 ms [Finished 0.458 s after backend start]
> 2023-07-02T19:12:35.523Z daemon INFO Starting daemon from /app/arduino-ide/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
> 2023-07-02T19:12:35.523Z discovery-log INFO start
> 2023-07-02T19:12:35.523Z discovery-log INFO start new deferred
> 2023-07-02T19:12:35.523Z root INFO Backend HostedPluginLocalizationService.initialize: 0.3 ms [Finished 0.459 s after backend start]
> 2023-07-02T19:12:35.523Z root INFO Backend BoardDiscovery.onStart: 0.6 ms [Finished 0.459 s after backend start]
> 2023-07-02T19:12:35.523Z root INFO Backend FileDownloadEndpoint.configure: 0.5 ms [Finished 0.459 s after backend start]
> 2023-07-02T19:12:35.523Z root INFO Backend AuthenticationServiceImpl.onStart: 0.3 ms [Finished 0.460 s after backend start]
> 2023-07-02T19:12:35.524Z root INFO Backend PluginApiContribution.configure: 0.3 ms [Finished 0.460 s after backend start]
> 2023-07-02T19:12:35.524Z root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 0.461 s after backend start]
> 2023-07-02T19:12:35.524Z root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 0.461 s after backend start]
> 2023-07-02T19:12:35.524Z root INFO Backend PlotterBackendContribution.configure: 0.2 ms [Finished 0.461 s after backend start]
> 2023-07-02T19:12:35.525Z root INFO Theia app listening on http://localhost:34883.
> 2023-07-02T19:12:35.526Z root INFO Finished starting backend application: 2.1 ms [Finished 0.463 s after backend start]
> 2023-07-02T19:12:35.526Z root WARN The local plugin referenced by local-dir:/home/rdlu/.arduinoIDE/plugins does not exist.
[... cut ...]
With regards to wayland and chromium apps it is considered experimental.
It is also worth noting that this might have only fixed it on my system because of how it is configured.

Finally the above step (running `arduino-ide --log-level=warn --enable-features=UseOzonePlatform` before `arduino-ide --log-level=warn --enable-features=UseOzonePlatform --ozone-platform=wayland`) seemed to fix the issue for me on a non-flatpak build of the app. I cannot say why or if this will work for you but please do let me know if this helps.

- Joshua

per1234, 2023-09-28:
Hi @Integral-Tech. We are still waiting for you to provide the information requested in the previous reply (https://github.com/arduino/arduino-ide/issues/2107#issuecomment-1600466676) so that we can proceed with the investigation or evaluation of this.

Please comment here to add the requested information.

Integral-Tech, 2023-10-02:
> Hi @Integral-Tech. We are still waiting for you to provide the information requested in the previous reply ([#2107 (comment)](https://github.com/arduino/arduino-ide/issues/2107#issuecomment-1600466676)) so that we can proceed with the investigation or evaluation of this.
> 
> Please comment here to add the requested information.

@kittaakos It also crashes with ```--disable-gpu``` flag.

```
╰─ arduino-ide --enable-features=UseOzonePlatform --ozone-platform=wayland --disable-gpu
Arduino IDE 2.2.1
Checking for frontend application configuration customizations. Module path: /opt/arduino-ide/resources/app/lib/backend/electron-main.js, destination 'package.json': /opt/arduino-ide/resources/app/package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.2.1" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "0.34.0" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2023-08-31T14:18:20.829Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.2.1","cliVersion":"0.34.0","buildDate":"2023-08-31T14:18:20.829Z"}
Starting backend process. PID: 336151
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/integral/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-10-02T14:40:24.832Z root INFO Backend Object.initialize: 8.1 ms [Finished 0.671 s after backend start]
2023-10-02T14:40:24.833Z root INFO Backend Object.configure: 6.0 ms [Finished 0.671 s after backend start]
2023-10-02T14:40:24.833Z root INFO Backend a.onStart: 2.2 ms [Finished 0.671 s after backend start]
2023-10-02T14:40:24.834Z root INFO Backend a.configure: 0.7 ms [Finished 0.711 s after backend start]
2023-10-02T14:40:24.835Z root INFO Backend u.onStart: 0.5 ms [Finished 0.711 s after backend start]
2023-10-02T14:40:24.836Z root INFO Backend d.configure: 0.2 ms [Finished 0.711 s after backend start]
2023-10-02T14:40:24.837Z root INFO Backend d.initialize: 40.5 ms [Finished 0.712 s after backend start]
2023-10-02T14:40:24.837Z root INFO Backend c.onStart: 0.4 ms [Finished 0.712 s after backend start]
2023-10-02T14:40:24.838Z root INFO Backend s.configure: 1.3 ms [Finished 0.713 s after backend start]
2023-10-02T14:40:24.840Z config INFO >>> Initializing CLI configuration...
2023-10-02T14:40:24.840Z root INFO Backend c.initialize: 0.7 ms [Finished 0.714 s after backend start]
2023-10-02T14:40:24.840Z root INFO Backend P.onStart: 1.1 ms [Finished 0.714 s after backend start]
2023-10-02T14:40:24.841Z config INFO Loading CLI configuration from /home/integral/.arduinoIDE/arduino-cli.yaml...
2023-10-02T14:40:24.842Z daemon INFO Starting daemon from /opt/arduino-ide/resources/app/lib/backend/resources/arduino-cli...
2023-10-02T14:40:24.842Z root INFO Backend s.initialize: 0.5 ms [Finished 0.716 s after backend start]
2023-10-02T14:40:24.842Z root INFO Backend w.onStart: 1.3 ms [Finished 0.717 s after backend start]
2023-10-02T14:40:24.842Z root INFO Backend r.configure: 2.5 ms [Finished 0.717 s after backend start]
2023-10-02T14:40:24.843Z discovery-log INFO start
2023-10-02T14:40:24.843Z discovery-log INFO start new deferred
2023-10-02T14:40:24.843Z root INFO Backend d.initialize: 0.5 ms [Finished 0.718 s after backend start]
2023-10-02T14:40:24.843Z root INFO Backend y.onStart: 2.7 ms [Finished 0.720 s after backend start]
2023-10-02T14:40:24.843Z root INFO Backend r.configure: 2.5 ms [Finished 0.720 s after backend start]
2023-10-02T14:40:24.843Z root INFO Backend s.onStart: 0.8 ms [Finished 0.722 s after backend start]
2023-10-02T14:40:24.844Z root INFO Backend h.configure: 0.7 ms [Finished 0.722 s after backend start]
2023-10-02T14:40:24.844Z root INFO Backend a.configure: 0.6 ms [Finished 0.723 s after backend start]
2023-10-02T14:40:24.844Z root INFO Backend s.configure: 0.2 ms [Finished 0.723 s after backend start]
2023-10-02T14:40:24.844Z root INFO Backend l.configure: 0.2 ms [Finished 0.724 s after backend start]
2023-10-02T14:40:24.849Z root INFO Theia app listening on http://localhost:40961.
2023-10-02T14:40:24.850Z root INFO Finished starting backend application: 6.8 ms [Finished 0.729 s after backend start]
2023-10-02T14:40:24.859Z root WARN The local plugin referenced by local-dir:/home/integral/.arduinoIDE/plugins does not exist.
2023-10-02T14:40:24.859Z root WARN The local plugin referenced by local-dir:/home/integral/.arduinoIDE/extensions does not exist.
2023-10-02T14:40:24.882Z daemon INFO time="2023-10-02T22:40:24+08:00" level=info msg="Using config file: /home/integral/.arduinoIDE/arduino-cli.yaml"
time="2023-10-02T22:40:24+08:00" level=info msg="arduino-cli version 0.34.0"
[... cut ...]
time="2023-10-02T22:40:25+08:00" level=info msg="sending command START_SYNC to discovery builtin:mdns-discovery"
time="2023-10-02T22:40:25+08:00" level=info msg="from discovery builtin:mdns-discovery received message type: start_sync，消息：OK"
time="2023-10-02T22:40:25+08:00" level=info msg="from discovery builtin:serial-discovery received message type: start_sync，消息：OK"
2023-10-02T14:40:25.572Z root INFO Deploy plugins list: 83.9 ms [Finished 1.452 s after backend start]
zsh: segmentation fault (core dumped)  arduino-ide --enable-features=UseOzonePlatform --ozone-platform=wayland 
```

jgamage91, 2023-10-11:
Siimlar issue. the IDE does not work in Wayland SWAY.

firefoxlover, 2023-12-09:
Just wanting to add that Flatpak works for Apps needing USB access, and it isolates apps, has a permission system and works across all Distros. So I would really encourage you to support the Flatpak as the current Linux application model is simply not future proof and very insecure. There are already immutable distributions like steamOS, Fedora Atomic, Opensuse microOS etc. that rely on Faltpak nearly only