arduino/arduino-ide #2115: Executable file not found in $PATH

rei-vilo, 2023-06-26:
### Describe the problem

Arduino IDE 2.0 FlatPak and AppImage

```
/home/reivilo/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-objcopy -O ihex /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5CFF790619DC8A9CFB4AC4A8A887325E/Blink.ino.elf /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5CFF790619DC8A9CFB4AC4A8A887325E/Blink.ino.hex
adafruit-nrfutil dfu genpkg --dev-type 0x0052 --sd-req 0x00B6 --application /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5CFF790619DC8A9CFB4AC4A8A887325E/Blink.ino.hex /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5CFF790619DC8A9CFB4AC4A8A887325E/Blink.ino.zip
exec: "adafruit-nrfutil": executable file not found in $PATH

Compilation error: exec: "adafruit-nrfutil": executable file not found in $PATH
```

With Arduino IDE 1.8.19

```
adafruit-nrfutil dfu genpkg --dev-type 0x0052 --sd-req 0x00B6 --application /tmp/arduino_build_731695/Blink.ino.hex /tmp/arduino_build_731695/Blink.ino.zip
Zip created at /tmp/arduino_build_731695/Blink.ino.zip
/home/reivilo/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-size -A /tmp/arduino_build_731695/Blink.ino.elf
Sketch uses 21228 bytes (2%) of program storage space. Maximum is 815104 bytes.
Global variables use 3096 bytes (1%) of dynamic memory, leaving 234472 bytes for local variables. Maximum is 237568 bytes.
```

### To reproduce

+ Select Adafruit Feather nRF52840
+ Launch **Verify**

### Expected behavior

Successful build

### Arduino IDE version

2.1.0 FlatPak and AppImage

### Operating system

Linux

### Operating system version

Linux 450-G5 6.1.0-9-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux 

### Additional context

Running the command in a Terminal succeeds.

̀̀```
$ adafruit-nrfutil dfu genpkg --dev-type 0x0052 --application /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5444C4324CC846FCD36DD91858EE6709/Blink.ino.hex /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5444C4324CC846FCD36DD91858EE6709/Blink.ino.zip
Zip created at /home/reivilo/.var/app/cc.arduino.IDE2/cache/arduino/sketches/5444C4324CC846FCD36DD91858EE6709/Blink.ino.zip

$ echo $PATH
/home/reivilo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/reivilo/Applications:/home/reivilo/Applications/bin:/home/reivilo/.local/bin:/home/reivilo/Projets/Zephyr/zephyr:/usr/sbin

$ which adafruit-nrfutil
/home/reivilo/Applications/adafruit-nrfutil
```

Also tested, to no avail

+ Add `export PATH=$PATH:/usr/local/bin` to `.bashrc`
+ Add export PATH=$PATH:/usr/local/bin` to `.bash_profile`
+ Set `sudo flatpak override cc.arduino.IDE2 --filesystem=/usr/local/bin`

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-06-26:
FlatPack IDE2

// CC @joshuag1000

joshuag1000, 2023-06-26:
@rei-vilo with regards to flatpak this issue should be fixed. 
I have pushed an update to the flatpak release that should be available in a few hours.

To use the Adafruit nrfutil boards I have included the adafruit-nrfutil binary within the release.

This means to use these boards all you need to do is add the URL Adafruit provides to the board manager and install the board. 
For the flatpak release installing the adafruit-nrfutil pip package is not required.

Please could you double check that this update to the flatpak release fixes your issue (only on the flatpak version as I don't maintain the appimage) and just let me know.

Many thanks,
Joshua

closed 2023-06-27 by rei-vilo

SamuTheCoder, 2025-01-22:
For those of you who installed Arduino IDE using snap, please reinstall a version from the Arduino's website. I lost hours and hours on this, installed adafruit nrfutil and used symlinks,  virtual environments and changing PATH variable and all and it never worked. I don't know what is wrong with the snap version of Arduino, but it seems that it has some trouble with the PATH.

After installing the appimage, I never had this error again