arduino/arduino-ide #2116: Wrong code highlighting

FloBab73, 2023-06-27:
### Describe the problem

Marking a word should highlight the same word in other places in the code. Sometimes the behaviour isn't as expected and not consistent, when the words don't match exactly. Some points might be up for discussion what the intended behaviour is, but some behaviours are wrong imo.

### To reproduce

Version: 2.1.0
Date: 2023-04-19T15:31:10.185Z
CLI Version: 0.32.2

Copy the following code in the Arduino IDE: 
```
requestCredentials
helloworld

void loop() {
  requestCredentials()
  helloworld
  hello
}

void requestCredetials(){
}
```
(notice the last requestCredentials has a missing n)

Following things are noticable and at least the first one should be fixed:
1. highlighting `requestCredentials` (line 5) also highlights `requestCredetials` (line 10), notice the missing `n`. This also works the other way round
    - thats very evil because you don't notice the typo and wonder why the compiler throws an error
2. highlighting `requestCredentials` (line 1) also highlights `helloworld` (line 2) , even though the words are completely different
3. highlighting `helloworld` (line 6) shows different results when you delete `helloworld `in line 2, `hello `in line 7 only gets highlighted after deletion

### Expected behavior

Different words shouldn't be highlighted when highlighting another word, especially when it is recognised as functions or variables. 

### Arduino IDE version

Version: 2.1.0 Date: 2023-04-19T15:31:10.185Z CLI Version: 0.32.2

### Operating system

Windows

### Operating system version

Windows 10 Enterprise 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-06-27:
Thanks for reporting the defect to us. I could not find the steps to reproduce the bug consistently. It works from VS Code, Theia Blueprint 1.38, and the monaco playground. I have noticed that I had to change the language mode in IDE2 to trigger the bug.

https://github.com/arduino/arduino-ide/assets/1405703/ebbd56c0-9fe1-4325-ab25-684fd3e3ced7

We might need to update the `language-client` version [here](https://github.com/arduino/vscode-arduino-tools/blob/6b9ee4b8feac6116f50876eb7c2d566835f87f9d/package.json#L51C6-L51C27). At least, this is where I would start for looking for a solution.

kittaakos, 2023-06-27:
I am labeling it as a problem with the Arduino Language server. I have debugged the code, and I can confirm the following from the https://github.com/arduino/vscode-arduino-tools/:

Given the above snippet, when the selection is on the `requestCredetials` method, the VSIX is sending this params to the language server:

`document`:
```json
{
    "uri": {
        "$mid": 1,
        "path": "/Users/a.kitta/Documents/Arduino/xxx/xxx.ino",
        "scheme": "file"
    },
    "fileName": "/Users/a.kitta/Documents/Arduino/xxx/xxx.ino",
    "isUntitled": false,
    "languageId": "cpp",
    "version": 1,
    "isClosed": false,
    "isDirty": false,
    "eol": 1,
    "lineCount": 11
}
```

`position`:
```json
{
    "line": 9,
    "character": 18
}
```

highlighted `ranges` response from the language server:

```json
[
    {
        "range": {
            "start": {
                "line": 4,
                "character": 2
            },
            "end": {
                "line": 4,
                "character": 20
            }
        },
        "kind": 1
    },
    {
        "range": {
            "start": {
                "line": 9,
                "character": 5
            },
            "end": {
                "line": 9,
                "character": 22
            }
        },
        "kind": 1
    }
]
```

One can see that there are two highlighted ranges in the resolved response. I think this bug is outside of the context of this repository, but I will keep it open for visibility. Maybe other will notice the same.

php4fan, 2023-12-09:
I have just observed this in the Arduino IDE:

[image]

As you can see I mistyped `picChar` (there's no such variable, this would cause a compile error) instead of `pixChar` (this is an existing parameter of the function I'm in, this is what I intended to type).

And yet the IDE behaves as if I had typed it as intended.

Sounds to me like it could be the same bug

Eheran1, 2024-03-08:
Why is this still not fixed? Note how completely different code sections are highlighted and not just that specific example, this happens all the time. Also, the markers on the scroll bar have additional errors where they show matches that do not exist in the actual code and not show matches that do exist (second screenshot).
[image]

[image]