arduino/arduino-ide #2122: IDE2.11 does not start in Windows10

wavoigt, 2023-07-01:
### Describe the problem

If i try to start IDE 2.11 in Windows 10, nothing happens.
In Windows Task manager, it shows up with 0% CPU usage and 26,5 MB Ram usage.

### To reproduce

n.a.

### Expected behavior

n.a.

### Arduino IDE version

2.11

### Operating system

Windows

### Operating system version

10 22H2, latest updates

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-07-01:
Thanks for your report @wavoigt. I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Close all Arduino IDE windows if the IDE is already running.
1. Start [Windows "**File Explorer**"](https://support.microsoft.com/windows/find-and-open-file-explorer-ef370130-1cca-9dc5-e0df-2f7416fe1cb1).
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<username>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as Arduino IDE starts (or fails to start).
1. Wait for Arduino IDE to finish starting, or failing to start.
1. Switch back to the PowerShell window if the Arduino IDE window took the focus.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

wavoigt, 2023-07-01:
PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Failed to start the electron application.
Error [ERR_REQUIRE_ESM]: Must use import to load ES Module: C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\node_modules\p-map\index.js
require() of ES modules is not supported.
require() of C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\node_modules\p-map\index.js from C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\index.js is an ES module file as it is a .js file whose nearest parent package.json contains "type": "module" which defines all .js files in that package scope as ES modules.
Instead rename C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\node_modules\p-map\index.js to end in .cjs, change the requiring code to use import(), or remove "type": "module" from C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\node_modules\p-map\package.json.

    at new NodeError (node:internal/errors:370:5)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1128:13)
    at Module.load (node:internal/modules/cjs/loader:982:32)
    at Module._load (node:internal/modules/cjs/loader:823:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1006:19)
    at require (node:internal/modules/cjs/helpers:93:18)
    at Object.<anonymous> (C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\index.js:2:14)
    at Module._compile (node:internal/modules/cjs/loader:1110:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1140:10) {
  code: 'ERR_REQUIRE_ESM'
}

wavoigt, 2023-07-01:
@per1234
I found the solution:
the files in:
C:\Program Files\Arduino IDE\resources\app\node_modules\p-filter\node_modules\p-map\
were outdated, obviously from the old 2.1.0 installation or the last nightly build, and the installer of 2.1.1 did not overwrite them. I de-installed the old version completely and installed 2.1.1 in a clean folder.
Now it's working fine.
Thank you so much for your effort.

Mohsnb, 2023-12-09:
Hello,
1.Make a temp folder. (eg: "D:\arduiotemp\temp")
2.Create a .BAT  file and put this lines to it:
```
@echo off
@set TEMP=D:\arduiotemp\temp&& "C:\Users\Mohsen\Downloads\Programs\arduino-ide_nightly\Arduino IDE.exe" >NUL
```

Run this bat file as administrator.

GRBlake488, 2024-02-10:
I have the same problem. Here is my PowerShell file:

PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Arduino IDE 2.3.0
Checking for frontend application configuration customizations. Module path: C:\Program Files\Arduino IDE\resources\app\lib\backend\electron-main.js, destination 'package.json': C:\Program Files\Arduino IDE\resources\app\package.json
Setting 'theia.frontend.config.appVersion' application configuration value to: "2.3.0" (type of string)
Setting 'theia.frontend.config.cliVersion' application configuration value to: "0.35.2" (type of string)
Setting 'theia.frontend.config.buildDate' application configuration value to: "2024-02-07T14:04:32.754Z" (type of string)
Frontend application configuration after modifications: {"applicationName":"Arduino IDE","defaultTheme":{"light":"arduino-theme","dark":"arduino-theme-dark"},"defaultIconTheme":"none","electron":{"windowOptions":{}},"defaultLocale":"","validatePreferencesSchema":false,"preferences":{"window.title":"${rootName}${activeEditorShort}${appName}","files.autoSave":"afterDelay","editor.minimap.enabled":false,"editor.tabSize":2,"editor.scrollBeyondLastLine":false,"editor.quickSuggestions":{"other":false,"comments":false,"strings":false},"editor.maxTokenizationLineLength":500,"editor.bracketPairColorization.enabled":false,"breadcrumbs.enabled":false,"workbench.tree.renderIndentGuides":"none","explorer.compactFolders":false},"appVersion":"2.3.0","cliVersion":"0.35.2","buildDate":"2024-02-07T14:04:32.754Z"}
Starting backend process. PID: 21324
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///c%3A/Users/georg/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2024-02-10T03:39:50.604Z root INFO Backend Object.initialize: 9.1 ms [Finished 1.276 s after backend start]
2024-02-10T03:39:50.606Z root INFO Backend Object.configure: 7.4 ms [Finished 1.276 s after backend start]
2024-02-10T03:39:50.606Z root INFO Backend a.onStart: 4.1 ms [Finished 1.276 s after backend start]
2024-02-10T03:39:50.608Z root INFO Backend a.configure: 0.6 ms [Finished 1.328 s after backend start]
2024-02-10T03:39:50.609Z root INFO Backend d.onStart: 0.4 ms [Finished 1.329 s after backend start]
2024-02-10T03:39:50.612Z root INFO Backend d.configure: 0.2 ms [Finished 1.329 s after backend start]
2024-02-10T03:39:50.614Z root WARN Backend d.initialize took longer than the expected maximum 50 milliseconds: 53.0 ms [Finished 1.330 s after backend start]
2024-02-10T03:39:50.615Z root INFO Backend c.onStart: 1.0 ms [Finished 1.330 s after backend start]
2024-02-10T03:39:50.616Z root INFO Backend s.configure: 2.7 ms [Finished 1.333 s after backend start]
2024-02-10T03:39:50.622Z config INFO >>> Initializing CLI configuration...
2024-02-10T03:39:50.622Z root INFO Backend c.initialize: 1.5 ms [Finished 1.333 s after backend start]
2024-02-10T03:39:50.622Z root INFO Backend R.onStart: 2.0 ms [Finished 1.335 s after backend start]
2024-02-10T03:39:50.624Z config INFO Loading CLI configuration from c:\Users\georg\.arduinoIDE\arduino-cli.yaml...
2024-02-10T03:39:50.626Z daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\lib\backend\resources\arduino-cli.exe...
2024-02-10T03:39:50.626Z root INFO Backend s.initialize: 1.3 ms [Finished 1.337 s after backend start]
2024-02-10T03:39:50.626Z root INFO Backend w.onStart: 3.3 ms [Finished 1.339 s after backend start]
2024-02-10T03:39:50.626Z root INFO Backend r.configure: 5.6 ms [Finished 1.340 s after backend start]
2024-02-10T03:39:50.626Z discovery-log INFO start
2024-02-10T03:39:50.626Z discovery-log INFO start new deferred
2024-02-10T03:39:50.626Z root INFO Backend d.initialize: 1.1 ms [Finished 1.342 s after backend start]
2024-02-10T03:39:50.626Z root INFO Backend y.onStart: 6.4 ms [Finished 1.348 s after backend start]
2024-02-10T03:39:50.626Z root INFO Backend r.configure: 5.4 ms [Finished 1.348 s after backend start]
2024-02-10T03:39:50.627Z root INFO Backend s.onStart: 1.1 ms [Finished 1.351 s after backend start]
2024-02-10T03:39:50.627Z root INFO Backend h.configure: 1.0 ms [Finished 1.352 s after backend start]
2024-02-10T03:39:50.627Z root INFO Backend a.configure: 0.4 ms [Finished 1.352 s after backend start]
2024-02-10T03:39:50.627Z root INFO Backend l.configure: 0.1 ms [Finished 1.352 s after backend start]
2024-02-10T03:39:50.635Z root INFO Theia app listening on http://127.0.0.1:49186.
2024-02-10T03:39:50.635Z root INFO Finished starting backend application: 9.0 ms [Finished 1.361 s after backend start]
2024-02-10T03:39:50.674Z root WARN The local plugin referenced by local-dir:/c%3A/Users/georg/.arduinoIDE/plugins does not exist.
2024-02-10T03:39:50.674Z root WARN The local plugin referenced by local-dir:/c%3A/Users/georg/.arduinoIDE/extensions does not exist.
2024-02-10T03:39:50.675Z root WARN The local plugin referenced by local-dir:C:\Users\georg\.arduinoIDE\plugins does not exist.
2024-02-10T03:39:50.696Z config INFO Loaded CLI configuration: {"board_manager":{"additional_urls":[]},"build_cache":{"compilations_before_purge":10,"ttl":"720h0m0s"},"daemon":{"port":"50051"},"directories":{"data":"C:\\Users\\georg\\AppData\\Local\\Arduino15","downloads":"C:\\Users\\georg\\AppData\\Local\\Arduino15\\staging","user":"C:\\Users\\georg\\OneDrive\\Documents\\Arduino"},"library":{"enable_unsafe_install":false},"logging":{"file":"","format":"text","level":"info"},"metrics":{"addr":": [... cut ...]
The following problems were detected during the gRPC client initialization:
[platform-index] - code: 9, message: Error loading hardware platform: discovery teensy:teensy-discovery not found

    at C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:622866
    at E.evaluateErrorStatus (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:622882)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async s.<anonymous> (C:\Program Files\Arduino IDE\resources\app\lib\backend\main.js:2:622552)