arduino/arduino-ide #2124: macOS Apple Silicon (arm64) release is missing from GitHub

HelgeSeidel, 2023-07-01:
### Describe the problem

The latest release V2.1.1 doesn't contain a build for Apple Silicion (arm64). Same as #2044. I've noticed based on the dates in previous versions it has been added later.

### To reproduce

https://github.com/arduino/arduino-ide/releases/tag/2.1.1 < missing arm64
https://github.com/arduino/arduino-ide/releases/tag/2.1.0 < has arm64

### Expected behavior

When a new release is made also include the arm64 version.

### Arduino IDE version

2.1.1

### Operating system

macOS

### Operating system version

13.4.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

closed 2023-07-03 by kittaakos

HelgeSeidel, 2023-07-04:
@kittaakos , @per1234 2.1.1 for Apple Silicon (arm64) is available on GitHub, but the IDE can't find any update.

Error message: There are no recent updates available for the Arduino IDE

kittaakos, 2023-07-05:
> but the IDE can't find any update.

Thanks for getting back to us. This is expected for now. Unfortunately, building the artifacts for the release has become more complicated as we need to build Linux on 18.04 Ubuntu which GH Actions do not support, plus we have to build an M1 for macOS, which is also not supported by GH Actions.

Please bear with us.

kittaakos, 2023-07-07:
> is available on GitHub, but the IDE can't find any update.

This should be fixed now. If there is a problem, please ping me. Thank you!