arduino/arduino-ide #2125: Dark title bar when dark mode enabled

balazscseh, 2023-07-02:
### Describe the request

I want the title bar to be white when i use light theme of the app, and be dark when i use dark theme.

Also it would be great if there would be an option to change this automatically, based on system settings.

https://learn.microsoft.com/en-us/windows/apps/desktop/modernize/apply-windows-themes#enable-a-dark-mode-title-bar-for-win32-applications

### Describe the current behavior

No matter if i enable dark mode in Windows settings, or in the app, the title bar remains white.

[image]

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

11

### Additional context

I found almost the same previous request in the issue tracker BUT i also found #10044 in Theia's pull requests, if i understand correctly, it is possible to change the title bar to dark with custom settings.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-07-02:
> I found almost the same previous request in the issue tracker

Are you referring to this issue?: https://github.com/arduino/arduino-ide/issues/321

I see that was closed as fixed by https://github.com/arduino/arduino-ide/pull/2027. I'm not sure what the relevant change was in that PR though. I see that https://github.com/eclipse-theia/theia/pull/10044 was shipped in Theia 1.19.0, so already part of the Theia version 1.31.1 that was in use by Arduino IDE prior to the bump in https://github.com/arduino/arduino-ide/pull/2027 so it wasn't that the bump pulled in the support added in https://github.com/eclipse-theia/theia/pull/10044.

I believe the idea is that the issue can be resolved by the removal of the Windows native title bar entirely by changing the `window.titleBarStyle` [advanced setting](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings) to `custom`. The IDE's theming should then apply to the custom title bar. However, that setting doesn't work currently (https://github.com/arduino/arduino-ide/issues/1733). I see there is work in progress to make it functional: https://github.com/arduino/arduino-ide/pull/1870

kittaakos, 2023-07-03:
> I'm not sure what the relevant change was in that PR though.

Updating the electron version was the fix. See here: https://github.com/arduino/arduino-ide/pull/2027/files#diff-60e7eeb6bc8be48e5693ebc9be980d76bb7973ab861bf167c666b519d437f501R26.

If you get the nightly, this is how it works: https://www.arduino.cc/en/software#nightly-builds.

> Also it would be great if there would be an option to change this automatically, based on system settings.

It works like you asked for. However, IDE2 needs a restart to get the OS theme into the IDE2 theme.

https://github.com/arduino/arduino-ide/pull/1870 is just icing on supporting the custom title bar beside the native one.

If you're wondering why this feature is unavailable on 2.2.1 but in the nightly build, please see https://github.com/arduino/arduino-ide/pull/2119#issuecomment-1616342221.

I am closing at resolved.

closed 2023-07-03 by kittaakos

balazscseh, 2023-07-03:
Thanks guys!