arduino/arduino-ide #2127: Don't work popup tooltips for functions

EarlVadim, 2023-07-04:
### Describe the problem

tooltips for functions and variables work only for sketches located in the default folder %User/Documents/...%. 
But if in the Preferences the home folder is redefined to another one, then these hints no longer work.

### To reproduce

Change Home folder in Preferences.
Create New sketch (even empty) and check working popup hints on "loop" or "setup" functions..
Save sketch to Home folder.
Check for broken popup tooltips.

### Expected behavior

Broken tooltips.

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

Windows11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-07-04:
Hi @EarlVadim. This sounds like a known bug that has already been fixed:

- https://github.com/arduino/arduino-cli/issues/2156
- https://github.com/arduino/arduino-ide/issues/2034

Unfortunately we weren't able to include the fix in the Arduino IDE 2.1.1 release but it will definitely be in the next release.

For now, you can only get the fix by using a beta tester version of Arduino IDE. Please give that version a try and then reply here to let me know if the problem still occurs.

I'll provide instructions you can follow to install that version:

1. Sign in to your [**GitHub**](https://github.com/) account.
   (GitHub only allows downloads of the beta tester builds when you are signed in.)
1. Open this page in your browser:
   <https://github.com/arduino/arduino-ide/actions/runs/5243125693#artifacts>
1. Click the "**Windows_X86-64_zip**" link under the "**Artifacts**" section of the page.
1. Wait for the download to finish.
1. [Unzip](https://support.microsoft.com/windows/zip-and-unzip-files-f6dde0a7-0fec-8294-e1d3-703ed85e7ebc#:~:text=To%20unzip%20files,-Open%20File%20Explorer) the downloaded file.
1. Unzip the ZIP file you find inside the unzipped file to any convenient location on your computer.
   (yes, there are two layers of ZIP files 🙃)
1. Run the `Arduino IDE.exe` file from the unzipped folder.

When you install the IDE via this ZIP file, it will not interfere with any other installations of Arduino IDE on your computer so you are welcome to have both beta tester versions as well as the release version of Arduino IDE installed on your computer at the same time.

EarlVadim, 2023-07-04:
Hello @per1234 .
Thank you for quick answer..
Yes, snapshot  d79bc0d is worked.
Thank you again.

per1234, 2023-07-08:
Hi @EarlVadim. Please provide detailed instructions I can follow to reproduce the faults you reported.

EarlVadim, 2023-07-11:
Hi, @per1234 

I can't write the instruction, I see these errors on my own projects. 
I'll just give two screenshots.
In the first screenshot, popup hints work, but not all repeated selections are displayed. project for ESP32.

[image]

on the second screenshot, all repetitions are displayed, but pop-up hints do not work. project for esp8266.

[image]

both projects are open at the same time, but their behavior is different.
hope you find the reason.

EarlVadim, 2023-09-03:
Hello again..
In new version 2.2.1 also has big problems.

[image]

The cursor in the picture is on "voltageInput" but highlighted duplicate values are anywhere.

per1234, 2023-11-03:
Hi @EarlVadim. Please comment to let us know whether you are still experiencing this problem.

If you are, I'm going to ask you to provide some additional information that might allow us to identify the cause (we are not able to reproduce the fault on our own machines, which makes it difficult to investigate):

I'm going to ask you to provide the debug logs that are produced by the "**Arduino Language Server**" tool. 

You can do that by following these instructions:

## A. Enable Language Server Logging

1. Use the <kbd>**Ctrl**</kbd>/<kbd>**⌘**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. Type `arduino.language.log` in the "**Search Settings**" field.
1. Check the box under the "**Arduino > Language: Log**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

## B. Gather Logs

1. Perform whatever procedure causes the problem you are experiencing in Arduino IDE.
1. Select **Sketch > Show Sketch Folder** from the Arduino IDE menus.
   The folder containing the sketch will open in your file manager.
1. Right click the sketch folder.
   A context menu will open.
1. Select "**Send to**" from the context menu.
   A **Send to**" sub-menu will open.
1. Select "**Compressed (zipped) folder**" from the "**Send to**" menu.
1. Attach the "ZIP" file that was created to a comment here on this GitHub issue.
   ⚠ Don't use an email reply to make the comment. You must make the comment via the web interface here on the GitHub website in order for the attachment to be made:
   https://github.com/arduino/arduino-ide/issues/2127
   You can do this by drag and dropping the file onto the comment field, or by clicking the paper clip icon on the comment composer toolbar.

## C. Disable Language Server Logging

When language server logging is enabled, log files are saved to every sketch you open in Arduino IDE and they steadily grow in size whenever you are editing the sketch. So the logging is something you would usually want to disable whenever generation of logs is not needed.

1. Use the <kbd>**Ctrl**</kbd>/<kbd>**⌘**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut to open the "**Command Palette**" in Arduino IDE.
1. Select the "**Preferences: Open Settings (UI)**" command from the "**Command Palette**".
   The "**Preferences**" tab will now open in the editor panel.
1. Type `arduino.language.log` in the "**Search Settings**" field.
1. Uncheck the box under the "**Arduino > Language: Log**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

You are now welcome to delete the `.log` files that were added to the sketch folder.