arduino/arduino-ide #2130: Really long lines result in going to next line in error.

V205Github, 2023-07-07:
### Describe the problem

Follow instructions in "to reproduce"
The ');' is bumped down to the next line even though there is so much space.
Same case for the Serial print. It should be on the same line until end of the window.

[image]

let me know If not clear enough.
Thank you.

### To reproduce

Info:
Arduino IDE 2.1.1
OS: MacOS
OS version: macOS Monterey 12.6.2
Language: English
Open Arduino IDE 2.
Paste this in to a new sketch: 
`/*
  Test
  By V205
*/

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); // begin serial comunication.
  delay(1000);
  Serial.println(F("STARTING " __FILE__ " from " __DATE__ __TIME__));
  //delay(9000);
  // while (mySgp40.begin(/*duration = */ 10000) != true) {
  //   Serial.println(F("failed to connect to SGP 40 , please check the wiring")); // why is this on a new line? Why is it commented?
  //   delay(1000);
  // }
  Serial.println("This Is a Really Long Line..............Results in errors....................");
}

void loop() {
  // put your main code here, to run repeatedly:

}
`

### Expected behavior

I would expect the serial print to be on one line until the end of the window.
I would expect the ); to be on the previous line.
Let me know if you need more info.
[image]

### Arduino IDE version

2.1.1

### Operating system

macOS

### Operating system version

12.6.2

### Additional context

Sorry, but I did not use the nightly build because I have not installed it yet.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

V205Github, 2023-07-07:
Take a look at the commented section and serial print in the code.
Thanks

per1234, 2023-07-08:
Hi @V205Github. Thanks for taking the time to submit an issue report.

It looks like you have enabled word wrapping in the IDE. This is fairly easy to do accidentally because the default keyboard shortcut for cycling through the word wrap settings (<kbd>**Option**</kbd>+<kbd>**Z**</kbd>) is similar to the frequently used keyboard shortcut for Undo (<kbd>**Command**</kbd>+<kbd>**Z**</kbd>).

I'll provide instructions for two different ways you can disable word wrapping. Pick whichever one is more convenient for you:

## Via Keyboard Shortcut

1. Press <kbd>**Option**</kbd>+<kbd>**Z**</kbd> repeatedly until the editor has the desired wrapping behavior.

The keyboard shortcut cycles through the four different word wrap settings rather than being a simple on/off toggle, so you may have press the keyboard shortcut multiple times.

If you find that the default <kbd>**Option**</kbd>+<kbd>**Z**</kbd> keyboard shortcut is problematic, you are welcome to change the assigned keyboard shortcut via the dialog opened from the **Arduino IDE > Advanced > Keyboard Shortcuts** menu in Arduino IDE.

## Via Advanced Settings

You can configure the word wrapping behavior of Arduino IDE in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings)

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "Command Palette".
   A menu will appear on the editor toolbar.
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
   **ⓘ** You can scroll down through the list of commands to find it or type the name in the field.
   A "**Preferences**" tab will open in the Arduino IDE main panel.
1. Type `editor.wordWrap` in the "**Search Settings**" field of the "**Preferences**" tab.
1. Select the desired word wrap setting from the menu under the "**Editor: Word Wrap**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

---

Since the default value of the "**Editor: Word Wrap**" setting is "off", as you expected, I'll close this issue as no action needed.

closed 2023-07-08 by per1234

V205Github, 2023-07-08:
Thank you! Probably accidentally pressed option z instead of command z.
Is there a way to permanently disable that shortcut? 
Thank you.

per1234, 2023-07-08:
>Is there a way to permanently disable that shortcut?

Yes, but we try to keep the conversations in the issue tracker here tightly focused on information directly relevant to bugs or feature requests for the Arduino IDE 2.x codebase. Questions about using Arduino IDE are more appropriate for the Arduino Forum. Please ask over there and I'll be happy to provide instructions you can follow to disable the shortcut.

V205Github, 2023-07-08:
Will do.
Thank you.
Will post link here.

V205Github, 2023-07-08:
Link: https://forum.arduino.cc/t/disable-line-wrapping-shortcut-option-z-in-arduino-ide-2-x-x/1146052

V205Github, 2023-07-08:
Oh and sorry I did not use code tags properly. No idea how to do it in GitHub.