arduino/arduino-ide #2132: Add autocorrect/complete to IDE sketch comments and title?

V205Github, 2023-07-09:
### Describe the request

Hi. Could a Autocorrect/complete feature be added in code comments and sketch naming?
Thank you.
Also maybe add an option for disabling it.
Sorry if there already is this feature.

### Describe the current behavior

Add a comment in any sketch.
Try typing a typo and autocorrect/complete won’t fix it.

### Arduino IDE version

2.1.1

### Operating system

macOS

### Operating system version

Monterey 12.6.2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-07-10:
> Could a Autocorrect/complete feature be added in code comments and sketch naming?

Yes, it should be installed manually as a VS Code extension. This request is very similar to https://github.com/arduino/arduino-ide/issues/1342#issuecomment-1228702088.

IDE2 also has documentation on installing a 3rd party theme; the structure of the extension is the same. Please follow that documentation if you get stuck: https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#3rd-party-themes

1. Download version `2.0.1` of the  [`streetsidesoftware/vscode-spell-checker`](https://github.com/streetsidesoftware/vscode-spell-checker) from https://open-vsx.org/api/streetsidesoftware/code-spell-checker/2.0.1/file/streetsidesoftware.code-spell-checker-2.0.1.vsix.
2. When IDE2 is not running, copy the downloaded `.vsix` file into your `plugins` folder. The plugins folder is under `~/.arduinoIDE/plugins` on Linxu and macOS and on `C:\\<username>\\.arduinoIDE\\plugins` folder, where `<username>` is your Windows username. If the `plugins` folder does not exist, create it.
3. Start IDE2. ✨

I verified the steps a few minutes ago with `2.1.1`, and it works.

https://github.com/arduino/arduino-ide/assets/1405703/dfb4e90b-c862-46ff-a4b0-64abacc91472

Notes:
 - You must pick version `2.0.1` of the `vscode-spell-checker`. More recent versions are not compatible with IDE `2.1.1`.
 - Reference the supported functionality of the extension [here](https://open-vsx.org/extension/streetsidesoftware/code-spell-checker/2.0.1#functionality).
 - Theia/IDE2 has compatibility issues with the spell checker VS Code extensions:
    ```
    2023-07-10T10:00:54.480Z root INFO [hosted-plugin: 7512] Error: Command with id 'vscode.executeDocumentRenameProvider' is not registered.
        at t.CommandRegistryMainImpl.$executeCommand (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6709350)
        at t.RPCProtocolImpl.doInvokeHandler (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6674232)
        at t.RPCProtocolImpl.invokeHandler (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6673983)
        at t.RPCProtocolImpl.receiveRequest (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6673376)
        at t.RPCProtocolImpl.receiveOneMessage (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6672869)
        at file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6671373
        at file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:2269356
        at r.invoke (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:2269446)
        at s.fire (file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:2271240)
        at file:///Applications/Arduino%20IDE%202.1.1.app/Contents/Resources/app/lib/bundle.js:2:6675026
    ```

V205Github, 2023-07-10:
Wait. So does the plugin work with IDE 2.x.x?
You mentioned an error.
Bit complex but I will try it.
Thank you.