arduino/arduino-ide #2137: Arduino board continuously dis/reconnects in Ubuntu.

aliphys, 2023-07-12:
### Describe the problem

Arduino board continuously dis/reconnects in the IDE. I cannot upload any sketch.

### To reproduce

1. Open the AppImage IDE by running the `./arduino-ide_2.1.1_Linux_64bit.AppImage` command in the directory
2. Connect the Nano 33 BLE Sense (Or Portenta H7)
3. The board constantly disconnects and reconnects as shown in this video [Screencast from 2023-07-12 12:40:49.webm](https://github.com/arduino/arduino-ide/assets/75624145/568c3c30-7738-462f-93e7-c59cb88d5cca)
4. This is the console output that is constantly repeated
     ```
    2023-07-12T10:48:12.052Z discovery-log INFO {"eventType":"add","port":{"matchingBoardsList":[{"name":"Arduino Nano 33 BLE","fqbn":"arduino:mbed_nano:nano33ble","isHidden":false}],"port":{"address":"/dev/ttyACM0","label":"/dev/ttyACM0","protocol":"serial","protocolLabel":"Serial Port (USB)","propertiesMap":[["pid","0x005a"],["serialNumber","00000000000000007877035BF3A113E9"],["vid","0x2341"]],"hardwareId":"00000000000000007877035BF3A113E9"}},"error":""}
    2023-07-12T10:48:12.065Z root INFO Attached boards and available ports changed:
    2023-07-12T10:48:12.066Z root INFO  - Attached board: Arduino Nano 33 BLE [arduino:mbed_nano:nano33ble] on /dev/ttyACM0 Serial Port (USB)
    2023-07-12T10:48:12.066Z root INFO ------------------------------------------
    2023-07-12T10:48:12.082Z discovery-log INFO {"eventType":"remove","port":{"matchingBoardsList":[],"port":{"address":"/dev/ttyACM0","label":"","protocol":"serial","protocolLabel":"","propertiesMap":[],"hardwareId":""}},"error":""}
    2023-07-12T10:48:12.084Z daemon INFO INFO[0007] from discovery builtin:serial-discovery received message type: remove, port: /dev/ttyACM0 
     ```

### Expected behavior

I can select the board from the drop down list and upload to it. 

### Arduino IDE version

2.1.1

### Operating system

Linux

### Operating system version

Ubuntu 22.04

### Additional context

- The current user is added to the dialout group via the `sudo adduser user dialout` command
- To eliminate potential Serial conflicts, the `brltty` tool is uninstalled `sudo apt remove brltty`
- Output of `sudo dmesg | tail -f` 
    ```
    user@user-laptop:~/Downloads$ sudo dmesg | tail -f
    [sudo] password for user: 
    [ 1830.373843] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
    [ 1830.423944] usb 1-3: USB disconnect, device number 24
    [ 1830.731637] usb 1-3: new full-speed USB device number 25 using xhci_hcd
    [ 1830.883801] usb 1-3: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
    [ 1830.883821] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [ 1830.883830] usb 1-3: Product: Arduino Nano 33 BLE
    [ 1830.883837] usb 1-3: Manufacturer: Arduino
    [ 1830.883844] usb 1-3: SerialNumber: 00000000000000007877035BF3A113E9
    [ 1830.887146] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
    [ 1830.936194] usb 1-3: USB disconnect, device number 25

    ```

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

aliphys, 2023-07-12:
<details>
<summary>Full Console Output</summary>

```

user@user-laptop:~/Downloads$ ./arduino-ide_2.1.1_Linux_64bit.AppImage 
Arduino IDE 2.1.1
Starting backend process. PID: 24413
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[24436:0712/124802.772635:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/user/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-07-12T10:48:04.043Z root INFO Backend Object.initialize: 5.8 ms [Finished 1.573 s after backend start]
2023-07-12T10:48:04.043Z root INFO Backend Object.configure: 4.9 ms [Finished 1.573 s after backend start]
2023-07-12T10:48:04.043Z root INFO Backend MessagingContribution.onStart: 1.7 ms [Finished 1.573 s after backend start]
2023-07-12T10:48:04.044Z root INFO Backend OSBackendApplicationContribution.configure: 0.5 ms [Finished 1.626 s after backend start]
2023-07-12T10:48:04.045Z root INFO Backend DefaultWorkspaceServer.onStart: 0.4 ms [Finished 1.626 s after backend start]
2023-07-12T10:48:04.046Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.2 ms [Finished 1.626 s after backend start]
2023-07-12T10:48:04.046Z root WARN Backend PluginLocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 52.8 ms [Finished 1.626 s after backend start]
2023-07-12T10:48:04.046Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.4 ms [Finished 1.626 s after backend start]
2023-07-12T10:48:04.047Z root INFO Backend ElectronTokenBackendContribution.configure: 0.8 ms [Finished 1.627 s after backend start]
2023-07-12T10:48:04.048Z config INFO >>> Initializing CLI configuration...
2023-07-12T10:48:04.048Z root INFO Backend PluginDeployerContribution.initialize: 0.5 ms [Finished 1.627 s after backend start]
2023-07-12T10:48:04.048Z root INFO Backend ConfigServiceImpl.onStart: 0.9 ms [Finished 1.628 s after backend start]
2023-07-12T10:48:04.049Z config INFO Loading CLI configuration from /home/user/.arduinoIDE/arduino-cli.yaml...
2023-07-12T10:48:04.049Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 1.629 s after backend start]
2023-07-12T10:48:04.049Z root INFO Backend ArduinoDaemonImpl.onStart: 0.9 ms [Finished 1.630 s after backend start]
2023-07-12T10:48:04.050Z root INFO Backend NodeFileUploadService.configure: 1.7 ms [Finished 1.630 s after backend start]
2023-07-12T10:48:04.050Z daemon INFO Starting daemon from /tmp/.mount_arduinyHzS6p/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-07-12T10:48:04.050Z discovery-log INFO start
2023-07-12T10:48:04.050Z discovery-log INFO start new deferred
2023-07-12T10:48:04.050Z root INFO Backend HostedPluginLocalizationService.initialize: 0.5 ms [Finished 1.631 s after backend start]
2023-07-12T10:48:04.050Z root INFO Backend BoardDiscovery.onStart: 1.0 ms [Finished 1.631 s after backend start]
2023-07-12T10:48:04.050Z root INFO Backend FileDownloadEndpoint.configure: 0.8 ms [Finished 1.632 s after backend start]
2023-07-12T10:48:04.050Z root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 1.633 s after backend start]
2023-07-12T10:48:04.050Z root INFO Backend PluginApiContribution.configure: 0.4 ms [Finished 1.633 s after backend start]
2023-07-12T10:48:04.051Z root INFO Backend HostedPluginReader.configure: 0.3 ms [Finished 1.633 s after backend start]
2023-07-12T10:48:04.051Z root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 1.634 s after backend start]
2023-07-12T10:48:04.051Z root INFO Backend PlotterBackendContribution.configure: 0.9 ms [Finished 1.635 s after backend start]
2023-07-12T10:48:04.053Z root INFO Theia app listening on http://localhost:43403.
2023-07-12T10:48:04.054Z root INFO Finished starting backend application: 3.8 ms [Finished 1.637 s after backend start]
2023-07-12T10:48:04.067Z root WARN The local plugin referenced by local-dir:/home/user/.arduinoIDE/plugins does not exist.
2023-07-12T10:48:04.067Z root WARN The local plugin referenced by local-dir:/home/user/.arduinoIDE/extensions does not exist.
2023-07-12T10:48:04.073Z config INFO Loaded CLI configuration: {"board_manager":{"additional_urls":[]},"build_cache":{"compilations_before_purge":10,"ttl":"720h0m0s"},"daemon":{"port":"50051"},"directories":{"data":"/home/user/.arduino15","downloads":"/home/user/.arduino15/staging","user":"/home/user/Arduino"},"library":{"enable_unsafe_install":false},"logging":{"file":"","format":"text","level":"info"},"metrics":{"addr":":9090","enabled":true},"output":{"no_color":false},"sketch":{"always_export_binaries":false},"updater":{"enable_notification":true}}
2023-07-12T10:48:04.073Z config INFO 'directories.data' and 'directories.user' are set in the CLI configuration model.
2023-07-12T10:48:04.073Z config INFO Loaded the CLI configuration.
2023-07-12T10:48:04.362Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-07-12T10:48:04.362Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-07-12T10:48:04.362Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-07-12T10:48:04.362Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.8" with engines:
[... cut ...]
2023-07-12T10:48:14.262Z root INFO Attached boards and available ports changed:
2023-07-12T10:48:14.262Z root INFO  - Detached board: Arduino Nano 33 BLE [arduino:mbed_nano:nano33ble] from /dev/ttyACM0 Serial Port (USB)
2023-07-12T10:48:14.262Z root INFO ------------------------------------------
Closing channel on service path '/services/electron-window-ext'.
Closing channel on service path '/services/electron-window'.
Closing channel on service path '/services/ide-updater'.
Stored workspaces roots: 
No sketches were scheduled for deletion.

```

</details>

aliphys, 2023-07-12:
Related Github issues:
- https://github.com/arduino/arduino-ide/issues/1336
- https://github.com/arduino/arduino-ide/issues/1306
- https://github.com/arduino/arduino-ide/issues/1319
- https://github.com/arduino/arduino-ide/issues/1788

Related Arduino Forum Posts:
- https://forum.arduino.cc/t/ubuntu-arduino-ide-not-showing-any-ports/1043925
- https://forum.arduino.cc/t/arduino-linux-ubuntu-arduino-cant-access-dev-ttyusb0-solved/568027

kittaakos, 2023-07-13:
Great bug report, @aliphys 🙏 

This issue is related to https://github.com/arduino/arduino-ide/issues/1319.

> Arduino board continuously dis/reconnects

Does it happen in your local env? If yes, please help investigate it if you have the capacity.

1. The [nightly build](https://www.arduino.cc/en/software#nightly-builds) contains an important bug fix ([arduino/arduino-ide#2101](https://github.com/arduino/arduino-ide/pull/2101)). If you download and start the nightly with the `--log-level debug` flags as described [here](https://github.com/arduino/arduino-ide/issues/2100#issuecomment-1594818955) (with the macOS way), it will produce a correct log file. If you can reproduce the bug in IDE2, please quit the app, and share the most recent log file. The log files are at `~/.config/Arduino IDE/` on Linux.

2. We can use the Arduino CLI bundled inside the IDE2 to verify what is discovered. The Arduino CLI is at `/path/to/Arduino IDE/resources/app/node_modules/arduino-ide-extension/build/arduino-cli` on Linux. It's inside the app, so you might need the ZIP build artifact. Please verify the version first with the `./arduino-cli version` command. It must be `0.33.1`. Then if you run the `./arduino-cli board list watch --format json` command and attach the Arduino board, the CLI will produce JSON output to the console. The command is in watch mode, so if you attach/detach boards, it will update the terminal output. Please share the JSON output with us.

Please do not hesitate to ask if you need details on the instructions.

Thank you!

aliphys, 2023-07-14:
Hello @kittaakos :wave: 

I have followed your two points and documented the outcome:

---

> The [nightly build](https://www.arduino.cc/en/software#nightly-builds) contains an important bug fix (https://github.com/arduino/arduino-ide/pull/2101). If you download and start the nightly with the --log-level debug flags as described https://github.com/arduino/arduino-ide/issues/2100#issuecomment-1594818955 (with the macOS way), it will produce a correct log file. If you can reproduce the bug in IDE2, please quit the app, and share the most recent log file. The log files are at ~/.config/Arduino IDE/ on Linux.

I unzipped the  `arduino-ide_nightly-20230713_Linux_64bit` nightly build and ran the command `./arduino-IDE --log-level debug`. This is the initial output when the IDE starts up.

<details>
<summary>Terminal Output Upon IDE Startup</summary>

```

user@user-laptop:~/Downloads/arduino-ide_nightly-20230713_Linux_64bit$ ./arduino-ide --log-level debug
Arduino IDE 2.1.2-nightly-20230713
Starting backend process. PID: 45970
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///home/user/.arduinoIDE'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-07-14T03:18:51.069Z root INFO Backend Object.initialize: 12.2 ms [Finished 1.520 s after backend start]
2023-07-14T03:18:51.070Z root INFO Backend Object.configure: 9.3 ms [Finished 1.520 s after backend start]
2023-07-14T03:18:51.070Z root INFO Backend MessagingContribution.onStart: 3.7 ms [Finished 1.520 s after backend start]
2023-07-14T03:18:51.071Z root INFO Backend OSBackendApplicationContribution.configure: 0.9 ms [Finished 1.595 s after backend start]
2023-07-14T03:18:51.073Z root INFO Backend DefaultWorkspaceServer.onStart: 0.8 ms [Finished 1.595 s after backend start]
2023-07-14T03:18:51.075Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.3 ms [Finished 1.596 s after backend start]
2023-07-14T03:18:51.076Z root WARN Backend PluginLocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 75.3 ms [Finished 1.596 s after backend start]
2023-07-14T03:18:51.077Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.6 ms [Finished 1.596 s after backend start]
2023-07-14T03:18:51.078Z root DEBUG Starting the deployer with the list of resolvers [
  LocalDirectoryPluginDeployerResolverWithFallback {},
  GithubPluginDeployerResolver {
    unpackedFolder: '/tmp/github-remote',
    request: ElectronBackendRequestService {}
  },
  HttpPluginDeployerResolver {
    unpackedFolder: '/tmp/http-remote',
    request: ElectronBackendRequestService {}
  },
  LocalVSIXFilePluginDeployerResolver {
    environment: PluginVSCodeEnvironment { environments: [EnvVariablesServer] }
  }
]
2023-07-14T03:18:51.081Z root INFO Backend ElectronTokenBackendContribution.configure: 1.8 ms [Finished 1.598 s after backend start]
2023-07-14T03:18:51.097Z config INFO >>> Initializing CLI configuration...
2023-07-14T03:18:51.097Z root INFO Backend PluginDeployerContribution.initialize: 1.1 ms [Finished 1.599 s after backend start]
2023-07-14T03:18:51.098Z root INFO Backend ConfigServiceImpl.onStart: 1.9 ms [Finished 1.600 s after backend start]
2023-07-14T03:18:51.100Z config INFO Loading CLI configuration from /home/user/.arduinoIDE/arduino-cli.yaml...
2023-07-14T03:18:51.100Z root DEBUG Found the list of default plugins ID on env: local-dir:/home/user/Downloads/arduino-ide_nightly-20230713_Linux_64bit/resources/app/plugins
2023-07-14T03:18:51.100Z root DEBUG Found the list of plugins ID on env: local-dir:/home/user/.arduinoIDE/plugins
2023-07-14T03:18:51.100Z root DEBUG Found the list of default plugins ID from CLI: undefined
2023-07-14T03:18:51.101Z daemon INFO Starting daemon from /home/user/Downloads/arduino-ide_nightly-20230713_Linux_64bit/resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-07-14T03:18:51.101Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 1.0 ms [Finished 1.603 s after backend start]
2023-07-14T03:18:51.102Z root DEBUG HTTP file upload URL path: /file-upload
2023-07-14T03:18:51.102Z root DEBUG Backend file upload cache path: /tmp/theia_upload
2023-07-14T03:18:51.102Z root INFO Backend ArduinoDaemonImpl.onStart: 2.0 ms [Finished 1.604 s after backend start]
2023-07-14T03:18:51.102Z root INFO Backend NodeFileUploadService.configure: 3.8 ms [Finished 1.604 s after backend start]
2023-07-14T03:18:51.102Z discovery-log INFO start
2023-07-14T03:18:51.102Z discovery-log INFO start new deferred
2023-07-14T03:18:51.103Z root INFO Backend HostedPluginLocalizationService.initialize: 4.3 ms [Finished 1.609 s after backend start]
2023-07-14T03:18:51.103Z root INFO Backend BoardDiscovery.onStart: 19.9 ms [Finished 1.625 s after backend start]
2023-07-14T03:18:51.103Z root INFO Backend FileDownloadEndpoint.configure: 16.3 ms [Finished 1.625 s after backend start]
2023-07-14T03:18:51.103Z root INFO Backend AuthenticationServiceImpl.onStart: 1.6 ms [Finished 1.629 s after backend start]
2023-07-14T03:18:51.103Z root INFO Backend PluginApiContribution.configure: 1.3 ms [Finished 1.629 s after backend start]
2023-07-14T03:18:51.104Z root INFO Backend HostedPluginReader.configure: 0.6 ms [Finished 1.630 s after backend start]
[... cut ...]
    "serialNumber": "00000000000000007877035BF3A113E9",
    "vid": "0x2341"
  },
  "boards": [
    {
      "name": "Arduino Nano 33 BLE",
      "fqbn": "arduino:mbed_nano:nano33ble"
    }
  ]
}
{
  "type": "remove",
  "address": "/dev/ttyACM0",
  "protocol": "serial",
  "properties": {}
}
```
</details>

---

**Note:** I do not see any log file (such as `arduino-ide-debug.txt` refered to in https://github.com/arduino/arduino-ide/issues/2100#issuecomment-1597250665 ) in the mentioned folder. So I copied from the Terminal. 
[image]

ideadapt, 2024-02-06:
I had a similar issue, which was caused by brltty package on ubuntu. `sudo dmesg -w` output when connecting the USB-C cable to Arduino (other than USB-C cables do work ...):
```
[144808.883753] input: BRLTTY 6.4 Linux Screen Driver Keyboard as /devices/virtual/input/input450
[144808.885641] usb 3-2: usbfs: interface 0 claimed by ch341 while 'brltty' sets config #1
[144808.886135] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[144808.886178] ch341 3-2:1.0: device disconnected
```
The solution is also mentioned at https://www.reddit.com/r/archlinux/comments/q8uv43/help_arduino_not_recognizing_ttyusb/

Igorpereirag, 2024-03-02:
check if the ch340 drive is available or loaded on the system?