arduino/arduino-ide #2139: My company uses a proxy server and at the first launch of Arduino IDE 2.1.1 is not launched normally, the program remains in anticipation.

Dmitry-Chernikov, 2023-07-14:
### Describe the problem

My company uses a proxy server and at the first launch of Arduino IDE 2.1.1 is not launched normally, the program remains in anticipation. Previously, in the version of Arduino 1.8 it was possible to configure the file of automatic settings of the proxy server C: \ proxy.pac because a new version does not work for this.
Proxy.manual.hostname =
Proxy.manual.password =
Proxy.manual.port =
Proxy.manual.type = http
Proxy.manual.username =
proxy.pac.url = C: \ proxy.pac
Proxy.type = Auto

### To reproduce

[logfile.log](https://github.com/arduino/arduino-ide/files/12046495/logfile.log)
Link to video 14 days
https://dropmefiles.com/wbPos
The ability to configure version 1.8
[image]

### Expected behavior

So that at the first launch, it allows you to configure the server proxy automatic settings file. Or did not immediately connect to adrduino.cc repositories

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

windows 7

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Dmitry-Chernikov, 2023-07-14:
Linux is the same problem!
https://github.com/arduino/arduino-ide/issues/2134

kittaakos, 2023-07-14:
Thank you for taking the time and filing a bug report.

We are aware of proxy detection issues: https://github.com/arduino/arduino-ide/issues/1438. Please reference https://github.com/arduino/arduino-ide/issues/1438#issue-1374239570 for a workaround. You must manually adjust the [Arduino CLI configuration](https://arduino.github.io/arduino-cli/latest/configuration/) file.

The auto proxy detection is yet something we have to support in IDE2: https://github.com/arduino/arduino-ide/issues/29.

We prefer keeping one open GitHub issue for the same topic. I am closing it as a duplicate. Thanks for your understanding.

closed 2023-07-14 by kittaakos