arduino/arduino-ide #2140: Avoid recompiling if the code has not changed

Silver-Fang, 2023-07-15:
### Describe the request

I'm sharing a board with other people, each of us using different code. So every time I use it, I have to re-upload the same code to overwrite the previous person's code. Given the amount of Sketch code I have, the compile time is not negligible.
In my experience, Visual Studio saves the last compilation. If the code is completely unchanged, the compilation step takes only a few seconds to complete. Can Arduino IDE do the same?

### Describe the current behavior

Even though my code never changes, the IDE has to recompile it every time.

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

23H2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-07-17:
Thanks for taking the time and submitting this feature request. We are aware of this:

 - https://github.com/arduino/arduino-ide/issues/1789,
 - https://github.com/arduino/arduino-cli/issues/1996, and
 - https://github.com/arduino/arduino-cli/issues/2204.

We prefer to keep one GitHub issue for one topic; I am closing this one as a duplicate. Please feel free to comment on the open topics if you have additional details. I appreciate your understanding.

closed 2023-07-17 by kittaakos

raffaeler, 2023-10-18:
@kittaakos Is there a roadmap including this?
But "my" #1789 has been closed, #1996 has been marked as spam and this issue is "not planned" and closed as dupe.

I think everyone wants to know whether adopting a different IDE or not.
I am not polemic, I just don't want to waste your or my time, simple as that and nothing more.