arduino/arduino-ide #2163: Search and replace not working as expected

tchilton, 2023-08-05:
### Describe the problem

In a sketch with multiple tabs, search and replace has a replace all, but it only works on the single tab you are working on. 

Secondly, the text boxes use the whole width of your screen, moving the sketch code down, then only gives a very small search and replace box, which is not resizable, this is not very user friendly if you have a longer expression. Can you make the boxes much bigger or allow them to be re-sized

### To reproduce

Create a sketch with multiple tabs. Add some text to each tab. Attempt to search and replace one piece of text to another and choose the replace all option.
Observe that the replace all did not work on any other tab than the one you are working on. Seems to be replacing only within the single tab, thus its name is misleading. 
 
The old 1.x IDE could do this without problems, but it had an extra option for all tabs, this is missing in the current IDE

### Expected behavior

A replace all should, as its name implies, replace all instances of the source text with the target text within the sketch.
The lack of a search in all sketch tabs option is not present in the 2.x IDE implying that its covered by the replace all option.  Note that having the option to choose the scope of a single tab or all tabs is quite useful too.

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

10 22H2

### Additional context

Search and replace is quite slow to use compared to the older IDE

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-08-05:
Hi @tchilton.

>it only works on the single tab you are working on.

There are two search interfaces in Arduino IDE 2.x, each with its own scope:

- Per-file
- Sketch

The sketch scoped search is accessed by clicking the magnifying glass icon on the "Activity Bar" at the left side of the Arduino IDE window:

[image]

Use that one when you need to search or replace on the entire sketch.

>a very small search and replace box, which is not resizable

It is resizable. For the per-file interface, hover the mouse pointer over the left border, then click and drag to the desired width.

closed 2023-08-05 by per1234

tchilton, 2023-08-05:
Thanks for the info. That's very odd as a user interface design, particularly given that very well known keyboard shortcut Ctrl-F opens the one that is focused only on the tab and it doesn't tell you its for the tab

My point about the search is that the search box should be larger by default, its very small. Even though you can change its size 
 with the little left bar (I found the control a little later), that only lasts in the current session, its lost when you quit and re-open.

I get that you are trying to modernise the UI and in general, there are some really good things in there, but there are also a lot of annoyances, it seems you need to spend a lot of time adjusting window sizes, rather than coding. Load time is a lot slower than the older IDE, particularly when you have a number of tabs and if you resize the window during that time, it doesn't display properly unless you resize again after the tabs all open etc. You can't open a second instance of the IDE from the desktop icon, only from the file menu, etc.