arduino/arduino-ide #2166: IDE does not start when last project is on unavailable network path

Niko-O, 2023-08-08:
### Describe the problem

When starting Arduino IDE 2.1.1, it tries to restore the last opened project. If that project is on a shared folder that is now unavailable, the IDE never starts. It does not show an error message. It does not even open a window.
Starting the IDE from CMD prints this:
```
Failed to start the electron application.
[Error: UNKNOWN: unknown error, stat '\\192.168.1.20\Share\Test\main\main.ino'] {
  errno: -4094,
  code: 'UNKNOWN',
  syscall: 'stat',
  path: '\\\\192.168.1.20\\Share\\Test\\main\\main.ino'
}
```
I had to remove the workspace-object from %AppData%\arduino-ide\config.json, then it started again.

### To reproduce

1. Create a new sketch
2. Save it on a shared folder (e.g. on a NAS or on a folder shared from a different computer)
3. Close Arduino IDE
4. Unplug your network cable / disconnect WiFi
5. Start Arduino IDE

### Expected behavior

The IDE should start up, and show a new, empty project (possibly the default sketch that's also shown when clicking on File -> New Sketch). It should ideally show a warning that the previous project (insert path here) could not be reopened.

### Arduino IDE version

2.1.1

### Operating system

Windows

### Operating system version

Windows 10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-08-18:
Thanks for the report and for collecting the forum posts. It was super helpful.

To get an `UNKNOWN` error for `syscall: 'stat'` on Windows:

```js
const fs = require('fs');
const start = performance.now();
try {
  fs.statSync('\\\\10.0.0.200\\path');
} finally {
  console.log('took', performance.now() - start, 'ms');
}
```

IDE2 must handle `UNKNOWN` code here:

https://github.com/arduino/arduino-ide/blob/9a6a457bc44fb16f646b9e61a9e8b336b3666412/arduino-ide-extension/src/node/sketches-service-impl.ts#L692-L695

The timeout is ~25s on my Windows env; IDE2 must use slow tests.

https://github.com/nodejs/node/issues/19965#issuecomment-380750573

Another topic IDE2 should consider is this:

https://github.com/arduino/arduino-ide/blob/9a6a457bc44fb16f646b9e61a9e8b336b3666412/arduino-ide-extension/src/node/sketches-service-impl.ts#L679

So if, for whatever reason, IDE2 cannot open a sketch due to `EACCESS`, the app will hang at startup.

per1234, 2023-08-18:
Despite quite some efforts, I was never able to reproduce this due to https://github.com/arduino/arduino-ide/issues/1797. I'm actually quite mystified as to how @Niko-O managed that.

However, I had seen some reports on the forum of a similar problem involving restoring sketches from normal paths. I wasn't able to reproduce it by saving a sketch to a removable drive, then removing the drive before restarting the IDE. However, I noticed the pattern that the sketch was on `D:\` in both reports. This is interesting because that drive letter is typically associated with a DVD drive on computers that have one. My hypothesis was that attempting to restore a sketch from a DVD drive without a disc produced an unusual exception Arduino IDE was not designed to handle. So I tried it out and sure enough I was able to produce a hang under these conditions:

### To reproduce

#### Equipment

- DVD drive on your computer
- A removable USB drive

#### Set up

1. Open the Windows "Start" menu.
1. Type `create and format hard disk partitions` in the search field.
1. Select "**Create and format hard disk partitions**" from the list of search results.
   The "**Disk Management**" window will open.
1. Find the "CD-ROM" drive in the bottom panel of the "**Disk Management**" window.
1. Note the drive letter that is assigned to the "CD-ROM" drive.
   **ⓘ** I'll refer to it as `D:\` through the rest of these instructions, but the demo will work for any drive letter.
1. Right click on the drive.
   A context menu will open.
1. Select "**Change Drive Letter and Paths...**" from the context menu.
   The "**Change Drive Letter and Paths for ...**" dialog will open.
1. Click the "**Change...**" button in the dialog.
1. Select any free drive letter from the "**Assign the following drive letter**" menu.
1. Click the "**OK**" button
1. Click the "**Yes**" button in the confirmation dialog.
1. Connect the removable drive to your computer.
1. If the drive was not automatically assigned the `D:\` drive letter, repeat the above procedure via the "**Disk Management**" utility to assign it `D:\`.
1. Start Arduino IDE if it is not already running.
1. Select **File > Save As...**" from the Arduino IDE menus.
1. Save the sketch on the removable drive.
1. Close all Arduino IDE windows other than the one with the sketch you just saved.
1. Wait a few seconds.
1. Select **File > Quit** from the Arduino IDE windows.
1. Eject and disconnect the removable drive from your computer.
1. Open the Windows "Start" menu.
1. Type `create and format hard disk partitions` in the search field.
1. Select "**Create and format hard disk partitions**" from the list of search results.
   The "**Disk Management**" window will open.
1. Find the "CD-ROM" drive in the bottom panel of the "**Disk Management**" window.
1. Right click on the drive.
   A context menu will open.
1. Select "**Change Drive Letter and Paths...**" from the context menu.
   The "**Change Drive Letter and Paths for ...**" dialog will open.
1. Click the "**Change...**" button in the dialog.
1. Select "**D**" from the "**Assign the following drive letter**" menu.
1. Click the "**OK**" button
1. Click the "**Yes**" button in the confirmation dialog.
1. Eject the disk from the DVD drive if one is loaded.

#### Demo

1. Start Arduino IDE.

🐛 The IDE never starts.

The logs show the error:

```text
Restoring workspace roots: d:\SavedOnD
Failed to start the electron application.
[Error: UNKNOWN: unknown error, stat 'd:\SavedOnD'] {
  errno: -4094,
  code: 'UNKNOWN',
  syscall: 'stat',
  path: 'd:\\SavedOnD'
}
```

### Arduino IDE version

69ae38e

### Operating system

Windows

### Operating system version

11

Niko-O, 2023-08-18:
> I'm actually quite mystified as to how @Niko-O managed that.
[image]

If opening a project form a path that starts with `\\Server\Share` does not work for you, mounting it as a network drive might work.

per1234, 2023-08-19:
> mounting it as a network drive might work.

Thanks for the suggestion. I tried that, but was not able to reproduce the fault via this approach of opening a sketch from a path on a network drive (which is not subject to https://github.com/arduino/arduino-ide/issues/1797) and then causing the IDE to attempt to restore the sketch from that path on a startup after I had disconnected the drive.

Instead of producing the fault reported here as I hoped, the IDE only exhibited the intended behavior of falling back on opening a new sketch under these conditions, just the same as it does if it attempts to restore a sketch from a path on a standard removable drive that was removed from the computer since the previous IDE session.

Niko-O, 2023-08-19:
@per1234 
I suspect that's because when you unmount the drive, Windows knows for sure that anything inside that path is definitely not accessible.
If I keep the drive mounted but unplug the network connection and then start the IDE, I can reproduce the issue.
```
Failed to start the electron application.
[Error: UNKNOWN: unknown error, stat 'x:\Test\main'] {
  errno: -4094,
  code: 'UNKNOWN',
  syscall: 'stat',
  path: 'x:\\Test\\main'
}
```
(I find it a bit weird that the drive letter is small here. Windows should not care, but it's weird nontheless.)

commit eb2a1d96 (2023-08-18): fix: handle `UNKNOWN` code on `syscall: 'stat'`

Closes #2166
PR #2174 (2023-08-20): fix: handle `UNKNOWN` code on `syscall: 'stat'`
### Motivation

IDE2 must gracefully handle when it cannot restore a sketch from a network drive at startup.

### Change description

This PR relaxes the logic of how IDE2 checks whether a sketch is accessible. From now on, `UNKNOWN` errors are caught, and the missing sketch is not restored.

### Other information

Closes #2166

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and de [... cut]
commit b2566556 (2023-08-20): fix: handle `UNKNOWN` code on `syscall: 'stat'`

Closes #2166
closed 2023-08-20 by kittaakos